sipeed/NanoKVM #2: NanoKVM can be disconnect HDMI connection to PC or can be support 16:10 resolution?

Qynklee, 2024-07-11:
Hi Sipeed team, thanks you so much for amazing project. 
I have a question about it, I using Parsec for remoting PC. 
But Parsec with free license only capture display in physical resolution or failback to virtual display (with virtual display can be change resolution very easy). 
If NanoKVM is connected to PC, I only remote in NanoKVM resolution. 
So I think NanoKVM should be have a mode disconnect HDMI connection or I can do something to enhance it.

Zepan, 2024-07-19:
Hi, what do you mean " a mode disconnect HDMI connection" ? 
BTW, we will opensource frontend this month

Qynklee, 2024-07-24:
> Hi, what do you mean " a mode disconnect HDMI connection" ? BTW, we will opensource frontend this month

I mean "a mode disconnect HDMI connection" is option when I dont want to use NanoKVM to remote control my PC. 
NanoKVM HDMI can be unplug or disconnect from HDMI in PC to make [Parsec ](https://parsec.app/) failback to virtual display. With virtual display, so easy to change display resolution.
Or,
Can I manual set display resolution to higher than 1920x1080 in Windows (display is created by NanoKVM to capture display via HDMI) ? NanoKVM maybe not work with higher resolution, it's OK because I use Parsec instead.
[image]

azurshinano, 2024-07-31:
> Hi, what do you mean " a mode disconnect HDMI connection" ? BTW, we will opensource frontend this month

Dear nanokvm development team, hello !

I noticed that there are three components in nanokvm:

jpg_stream: cpp

NanoKVM-Server: go

web: react

The firmware uses buildroot as the base image and is all riscv architecture.

So which parts do you plan to open source [^1] [^2]?

React does not seem to be enough to dispel people's concerns, and because you use the riscv architecture, it makes security analysis complicated. If you can't let everyone have confidence in the security of this kind of very low-level hardware [^3] , it seems difficult to promote it?

I really hope that this product can create brilliant results in the market. Thank you for your hard work!

[^1]: https://github.com/sipeed/NanoKVM/issues/1#issuecomment-2218134286

[^2]: https://github.com/sipeed/NanoKVM/issues/2#issuecomment-2238529958

[^3]: https://github.com/sipeed/NanoKVM/issues/1#issuecomment-2217429857

polyzium, 2024-08-09:
> React does not seem to be enough to dispel people's concerns

While that's true, IIRC React produces a very large bundle, and moreover, it recently got into a [controversy](https://github.com/facebook/react/pull/26380#issue-1621855149). I look forward to creating an alternative frontend in Vue or Solid in the near future.

bilogic, 2024-08-10:
I need to connect my PC's HDMI to a monitor as well, is there a way to do it? Using a HDMI splitter?

Z2Z-GuGu, 2024-09-11:
@Qynklee Hi, we have recently updated the KVM application for 2.0.7 to support more resolutions, including 1440 * 900 (16:10), which you can try after updating the webpage. We have also tried using software to turn off HDMI reception, but unfortunately LT6911 does not support this feature.And due to the chip limitations of LT6911, 1920 * 1080 is the maximum supported resolution, and we are powerless to improve the resolution.

Z2Z-GuGu, 2024-09-11:
> I need to connect my PC's HDMI to a monitor as well, is there a way to do it? Using a HDMI splitter?

If the computer only has one HDMI output, HDMI splitter is the only option. If the host has additional output ports, you can try using a conversion cable to connect NanoKVM. We have previously tested some products (including HDMI splitter, DP to HDMI, VGA to HDMI) that are compatible, please note that VGA to HDMI is only supported in versions 2.0.7 and above.

Qynklee, 2024-09-11:
> @Qynklee Hi, we have recently updated the KVM application for 2.0.7 to support more resolutions, including 1440 * 900 (16:10), which you can try after updating the webpage. We have also tried using [...]

Thanks for your hard work 😄  I think this issue is solved 😄

closed 2024-09-11 by Qynklee