sipeed/NanoKVM #3: Support for Rack Mount

SchoolGuy, 2024-07-14:
Hey guys!

I recently learnt from a friend about this cool project. Is it possible to get a mounting bracket that attaches to a standard 19'' Rack? If you guys love the idea, maybe a 1U chassis with keystones for the 5-pack that is fed by a single power plug from the back of the chassis?

Regards,
SchoolGuy

Zepan, 2024-07-19:
Thank you for your suggestion! we make a vote on twitter: https://x.com/SipeedIO/status/1814200599782576539
We will figure it out depend on vote result

SchoolGuy, 2024-07-19:
@Zepan Thanks for creating the vote. Why not go with the cube form and offer an adapter that can be 3D printed? Then you probably get the best out of both worlds. Your micro cube form factor is a very awesome feature!

Zepan, 2024-07-19:
wow, I see, we can make a 3D printed adapter, the OLED side outside~

marv7000, 2024-07-25:
I agree with this, a rack mount adapter would be very cool

closed 2024-08-25 by Zepan