sipeed/NanoKVM #4: Keyboard not detected?

waterytart, 2024-07-19:
So, I'm connecting the NanoKVM to my server. The first time I tried, it worked. It has since offered an update to v1.9.1, and I also copied some .ISO files to its internal storage. Now I tried again, and I can't get it to connect the keyboard on the remote PC. It won't type anything, either direct keyboard input, or through the soft keyboard.

Is there any way to troubleshoot this?

polyzium, 2024-08-09:
Funnily enough, I recently ran into the same problem, only to find that the USB connector wasn't all the way in.
However, it wouldn't hurt to take a look at your USB devices and see if the keyboard is recognized. On Linux, you need to type `lsusb` in the terminal and look for `flyingrtx licheervnano`. If there is such a device, you may have to click the mouse button in the webUI and select "Reset HID" several times. In other cases, you may need to restart the daemon (see [wiki](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/3_user_guide.html#%E5%BC%82%E5%B8%B8%E6%81%A2%E5%A4%8D), use Google Translate) or even reboot the KVM completely (please note that it would send a reset signal for some reason if you have the Full version with the ATX-B board wired and plugged in).

The software is not mature yet, but they're slowly open sourcing it, so hopefully people can iron out such bugs.

Z2Z-GuGu, 2024-08-12:
> So, I'm connecting the NanoKVM to my server. The first time I tried, it worked. It has since offered an update to v1.9.1, and I also copied some .ISO files to its internal storage. Now I tried [...]

Hello, we have solved this problem in the last few app updates, you can update the app to 2.0.3 (or higher) in the web page and try again.

Z2Z-GuGu, 2024-08-12:
> Funnily enough, I recently ran into the same problem, only to find that the USB connector wasn't all the way in. However, it wouldn't hurt to take a look at your USB devices and see if the keyboard [...]

Thanks for the feedback, we reproduced this reboot issue and found it to be a hardware design mistake, if you have the tools for soldering, you can try adding 1K resistor pull-down to GND on the GPIO A23 and A27 pins (https://wiki.sipeed.com/hardware/zh/kvm/assets/NanoKVM/1_intro/NanoKVM_2.jpg) . And we'll fix this on the latest hardware.

polyzium, 2024-08-12:
> > Funnily enough, I recently ran into the same problem, only to find that the USB connector wasn't all the way in. However, it wouldn't hurt to take a look at your USB devices and see if the keyboard is recognized. On Linux, you need to type `lsusb` in the terminal and look for `flyingrtx licheervnano`. If there is such a device, you may have to click the mouse button in the webUI and select "Reset HID" several times. In other cases, you may need to restart the daemon (see [wiki](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/3_user_guide.html#%E5%BC%82%E5%B8%B8%E6%81%A2%E5%A4%8D), use Google Translate) or even reboot the KVM completely (please note that it would send a reset signal for some reason if you have the Full version with the ATX-B board wired and plugged in).
> > The software is not mature yet, but they're slowly open sourcing it, so hopefully people can iron out such bugs.
> 
> Thanks for your feedback, we reproduced this reboot issue and found it to be a hardware design mistake, if you have the tools for soldering, you can try adding 1K resistor pull-down to GND on the GPIO A23 and A27 pins (https://wiki.sipeed.com/hardware/zh/kvm/assets/NanoKVM/1_intro/NanoKVM_2.jpg) . And we'll fix this on the latest hardware.

Just to be clear, you're showing the pinout of the LicheeRV Nano, which is what the Lite version is based on, do I have to do the same for the Full version considering that the ATX board is separate and the Full version is some sort of a "cluster" on top of the LicheeRV Nano base board?

Z2Z-GuGu, 2024-08-21:
> Just to be clear, you're showing the pinout of the LicheeRV Nano, which is what the Lite version is based on, do I have to do the same for the Full version considering that the ATX board is separate [...]

I made a simple example of pulling the two pins to GND with the 1.5K resistor on hand, which can be referred to in this diagram 
[image]

polyzium, 2024-08-21:
Understood, I'll give this a try when I get the chance

closed 2024-08-25 by Zepan

Dark-Existed, 2024-09-25:
> I made a simple example of pulling the two pins to GND with the 1.5K resistor on hand, which can be referred to in this diagram [image]

Is the diagram missing? For me, I am not that familiar with the hardware, Add a detail diagram please.
@Z2Z-GuGu

Z2Z-GuGu, 2024-09-26:
@Dark-Existed 

I'm very sorry, there are some problems with the image bed that I uploaded before, which caused the image to not be displayed, so I re-uploaded the image: [image]

Z2Z-GuGu, 2024-09-26:
Unfortunately, on some older versions of hardware, even if the RESET control pin is pulled down with a resistor, it cannot solve the restart problem. The only solution is to disconnect the RESET jumper. The pull-down resistor is only effective for the power on/off button, so there will be no additional operation on the computer's power button when rebooting NanoKVM. I will update the explanation on the wiki later