sipeed/NanoKVM #5: Video feed stops when resolution is changed

polyzium, 2024-08-09:
The MJPEG video feed (or should I say HDMI input) works as long as the resolution is constant. But when the resolution changes, the HDMI indicator on the OLED screen goes out and I get either a black screen or a static image in the web frontend. I have to restart the daemon as described in the wiki each time this happens, sometimes even rebooting the device completely.
After restarting the daemon via SSH, I occasionally see `img = NULL` among other lines. Something tells me that whatever is responsible for the HDMI input (or its driver) cannot get an image. I think this could be due to a misconfigured device initialization.

Perhaps I should also add that a cheap Macrosilicon MS2109 based USB capture device can easily adapt to arbitrary resolutions.

I raised this issue on the Telegram channel, but got no response.
I hope this will be fixed as soon as possible, it has become quite annoying, especially on machines with integrated GPUs.

Zepan, 2024-08-10:
Seem our PC/SBC can't reproduce this bug, can you tell me what device are you using?

polyzium, 2024-08-10:
For testing purposes I connected the NanoKVM to an old B75 PC with i3-3220, no PCIe GPU, only internal GPU.
I've also had a similar problem on my B550 main machine with R7 5700X and RTX 3060 Ti GPU, but that happens when I explicitly change the host resolution in the system settings. I can provide videos if needed.

polyzium, 2024-08-12:
@Zepan So I decided to make some videos to better explain the problem. I hope this issue will be fixed soon.
### Click one of the following labels to view the linked videos
## [B75 (NanoKVM)](https://streamable.com/71hxuj)
This is my old B75 PC with the i3-3220 CPU, no external PCIe GPU, only the internal GPU (connected to the HDMI of the motherboard). I was struggling to get it to display anything after the POST screen. Rebooting the PC also renders the KVM display useless until I restart the KVM daemon itself, which could sometimes result in an unrecoverable error I explained earlier if it could not open the capture device (I select the lines at 03:27), requiring a complete reboot of the KVM. In this video, I have also included the live camera feed of the KVM's OLED screen and also connected the power LED wires for better clarity.
## [B75 (Macrosilicon)](https://streamable.com/q88ltq)
Same old B75 PC, but the HDMI output is connected to a cheap Macrosilicon capture device instead, which in turn is connected to my computer for capture. I use the NanoKVM web interface off-screen to control the PC. As you can see, it has absolutely no problem handling arbitrary resolutions (although it is stretched to 1080p, but that's not really a problem).
## [Steam Deck (NanoKVM)](https://streamable.com/k4opw8)
This time I tried the same thing with my Steam Deck OLED connected to a dock (no power LED due to different form factor). Even worse, I could not get the KVM to output a picture after changing the resolution to 720p. Not even after restarting the daemon or rebooting the KVM.
## [Steam Deck (Macrosilicon)](https://streamable.com/oo1ieu)
Same thing, but Steam Deck connected to the Macrosilicon capture device. Again, using the KVM web interface offscreen to control the Deck. It has no problem displaying arbitrary resolutions, but again stretched to 1080p.

polyzium, 2024-08-15:
Bump.
Any progress on this bug? I strongly recommend that you give this a high priority, as it essentially renders the device useless (at least until rebooted).

Stoufiler, 2024-08-17:
Got exactly the same problem as you !

Can boot from my Intel NUC but when I access to BIOS I lost the HDMI 

This is the ouput of `/var/log/syslog`

```
Aug 17 12:53:58 kvm-6da7 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 17 12:53:58 kvm-6da7 kern.err kernel: [  179.201905] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 17 12:53:59 kvm-6da7 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 17 12:53:59 kvm-6da7 kern.err kernel: [  180.225905] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
```

polyzium, 2024-08-17:
@Stoufiler Hey,
The syslog lines look interesting. I tried searching for the source file and the function mentioned, but Google came up with nothing. Something tells me this might be a driver failure, in which case the problem is deeper than I first thought. Perhaps someone with Linux kernel experience can help?

Stoufiler, 2024-08-17:
Sorry was not `/var/log/syslog` but `/var/log/messages`

I've tried to force restart the stream by using `/etc/init.d/S95nanokvm restart_stream` while (I hope) i'm in bios mode got this output : 
```
# /etc/init.d/S95nanokvm restart_stream
OK
# [SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
try release sys ok
rmmod: can't unload module 'soph_vpss': Resource temporarily unavailable
insmod: can't insert '/mnt/system/ko/soph_base.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_rtos_cmdqu.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_fast_image.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_mipi_rx.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_snsr_i2c.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vi.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vpss.ko': File exists
[_mmf_init]-1139: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
alloc_buf_size(294912): 294912, phy_addr(8cae8000): 0x8cae8000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8cae8000, ion_data.name:ISP_SHARED_BUFFER_0
ISP Vipipe(0) Allocate pa(0x8cae8000) va(0x0x3fc2924000) size(291424)
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fc30432e8
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
20240817 13:51:19.387 0856 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20240817 13:51:19.396 0856 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
[... 97 lines cut ...]

------------MIPI info-------------------
 Devno EccErr CrcErr HdrErr WcErr fifofull   decode
     0      1      0      0     1        0 yuv422-8
  Physical:       D0       D1       D2       D3       D4       D5
                  8c       3d        0       8c       87        0
   Digital:       D0       D1       D2       D3    CK_HS  CK_ULPS  CK_STOP   CK_ERR   Deskew
             hs_idle  hs_idle   hs_hst   hs_hst        1        0        0        0    start
img = NULL
```

These lines seems to be interesting, it detect the output of the display : 
```

[SAMPLE_COMM_VO_StartChn]-351: u32Width:368, u32Height:552, u32Square:1
-- [I] camera and display open success

-- [I] camera size: 1280x720

-- [I] disp size: 552x368
```

Which is 552x368, but still have **img = NULL**

Will continue to dig

Stoufiler, 2024-08-17:
I've booted up my nuc to the OS itself, and restarted the stream, now I have this and stream is working :

```
OK
# [SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
try release sys ok
rmmod: can't unload module 'soph_vpss': Resource temporarily unavailable
insmod: can't insert '/mnt/system/ko/soph_base.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_rtos_cmdqu.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_fast_image.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_mipi_rx.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_snsr_i2c.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vi.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vpss.ko': File exists
[_mmf_init]-1139: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
alloc_buf_size(294912): 294912, phy_addr(8cae8000): 0x8cae8000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8cae8000, ion_data.name:ISP_SHARED_BUFFER_0
relese PoolId: 1, PhysAddr: 0x8cb30000, BlkSize: 1382400, BlkCnt: 2 Free: 0
try release poolid:1 phy:0x8cb30000
try release poolid:1 phy:0x8cc81800
relese PoolId: 4, PhysAddr: 0x8d805000, BlkSize: 1382400, BlkCnt: 1 Free: 0
try release poolid:4 phy:0x8d805000
ISP Vipipe(0) Allocate pa(0x8cae8000) va(0x0x3fdecea000) size(291424)
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fdf4092e8
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
20240817 14:00:07.581 1014 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20240817 14:00:07.590 1014 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647       
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1              
sensorName0    6911           

PQBIN message
gerritId:      
              commitId:      fc6ce647       
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1              
sensorName0    4653           

author:        lxowalle       desc:          gc4653 30fps   
createTime:    2024-03-26 13:41:34version:                      
tool Version:       v3.0.5.48           mode:      
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
maix multi-media init ok
try release vio ok
try release venc ok
[_mmf_vpss_init_new]-1487: CVI_VPSS_CreateGrp(grp:0) retry(0xc0068004)!
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
-- [W] pwm 10 already exported, unexport first automatically
-- [E] open sys config failed: 6

[SAMPLE_COMM_VO_StartChn]-351: u32Width:368, u32Height:552, u32Square:1
-- [I] camera and display open success

-- [I] camera size: 1280x720

-- [I] disp size: 552x368

kvm_cfg.fps = 30
kvm_cfg.qlty = 80
kvm_cfg.res = 720
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
```

Got the same disp size: 552x368

So this should not be the problem

polyzium, 2024-08-17:
This is weird, according to my tests it only displays 1080p source video, not at any other resolution. Why it displays 552x368 is a mystery to me.

Stoufiler, 2024-08-17:
I've plug a mini dvi to hdmi adpater between my nuc and the nanokvm

Maybe it's the base resolution of the adapter

I will buy a mini-HDMI to HDMI today to see if I have the same error

I think I found what is the resolution of 552x368 it's the OLED display and the 1280x720 is the resolution of the video on web page

polyzium, 2024-08-17:
The OLED is 128x64, this cannot be real. I believe the KVM takes 1080p video and downscales it to 720p/600p/480p depending on the setting you choose.

Stoufiler, 2024-08-17:
[image]

As you can see screen is well displayed on my monitor but nothing on nanoKVM

But mouse and keyboard works really through the kvm

I can move the mouse on the black screen and it's well reflated on my monitor either for the keyboard

polyzium, 2024-08-17:
Depending on the machine and whether or not you're using a splitter, connecting both the KVM and the monitor to the same machine may result in one of the two being prioritized. The best way to check this IMO is to use a remote desktop application like RustDesk when the machine is connected to the KVM.

Stoufiler, 2024-08-19:
Yes of course, 

But how will work rustdeck if I am in the bios ?

polyzium, 2024-08-19:
Boot into the OS, change resolution inside the OS. See if the video feed stops. Use remote desktop as a failsafe.

Z2Z-GuGu, 2024-08-21:
Thanks for your feedback. Recently, I have finally found a similar device and reproduced the issue encountered by @polyzium : HDMI cannot detect when switching resolutions. The HDMI input of NanoKVM is completed by a chip called LT6911, which theoretically converts any resolution of HDMI image received into 1920 * 1080 and transmits it to LicheeRV Nano through CSI. Therefore, when building the driver program for LT6911, I fixed the size to the standard output of LT6911 and it has been effective on many platforms I have tested. Unfortunately, LT6911 cannot output to the CSI channel in standard size on a small number of devices, which has resulted in erroneous log output information. Next, I will try to solve this problem and provide feedback on my progress here.

Z2Z-GuGu, 2024-08-21:
> This is weird, according to my tests it only displays 1080p source video, not at any other resolution. Why it displays 552x368 is a mystery to me. What's the source resolution?

Actually, I added a little Easter egg in the program. If you purchase both Sipeed's MaixCAM and NanoKVM Lite at the same time, you can execute`echo panel=st7701_hd228001c31 > /boot/uEnv.txt && touch /etc/kvm/display && reboot`to activate MaixCAM's small screen, which can be used as a small computer sub screen, and 552x368 is the size of this small screen. The demonstration video is on my [Bilibili channel](https://www.bilibili.com/video/BV1RtYieKEhZ)

polyzium, 2024-08-23:
>  which theoretically converts any resolution of HDMI image received into 1920 * 1080

Well, this is what my Macrosilicon device does. However, as I said, on computers with dedicated GPUs, the image is upscaled to 1080p by the GPU inside the computer, not the chip.

> Unfortunately, LT6911 cannot output to the CSI channel in standard size on a small number of devices

You might want to take some inspiration from PiKVM, they somehow managed to get this right on a RPi 4B, via CSI as well.

> Next, I will try to solve this problem and provide feedback on my progress here.

PLEASE PLEASE, if you fix this, you'll save my life. This is probably the most annoying bug I have ever encountered in NanoKVM.

OrpheeGT, 2024-08-29:
Hello, I'm facing the same issue with a HP N54L with VGA to HDMI adapter.
This adapter works with a real screen, so I bet this is some bug between adapter and nanoKVM

/var/log/messages is flooded with 

```
Aug 29 19:05:17 kvm-e457 kern.err kernel: [ 2677.332077] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 29 19:05:18 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 29 19:05:18 kvm-e457 kern.err kernel: [ 2678.356135] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 29 19:05:19 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 29 19:05:19 kvm-e457 kern.err kernel: [ 2679.380196] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 29 19:05:20 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 29 19:05:20 kvm-e457 kern.err kernel: [ 2680.404239] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 29 19:05:21 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 29 19:05:21 kvm-e457 kern.err kernel: [ 2681.428293] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Aug 29 19:05:22 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Aug 29 19:05:22 kvm-e457 kern.err kernel: [ 2682.452349] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
```

Tried to restart stream as found above, but same behavior with flooded 

```
# /etc/init.d/S95nanokvm restart_stream
OK
# [SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
try release sys ok
[_mmf_init]-1139: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2158: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1751: devNum =  1
[parse_sensor_name]-1832: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1861: bus_id =  4
[parse_sensor_i2caddr]-1872: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1883: mipi_dev =  0
[parse_sensor_laneid]-1894: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1905: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1927: mclk_en =  0
[parse_sensor_mclk]-1938: mclk =  0
alloc_buf_size(294912): 294912, phy_addr(8c305000): 0x8c305000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8c305000, ion_data.name:ISP_SHARED_BUFFER_0
ISP Vipipe(0) Allocate pa(0x8c305000) va(0x0x3fd50a9000) size(291424)
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fd57c82e8
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
20240829 19:07:01.769 1369 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
[... 228 lines cut ...]
VIInImgWidth            :1920
VIInImgHeight           :1080
VIDevFPS                :  60
VIFPS                   :   0
VISofCh0Cnt             : 298
VIPreFECh0Cnt           :   0
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x3
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :  32
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0
```

Can it be fixed ?

Thank you

polyzium, 2024-08-30:
Any updates on the fix? I strongly suggest you give this a high priority.

Z2Z-GuGu, 2024-09-04:
> Any updates on the fix? I strongly suggest you give this a high priority.

Hi, in the past week I have tried to solve this problem, fortunately now I have successfully achieved normal switching between different resolutions. We've recently been adapting to the web page and doing a lot of testing, which will be updated this week.

polyzium, 2024-09-04:
Good to hear! Looking forward to the fix.

OrpheeGT, 2024-09-09:
Hello !

I updated to 2.0.7.

I'm still having issues with vga2hdmi output on my N54L : 

```
# /etc/init.d/S95nanokvm restart_stream
OK
# [KVM_stream]:start
kvm_cfg.fps = 30
kvm_cfg.qlty = 60
kvm_cfg.width = 640
kvm_cfg.height = 480
VI RES = 640 * 480 
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
640 * 480
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
ISP Vipipe(0) Allocate pa(0x89898000) va(0x0x3fd44d4000) size(291424)
pstSize->u32Width = 640
pstSize->u32Height = 480
stSnsrMode.u16Width 640 stSnsrMode.u16Height 480 60.000000 wdrMode 0 pstSnsObj 0x3fd4c402e8
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[lt6911_probe] -208 = exit success
[... 285 lines cut ...]
VICsiIntStatus0         :0x0
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0

Module: [MIPI_RX], Build Time[#19 PREEMPT Mon Sep 9 11:27:10 CST 2024]

------------Combo DEV ATTR--------------
   Devno  WorkMode  DataType   WDRMode         LinkId        PN Swap  SyncMode  DataEndian  SyncCodeEndian
       0      MIPIYUV422_8BIT      NONE  2, 4, 3, 1, 0  0, 0, 0, 0, 0       N/A         N/A             N/A

------------MIPI info-------------------
 Devno EccErr CrcErr HdrErr WcErr fifofull   decode
     0      0      0      0     0        0  unknown
  Physical:       D0       D1       D2       D3       D4       D5
                   0        0        0        0        0        0
   Digital:       D0       D1       D2       D3    CK_HS  CK_ULPS  CK_STOP   CK_ERR   Deskew
             hs_idle  hs_idle   hs_hst   hs_hst        0        0        1        0     idle
img = NULL
```

OrpheeGT, 2024-09-09:
Installed 2.0.7

it seems to be an improvement, but still a bug.

At boot, I see the POST bios screen, but when it switch to grub menu, then screen goes crazy  : 
[image]
[image]
[image]
[image]

```
Sep  9 16:41:12 kvm-e457 kern.err kernel: [  828.133182] vb_get_pool_info:536():  NULL VB HANDLE
Sep  9 16:41:12 kvm-e457 kern.info kernel: [  828.133929] vpss_open
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:214:lt6911_probe(): read sensor id error.
Sep  9 16:41:13 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:222:lt6911_probe(): Sensor ID Mismatch! Use the wrong sensor??
Sep  9 16:41:13 kvm-e457 local5.warn : [SYS-WRN] sample_common_bin.c:36:SAMPLE_COMM_BIN_ReadParaFrombin(): Bin exist (/mnt/cfg/param/cvi_sdr_bin)
Sep  9 16:41:13 kvm-e457 kern.info kernel: [  828.736675] open channel no. 0
Sep  9 16:41:13 kvm-e457 kern.info kernel: [  828.737023] close channel no. 0
Sep  9 16:41:13 kvm-e457 kern.info kernel: [  828.737082] open channel no. 0
Sep  9 16:41:15 kvm-e457 auth.err getty[3965]: can't open '/dev/ttyGS0': No such file or directory^M
Sep  9 16:41:18 kvm-e457 kern.warn kernel: [  833.748182] LT6911--:-- INT!!##
Sep  9 16:41:18 kvm-e457 kern.warn kernel: [  834.064504] LT6911--:-- INT!!##
Sep  9 16:41:19 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep  9 16:41:19 kvm-e457 kern.err kernel: [  834.735461] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep  9 16:41:19 kvm-e457 kern.info kernel: [  834.742066] close channel no. 0
Sep  9 16:41:19 kvm-e457 kern.info kernel: [  834.742274] open channel no. 0
Sep  9 16:41:19 kvm-e457 kern.info kernel: [  834.742586] close channel no. 0
Sep  9 16:41:19 kvm-e457 kern.info kernel: [  834.742670] open channel no. 0
Sep  9 16:41:19 kvm-e457 kern.info kernel: [  834.764403] close channel no. 0
Sep  9 16:41:20 kvm-e457 kern.info kernel: [  835.216061] vi_event_handler exit
Sep  9 16:41:20 kvm-e457 kern.info kernel: [  835.716052] vpss_release
Sep  9 16:41:22 kvm-e457 kern.warn kernel: [  837.880971] LT6911--:-- INT!!##
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep  9 16:41:22 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:714:SAMPLE_COMM_VI_StopViChn(): CVI_VI_DisableChn failed with 0xc00e8007!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep  9 16:41:22 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:948:SAMPLE_COMM_VI_StopSingleViPipe(): CVI_VI_StopPipe failed with 0xc00e8007!
Sep  9 16:41:22 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:971:SAMPLE_COMM_VI_StopViPipe(): SAMPLE_COMM_VI_StopViPipe is fail
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep  9 16:41:22 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:606:SAMPLE_COMM_VI_StopDev(): CVI_VI_DisableDev failed with 0xc00e8007!
Sep  9 16:41:22 kvm-e457 kern.err kernel: [  837.937563] vb_get_pool_info:536():  NULL VB HANDLE
Sep  9 16:41:22 kvm-e457 kern.err kernel: [  837.938635] vb_get_pool_info:536():  NULL VB HANDLE
Sep  9 16:41:22 kvm-e457 kern.info kernel: [  837.939314] vpss_open
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep  9 16:41:22 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
[... cut ...]
Sep  9 16:42:05 kvm-e457 auth.err getty[424]: can't open '/dev/ttyGS0': No such file or directory^M
Sep  9 16:42:16 kvm-e457 auth.err getty[455]: can't open '/dev/ttyGS0': No such file or directory^M
Sep  9 16:42:26 kvm-e457 auth.err getty[479]: can't open '/dev/ttyGS0': No such file or directory^M
Sep  9 16:42:36 kvm-e457 auth.err getty[504]: can't open '/dev/ttyGS0': No such file or directory^M
Sep  9 16:42:46 kvm-e457 daemon.info init: Id "acm" respawning too fast: disabled for 5 minutes
```

OrpheeGT, 2024-09-09:
I restarted kvm_stream 

```
# /etc/init.d/S95nanokvm restart_stream
OK
# [KVM_stream]:start
kvm_cfg.fps = 60
kvm_cfg.qlty = 99
kvm_cfg.width = 1400
kvm_cfg.height = 1050
VI RES = 1400 * 1050 
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
1400 * 1050
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 1400
pstSize->u32Height = 1050
alloc_buf_size(294912): 294912, phy_addr(8a617000): 0x8a617000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8a617000, ion_data.name:ISP_SHARED_BUFFER_0
pstSize->u32Width = 1400
pstSize->u32Height = 1050
pstSize->u32Width = 1400
pstSize->u32Height = 1050
ISP Vipipe(0) Allocate pa(0x8a617000) va(0x0x3fcdf3e000) size(291424)
pstSize->u32Width = 1400
pstSize->u32Height = 1050
stSnsrMode.u16Width 1400 stSnsrMode.u16Height 1050 60.000000 wdrMode 0 pstSnsObj 0x3fce6ab2e8
pstSize->u32Width = 1400
pstSize->u32Height = 1050
pstSize->u32Width = 1400
pstSize->u32Height = 1050
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe]
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[lt6911_probe] -208 = exit success
pstSize->u32Width = 1400
pstSize->u32Height = 1050
20240909 16:50:34.595 1706 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20240909 16:50:34.602 1706 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    6911

PQBIN message
gerritId:
              commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

author:        lxowalle       desc:          gc4653 30fps
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1400
pstSize->u32Height = 1050
maix multi-media init ok
try release vio ok
try release venc ok
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
1400 * 1050
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
-- [I] camera and display open success

-- [I] camera size: 1400x1050
```

I really doubt a linux tty screen scale at 1400x1050, but hey, maybe...

[image]

OrpheeGT, 2024-09-09:
Logs from a reboot for my with my vga2hdmi : 

I shrinked where there were handred lines with : start_cam_signal = 0
replaced by "....."

```
start_cam_signal = 0
HDMI res modification event
new res: 640 * 480
CSI res: 640 * 480
start_cam_signal = 0
start_cam_signal = 0
start_cam_signal = 0
start_cam_signal = 0
start_cam_signal = 0
VI RES = 640 * 480 
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
640 * 480
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
ISP Vipipe(0) Allocate pa(0x89898000) va(0x0x3fe3950000) size(291424)
pstSize->u32Width = 640
pstSize->u32Height = 480
stSnsrMode.u16Width 640 stSnsrMode.u16Height 480 60.000000 wdrMode 0 pstSnsObj 0x3fe40bd2e8
pstSize->u32Width = 640
pstSize->u32Height = 480
pstSize->u32Width = 640
pstSize->u32Height = 480
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe]
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[... 693 lines cut ...]
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1400
pstSize->u32Height = 1050
maix multi-media init ok
try release vio ok
try release venc ok
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
1400 * 1050
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
-- [I] camera and display open success

-- [I] camera size: 1400x1050
```

Z2Z-GuGu, 2024-09-11:
> At boot, I see the POST bios screen, but when it switch to grub menu, then screen goes crazy

@OrpheeGT Hi, the new update supports more resolutions, but due to limitations in the chip encoder, there may be display tearing issues at certain resolutions (known to not work properly: 1680 * 1050; 1440 * 1050; 1368 * 768; 1366 * 768; 720 * 576). We are trying to modify the EDID information of the HDMI module to mask these abnormal resolutions.

polyzium, 2024-09-11:
> We are trying to modify the EDID information of the HDMI module to mask these abnormal resolutions.

Why don't you let users tweak the EDID like you can with PiKVM?

OrpheeGT, 2024-09-11:
Any executable script would be perfect as long as it is easy to launch and well documented ! :)

babbaj, 2024-11-09:
> > Why don't you let users tweak the EDID like you can with PiKVM?
> 
> Good idea! I will consider doing so when making modifications, but there are some potential risks associated with modifying EDID. If custom modifications are ultimately supported, they will also appear as separate executable scripts

Any update on this?

delumerlino, 2025-02-27:
Have the same problem with my N54L.

NanoKVM recognize 1440x1050 resolution.
I use a VGA to VGA+HDMI adapter, I can see output on the VGA screen (and for test on the HDMI), but NanoKVM remote desktop is crazy.

It is also very slow to start, because I cannot see the bios or post messages (black screen).
Any way to force resolution and/or refresh?

I'm on 2.2.1 firmware..

Thanks

Z2Z-GuGu, 2025-03-11:
@delumerlino Hello, I apologize for the long delay in responding to your issue.

> NanoKVM recognize 1440x1050 resolution. I use a VGA to VGA+HDMI adapter, I can see output on the VGA screen (and for test on the HDMI), but NanoKVM remote desktop is crazy.

Currently, our NanoKVM does not support a resolution of 1440×1050. Please try changing it. Below is the list of resolutions we support and do not support:
``` c
uint16_t hdmi_res_list[][2] = {
    {1920, 1080},
    {1600, 900},
    {1440, 1080},
    {1440, 900},
    {1280, 1024},
    {1280, 960},
    {1280, 800},
    {1280, 720},
    {1152, 864},
    {1024, 768},
    {800, 600},
};

uint16_t hdmi_unsupported_res_list[][2] = {
    {1680, 1050},
    {1440, 1050},
    {1400, 1050},
    {1368, 768},
    {1366, 768},
    {720, 576},
};
```
Due to the HDMI EDID not being fully understood yet, we can only restrict the output of these resolutions in the latest 2.2.2 software.

> It is also very slow to start, because I cannot see the bios or post messages (black screen), I have `[error] [cgo.go:93] failed to read kvm image: -1 ` Any way to force resolution and/or refresh?

Regarding the startup of the NanoKVM: Some hosts or BIOS settings may turn off the USB power during shutdown. Since the NanoKVM requires some time to start up, we recommend connecting an additional 5V power supply to the auxiliary power USB port to ensure that the NanoKVM remains operational at all times.

Z2Z-GuGu, 2025-03-11:
Regarding the progress on EDID: The HDMI to MIPI CSI chip used in the NanoKVM is quite unusual. Its internal I2C controller is shared between the SOC side (for reading resolution information) and the HDMI interface, which can lead to anomalies when both sides are accessed simultaneously. In recent testing, we found that the probability of encountering such anomalies is quite high.

Actually, the current open-source vision code includes some EDID modification code for certain versions of the chip. I will take some time to improve the EDID script, but I currently have no clear understanding of EDID content and will need to continue learning more about it.

delumerlino, 2025-03-11:
> [@delumerlino](https://github.com/delumerlino) Hello, I apologize for the long delay in responding to your issue.
> 
> > NanoKVM recognize 1440x1050 resolution. I use a VGA to VGA+HDMI adapter, I can see output on the VGA screen (and for test on the HDMI), but NanoKVM remote desktop is crazy.
> 
> Currently, our NanoKVM does not support a resolution of 1440×1050. Please try changing it. Below is the list of resolutions we support and do not support:
> 
> uint16_t hdmi_res_list[][2] = {
>     {1920, 1080},
>     {1600, 900},
>     {1440, 1080},
>     {1440, 900},
>     {1280, 1024},
>     {1280, 960},
>     {1280, 800},
>     {1280, 720},
>     {1152, 864},
>     {1024, 768},
>     {800, 600},
> };
> 
> uint16_t hdmi_unsupported_res_list[][2] = {
>     {1680, 1050},
>     {1440, 1050},
>     {1400, 1050},
>     {1368, 768},
>     {1366, 768},
>     {720, 576},
> };
> Due to the HDMI EDID not being fully understood yet, we can only restrict the output of these resolutions in the latest 2.2.2 software.
> 
> > It is also very slow to start, because I cannot see the bios or post messages (black screen), I have `[error] [cgo.go:93] failed to read kvm image: -1 ` Any way to force resolution and/or refresh?
> 
> Regarding the startup of the NanoKVM: Some hosts or BIOS settings may turn off the USB power during shutdown. Since the NanoKVM requires some time to start up, we recommend connecting an additional 5V power supply to the auxiliary power USB port to ensure that the NanoKVM remains operational at all times.

Thank you lots for your reply, I have shipped back the device.
About the power connection, I have noticed nanokvm was powered down by the host, and I had already connected to an external power supply but the problem persisted.

Z2Z-GuGu, 2025-03-12:
> Thank you lots for your reply, I have shipped back the device. About the power connection, I have noticed nanokvm was powered down by the host, and I had already connected to an external power supply but the problem persisted.

Hi, did you return it through official channels? If it wasn't through Sipeed's official channels, I might not receive your NanoKVM.

jagrob9, 2025-05-01:
Hey @Z2Z-GuGu,
any chance this will get fixed? Im having the same issue with the PCIe-Version unsing a mini-DP to HDMI cable. As soon es the resolution changes after boot the video stream stops. Found no way to get it back.

git-webmaster, 2025-06-30:
Same.
Are there any plans to add 1980 x 1200? I'm getting a black screen.

Thanks.

dormancygrace, 2026-10-11:
On NanoKVM OS Tailscale is an ordinary Alpine package on a writable root filesystem; application updates (apk) keep installed packages and their state, so the Tailscale login survives an update.

NanoKVM OS is an independent community firmware for the SG2002 NanoKVM (Cube, Lite, PCIe): a full SD image on Alpine Linux, updated with apk. The current release is an alpha; PCIe is the validated platform, the Cube and Lite profiles are included but not yet validated on hardware. I maintain it, so take this as a biased pointer: https://github.com/dormancygrace/NanoKVM-OS