sipeed/NanoKVM #6: no /dev/tun device for tailscale

lidless-eye, 2024-08-12:
unlike to description in _/etc/init.d/S98tailscaled_, /dev/tun device is missing.
users have to set FLAGS="--tun=userspace-networking" to run tailscaled daemon.

either new kernel with TUN or fix in init script is required.

tested with nanokvm app 2.0.4

wj-xiao, 2024-08-12:
An image upgrade is required. 
The 1.0.0 image doesn't have Tailscale support. Please update to the latest version [1.2.0](https://github.com/sipeed/NanoKVM/releases/tag/v1.2.0).

polyzium, 2024-08-12:
See [this Chinese wiki page](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/4_firmware.html) (use Google Translate) on how to upgrade your firmware

lidless-eye, 2024-08-12:
with new image, confirmed /dev/net/tun exists. thanks!

closed 2024-08-12 by lidless-eye