sipeed/NanoKVM #7: No IP Address

sammcj, 2024-08-14:
Just recieved my NanoKVM, it boots up but never gets an IP.

I've tried updating the OS (dd the image to the usb mounted partition) but that didn't help.

There seems to be very little in the way of documentation and a lot of the source code seems to be missing?

polyzium, 2024-08-14:
Have you updated to the latest firmware? I had the exact same problem until I flashed the latest image. One thing I noticed is that the NanoKVM sends the MAC address correctly, but doesn't get the IP. If your NanoKVM has a missing MAC, it may be a hardware defect.

sammcj, 2024-08-14:
Hey @polyzium,

It's not yet 100% clear if I've correctly updated the firmware or not. I downloaded the latest available release image, extracted it, plugged the device in so that the removable drive showed up, unmounted it, used dd to write the image to the device, ejected and power cycled the KVM.

Is that the same process you followed?

Also interested to know if your device has ethernet link lights?

Thanks for the info!

polyzium, 2024-08-14:
> plugged the device in so that the removable drive showed up

The removable drive provided by NanoKVM is a data partition for ISO mounting. It does not affect your firmware in any way.
If you have the Full version, you need to unscrew the case and lift the board out of the case. On the bottom side you will find a heat sink, behind it is the SD card slot with the 32GB Kioxia SD card. Remove the heatsink (it's held by a thermal pad), pull it out, plug it into your computer (mine doesn't have one, so I used a Steam Deck) and flash the firmware to it. Once you're done, do the same steps in reverse.

Let me know if this helps.

> Also interested to know if your device has ethernet link lights?

No. I checked my router settings instead.

sammcj, 2024-08-14:
Ohhhh gosh - I did not realise that, I actually assumed this revision of the full version must have had some onboard / emmc flash as there is no port for the microSD card in the case, I just opened it up now to see the SDcard inside 🤦 

That's really helpful advice @polyzium, I'm not sure if I missed this information in the repo or on the website somewhere, but I might submit a PR to the readme to add this information.

I updated the SD and now when it boots the OLED simply states "Couldn'tget IP!"

I might have to look at how I can get a serial console onto it to debug whats going on.

polyzium, 2024-08-14:
Make sure you have DHCP enabled on your network, otherwise it won't work. Also, like I said, you probably need to check your router settings to see if the KVM is being recognized at all, either by MAC or some other means.

sammcj, 2024-08-14:
Yeah DHCP is setup and working, I've got a pretty extensive home network/server environment, I've been monitoring the DHCP server logs and it doesn't seem to get any request, I'm suspecting either the NIC is faulty or is missing it's MAC address.

Z2Z-GuGu, 2024-08-15:
> Yeah DHCP is setup and working, I've got a pretty extensive home network/server environment, I've been monitoring the DHCP server logs and it doesn't seem to get any request, I'm suspecting either the NIC is faulty or is missing it's MAC address.
Hello, is there any change in the ETH status on the OLED after connecting to the network cable? If you are using the lite version, trying to connect to the development board and run the command 'cat /sys/class/net/eth0/carrier', if the result is 1, the network cable has been plugged in, please continue to run 'udhcpc -i eth0' to get the IP address.

sammcj, 2024-08-15:
> Hello, is there any change in the ETH status on the OLED after connecting to the network cable? 

Hey, I get link lights on my switch or any computer I connect it to and they can see the link layer is up at 100mbps, but the KVM doesn't aquire an IP.

I have the full version of the KVM, does the 'reserved serial interface' or USB HID provide a serial console tty?

polyzium, 2024-08-16:
I don't think the NanoKVM exposes a tty to the two serial ports. You might have some luck setting it up via the init scripts on the firmware. I believe getty is present on the firmware.

polyzium, 2024-08-20:
I believe they have released the firmware with your fix. Maybe it's time to close this issue?

Z2Z-GuGu, 2024-08-20:
Hello everyone, we reproduce this problem, and tried to make some software modifications, please refer to the [firmware flashing](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) to re-flash the latest [v1.2.1 image](https://github.com/sipeed/NanoKVM/releases/tag/v1.2.1) to test again. We have found that a small number of power supplies or too long network cables can also cause the problem of not being able to get IP or slow network speed, if the new image still cannot get IP, please try to change the power supply of the NanoKVM or use a shorter network cable.

Z2Z-GuGu, 2024-08-20:
> I believe they have released the firmware with your fix. Maybe it's time to close this issue?

Thank you for your feedback, but I'm not sure the new image will solve everyone's problems, and we may need more feedback.

polyzium, 2024-08-20:
My KVM has been working since day one, so I doubt it would change anything for me.

sammcj, 2024-08-20:
Testing the new firmware out now, will report back.

sammcj, 2024-08-20:
Unfotunately it's exactly the same:

[image]

I wrote the SD card with:

```
sudo dd if=/Users/samm/Downloads/20240820_NanoKVM_Rev1_2_1.img conv=sync of=/dev/disk11 bs=4M
Password:
230+0 records in
230+0 records out
964689920 bytes transferred in 67.133205 secs (14369788 bytes/sec)
```

and safetly ejected before removing.

Z2Z-GuGu, 2024-08-20:
> Hey, I get link lights on my switch or any computer I connect it to and they can see the link layer is up at 100mbps, but the KVM doesn't aquire an IP.
>
> I have the full version of the KVM, does the [...]

I'm sorry to see your reply so late, the USB HID port actually has RNDIS service enabled by default, please refer to [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/updating.html#Check-via-USB-RNDIS-Network-Interface) to connect your computer.

Z2Z-GuGu, 2024-08-20:
> Unfotunately it's exactly the same:
> 
> ![IMG_4746](https://private-user-images.githubusercontent.com/862951/359433340-d8902bcd-3288-4771-aa7a-8ac1d1cd0add.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQxNDkxODAsIm5iZiI6MTcyNDE0ODg4MCwicGF0aCI6Ii84NjI5NTEvMzU5NDMzMzQwLWQ4OTAyYmNkLTMyODgtNDc3MS1hYTdhLThhYzFkMWNkMGFkZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyMFQxMDE0NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MGY3MDJiZTcyMDU5ODk1MzRiZDRiMWNmMGJmM2U0MmVjZTA2NDM4MDgyZjY2NTgzMGVhOGM2NmE3M2M2NzQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5KLWgoBjcz5PSVn4G1ZcfIr_aC5g-KQOeJqUSLEwpnE)
> 
> I wrote the SD card with:
> 
> ```
> sudo dd if=/Users/samm/Downloads/20240820_NanoKVM_Rev1_2_1.img conv=sync of=/dev/disk11 bs=4M
> Password:
> 230+0 records in
> 230+0 records out
> 964689920 bytes transferred in 67.133205 secs (14369788 bytes/sec)
> ```
> 
> and safetly ejected before removing.
> 
> My DHCP server sees no requests for leases on v4 or v6 or errors.

You can try changing to a completely isolated power source, such as a power bank, or try again with a short network cable. If you still can't get the IP, connect to the computer as described above and check the network cable connection status by 'cat /sys/class/net/eth0/carrier'

sammcj, 2024-08-20:
Update: I figured it out!

So I tried the device on every switch in my house to no avil, then I tried connecting directly to my home server and did some debugging I found that the problem was indeed at the link layer.

I dug a little deeper and then realised that the NIC on the NanoKVM will not connect to any switch or network card in my house because it's limited to 100Mbit on the chipset, not just it's bus or due to performance constraints.

This means it will never work on any switches or connected to any network cards that support 1/2.5GbE, 1/5GbE or 1/10GbE as I think all or at least most of these modern standards no longer support 100MbE at the link layer.

My whole house (including my laptops) is 1/2.5GbE and a few 1/2.5/10GbE - so nothing would work with the old 100Mbit standard.

Very sorry for wasting your time looking for a software fix, I just didn't realise this was a hardware limitation of the Sipeed hardware it's based on.

closed 2024-08-20 by sammcj