sipeed/NanoKVM #8: How is wifi configured?

nycedwardd, 2024-08-14:
I can't find any documentation on how to setup wifi. Is there a text file that needs to be created on the sd card?

polyzium, 2024-08-14:
I don't think there is wifi on the NanoKVM, no matter which version you choose. Their wiki mentions some sort of expansion card, but that's it.
@Zepan @Z2Z-GuGu Could you please elaborate more on that?

Zepan, 2024-08-15:
The Current hardware version not support wifi. FW have support wifi, it is prepare work for next pcie KVM device

nycedwardd, 2024-08-15:
I'm using a Sipeed LicheeRV Nano which does have wifi. If the FW has support for it, how can it be configured?

polyzium, 2024-08-15:
I believe the Lite version is based on the LicheeRV Nano E, which doesn't have wifi. According to the [wiki](https://wiki.sipeed.com/hardware/en/lichee/RV_Nano/1_intro.html#Version-Comparison), the LicheeRV Nano W/WE has wifi.

nycedwardd, 2024-08-15:
It looks like the tools are there but the interface is not recognized.

davidcoulson, 2024-08-18:
> The Current hardware version not support wifi. FW have support wifi, it is prepare work for next pcie KVM device

Any info you can share on the PCIe KVM? I have the BliKVM PCI v2 and always looking for other devices that do something similar! :)

Z2Z-GuGu, 2024-08-20:
> 我正在使用 Sipeed LicheeRV Nano，它有 wifi。如果 FW 支持它，如何配置它？

NanoKVM's streaming service is bundled with the SG2002 chip, [see here](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/system/introduction.html#%E5%9B%BA%E4%BB%B6%EF%BC%88%E9%95%9C%E5%83%8F%EF%BC%89%E4%B8%8E%E5%BA%94%E7%94%A8%EF%BC%88APP%EF%BC%89), if you want to use the WiFi version of NanoKVM, you can look forward to our follow-up products.Or you can try DIY a piece of KVM-A, refer to our [pin definition](https://wiki.sipeed.com/hardware/assets/NanoKVM/guide/PinDefine.png)

closed 2024-08-25 by Zepan