sipeed/NanoKVM #9: Uploading ISO files

jcastro, 2024-08-15:
Hi!

I received my NanoKVM a few days ago, and it's been working quite well. I have a few comments on improvements, but I'm sure many of those will come.

I'm preparing a YouTube video for my channel (https://www.youtube.com/@JonatanCastro) and I wanted to know how I am supposed to upload an ISO file to boot from it. When I connect the SD card, I only see the boot partition on my macOS system (same on Windows VM), and through the UI, there's no option to upload, which would be great.

Also, the Wiki page seems to not be working, so it's hard to read more about it. I know it's in beta, so no worries, I'm just trying to figure things out!

Overall great product and great work!

polyzium, 2024-08-15:
The wiki is in Chinese and could be a roadblock sometimes, so I'm going to save you from headache here.
ISO files for mounting are stored on the onboard SD card's data partition (mounted to /data). They can be uploaded using the following ways:
1. Plugging the KVM into your computer via USB (HID port). You should see a 20-something GB removable drive, that is the data partition. This is where you upload your ISO files into.
2. Connecting to the KVM via SFTP. The default credentials are root:root. Once you are in, you need to navigate to /data and upload your ISOs to there.
3. Writing to the SD card. If you have the Full version, you need to get the SD card out (I described how to do that [here](https://github.com/sipeed/NanoKVM/issues/7#issuecomment-2288903849)). Mount the 20-something Gb data partition on the SD card and you should be able to upload your ISOs that way.

After you're done you should see the ISO images immediately pop up in the ISO mounting menu of the web UI. Let me know if this helps.

jcastro, 2024-08-15:
Thanks a lot @polyzium. I'm now uploading it through SFTP! Hopefully, upload through web UI is coming soon <3

As for the Wiki, I couldn't even access the Chinese version. I'm being redirected to a 404 page, so wiki.sipeed.com/nanokvm goes to https://wiki.sipeed.com/en/404.html?jump=5&from=https://wiki.sipeed.com/nanokvm

[image]

polyzium, 2024-08-15:
> Hopefully, upload through web UI is coming soon <3

I wish I could implement such a feature, I just hope they open source the backend part of their firmware sooner

> As for the Wiki, I couldn't even access the Chinese version.

Something tells me it might be a regional block. I have a Russian IP and it opens for me without any problems. Could you try if [this page](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/1_intro.html) opens for you?

jcastro, 2024-08-15:
> Something tells me it might be a regional block. I have a Russian IP and it opens for me without any problems. Could you try if [this page](https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/1_intro.html) opens for you?

Yep! that web opens up perfectly

tltangliang, 2024-08-24:
@Z2Z-GuGu  What's the password for root user ? I'm using scp but I don't know the root password and I cann't find it anywhere.

jcastro, 2024-08-24:
> @Z2Z-GuGu What's the password for root user ? I'm using scp but I don't know the root password and I cann't find it anywhere.

root and root

wj-xiao, 2024-08-26:
Due to the network and performance constraints, uploading large image file through browser can be frustrating.

I've add tips on how to upload image in the new version. 
We may add this feature in the future but not now.

closed 2024-08-26 by jcastro