sipeed/NanoKVM #11: Any plans to add Proxmox Qdevice support?

jcastro, 2024-08-18:
Hi!
I'm currently running a Proxmox cluster with 2 nodes + a Qdevice (to get quorum so the cluster works properly). I'm running it on my current KVM device that runs PiKVM; however, I would love to use NanoKVM instead.

[As you can see in this video](https://youtu.be/TXFYTQKYlno?si=j-LmGXYz4dW8Ghkw&t=728), it's only required to have:

-  SSH daemon package
-  corosync-qnetd and corosync-qdevice packages (and their dependencies, of course)

Thanks!

davidcoulson, 2024-08-18:
I'm running modelrockettier/corosync-qnetd for my qdevice.

I don't see a build yet for RISC-V but should be easy enough - I'm certainly in favor of running docker on a device rather than  having to support a bunch of random capabilities.

jcastro, 2024-08-19:
> I'm running modelrockettier/corosync-qnetd for my qdevice.
> 
> I don't see a build yet for RISC-V but should be easy enough - I'm certainly in favor of running docker on a device rather than having to support a bunch of random capabilities.

Isn't it too heavy docker for the nanokvm?

polyzium, 2024-08-20:
Good point. The NanoKVM has about 256MB of onboard RAM and half of that is used by the KVM software. I highly doubt it would have the resources left over for Docker.

Z2Z-GuGu, 2024-08-21:
NanoKVM is built on SG2002, which is just a 900MHz single core CPU with RISC-V core (small cores can only run RTOS). Running KVM programs takes up 60-90% of CPU time, and running Docker is too difficult for SG2002 
[image]

closed 2024-08-25 by Zepan