sipeed/NanoKVM #13: NanoKVM does not boot: nothing on OLED display, PC-USB and AUX LEDs lit

mivens, 2024-08-20:
I have just received a NanoKVM Full. 
It does not boot up. 
I have both the PC-USB and AUX USB ports connected and connected it to my ethernet switch.
There is a blue AUX LED and a red LED for the PC-USB lit on the side.
The blue 5V LED is lit on the side of the NanoKVM with the OLED screen.
But there is nothing on the OLED display and the ethernet port stays down.

I have tried multiple USB chargers as well as a USB power bank. 

[image]
[image]

polyzium, 2024-08-20:
Hi, try flashing the latest firmware by following [this guide](https://github.com/sipeed/NanoKVM/issues/7#issuecomment-2288903849) and see if it works. If not, the SD card inside the KVM may be defective and you may need to replace it.

Z2Z-GuGu, 2024-08-20:
Thank you for @polyzium 's reply. Please follow his steps to re burn the TF card. You can also refer to [disassembly]( https://wiki.sipeed.com/hardware/assets/NanoKVM/guide/fix1.png),[WiKi]((https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) If the blue LED does not start flashing after burning, the NanoKVM-A board needs to be removed and powered on again (starting like the Lite version) to see if the blue LED is flashing.

mivens, 2024-08-20:
Thank you - I wasn't sure if disassembling a newly received device was OK. 

After reflashing the TF card, the problem is resolved.

closed 2024-08-20 by mivens