sipeed/NanoKVM #14: Documentation gaps

patschi, 2024-08-23:
I have discovered NanoKVM just yesterday and I'm amazed about this project. I've been reading through the documentation, GitHub, aliexpress page, etc and found some documentation gaps I'd like to note and get clarification on:

- [ ] On GitHub it mentions latency of 90～230ms. But on aliexpress it states 100 ~ 160ms. What is correct now?
- [ ] All pages state there are expansion accessories like WiFI and PoE. Where are both expansions available? Which modules?
- [ ] The aliexpress link here is broken: https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/introduction.html#Purchase-Links
- [ ] The documentation refers to "TF Card", but doesn't specify which card. I'd appreciate if "microSD" (it is, right?) can be noted somewhere for clarification.
- [ ] Is it possible to change the text/content of what is shown on the OLED display in the NanoKVM Full-version?

Again - Great project!

Z2Z-GuGu, 2024-08-26:
Thanks for your feedback
+ The data on the aliexpress page comes from the results of our early testing, and now the wiki documentation shall prevail;
+ WiFi and PoE accessories for NanoKVM cube are currently under development, but their priority is lower than NanoKVM PCIE, which is a PCIe form KVM with WiFi functionality;
+ We will fix the aliexpress link as soon as possible;
+ Yes, the TF cards described in the document are all microSD cards;
+ At present, modification is not supported, but this part will be open sourced in the future, and users can freely modify the display of OLED.

patschi, 2024-08-26:
Appreciate your fast and helpful reply, thank you!

> * WiFi and PoE accessories for NanoKVM cube are currently under development, but their priority is lower than NanoKVM PCIE, which is a PCIe form KVM with WiFi functionality;

Makes sense. Already very curious about the PCIe form-model. While I doubt seeing this - but would be cool, to have the PCIe device also present the USB VID/HUB to the host, not just consuming ground/power via PCIe pins.

I'd appreciate if there could be some note saying that these expansion modules are still under active development and not yet available - as it might disappoint users who think it is already a thing when purchasing, and prevent further people - like me - asking specifically.

> * Yes, the TF cards described in the document are all microSD cards;

I'd appreciate some clarification on this in the official documentation as well. It's possible to find this out by googling, but in official documentation it would be way more fitting.

AkechiShiro, 2024-09-08:
Default login is mentioned where ?

polyzium, 2024-09-08:
> Default login is mentioned where ?

https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/quick_start.html#Viewing-Remote-Desktop
> Open a browser and enter the obtained IP address to access the login page. **The default username and password are admin/admin.**

Z2Z-GuGu, 2024-09-11:
@patschi Hello, I modified the wiki document a long time ago, but I forgot to submit it! I didn't realize this issue until yesterday and submitted new modifications: adding descriptions of TF cards where necessary, and fixing links on AliExpress
Regarding the accessories for NanoKVM, I inquired with the leadership about the follow-up plan. We will consider providing external PoE accessories after NanoKVM is stably shipped, but the WiFi accessories have been removed. Our new NanoKVM PCIe will provide a WiFi version, and the PCIe hardware is currently being tested and adapted.

closed 2024-09-18 by Z2Z-GuGu