sipeed/NanoKVM #16: NanoKVM sometimes locks up when restarting the KVM service

polyzium, 2024-08-23:
For frontend changes and troubleshooting scenarios, it is necessary to restart the KVM service. However, in some cases, it locks up and is no longer accessible via the webUI or SSH, requiring a complete power cycle of the device.

wj-xiao, 2024-08-26:
For frontend developing, I have updated the [README.md](https://github.com/sipeed/NanoKVM/blob/main/web/README.md#development). You don't need to restart all services every time.

Locking up may be a bug. We will attempt to reproduce it.

polyzium, 2024-08-26:
I think that would be more appropriate for CONTRIBUTING than README.

wj-xiao, 2024-08-27:
> I think this would be more appropriate for CONTRIBUTING than README, similar to what the [OpenBMC webui-vue project's CONTRIBUTING.md](https://github.com/openbmc/webui-vue/blob/master/CONTRIBUTING.md) does.

This project is still under development, I will add that once it's stable, possibly next month. So feel free to update the code.

> Now that I think about it, maybe you could set up a proxy like in the OpenBMC webui-vue so that all API requests are sent to the NanoKVM?

To enable proxy, some settings need to be configured due to CORS. I will update it today.

wj-xiao, 2024-09-10:
NanoKVM will automatically reboot after updating to 2.0.7. So maybe it's not locked up.

And 2.0.7 fixed this issue. The service won't be locked up due to excessive restarts.

wj-xiao, 2024-09-12:
Closed. Please reopen if it happens again.

closed 2024-09-12 by wj-xiao