sipeed/NanoKVM #17: Tailscale support

ajorg, 2024-08-25:
I'd like to be able to connect the device to my [Tailscale](https://github.com/tailscale/tailscale) network so that I can reach it from other hosts on the VPN.

For me this is one of the more important differences when comparing to PiKVM.

polyzium, 2024-08-26:
The latest kvmapp should have Tailscale support already, please update and check again

wj-xiao, 2024-08-26:
Please refer to the [documentation](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/tailscale.html).

But there is a bug in current version. The login button in Tailscale may be no response due to a permission issue. 

If you encounter this problem, you can fix it by execute the command in NanoKVM:
```shell
chmod +x /usr/bin/tailscale
chmod +x /usr/sbin/tailscaled
```

We will fix it in this week.

ajorg, 2024-08-27:
Mine is on 2.0.1 and doesn't show a Tailscale option in the UI. I see references in the documentation (in screenshots) to a 2.0.4, but when I check for updates, it says "2.0.1 / You already have the latest version."

wj-xiao, 2024-08-27:
Can you check the image version (setting -> About NanoKVM -> image version) cause we have fix that in the latest image.

If your image version is not 1.2.0, you can update image according to [this doc](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html).

There may be two reasons for this issue：
1. Wrong time. You can check the system time by executing `date`. If the time is wrong, fix it by executing `ntpdate time.nist.gov`.
2. Wrong DNS. Execute `ping cdn.sipeed.com`. If no record, `vim /etc/resolv.conf`, and add this line: `nameserver 223.5.5.5`.

ajorg, 2024-08-27:
> Can you check the image version (setting -> About NanoKVM -> image version) cause we have fix that in the latest image.

[image]

> Wrong time. You can check the system time by executing date. If the time is wrong, fix it by executing ntpdate time.nist.gov.

Looks right, and ntpd is running.

```txt
# date
Tue Aug 27 03:04:29 UTC 2024
# ntpdate time.nist.gov
27 Aug 03:04:31 ntpdate[3130]: the NTP socket is in use, exiting
# ps | grep ntp
  388 root     /usr/sbin/ntpd -g -p /var/run/ntpd.pid
 3150 root     grep ntp
```

> Wrong DNS. Execute ping cdn.sipeed.com. If no record, vim /etc/resolv.conf, and add this line: nameserver 223.5.5.5.

Also looks right:

```txt
# ping -c 2 cdn.sipeed.com
PING onlyv4.overseacdnweb.qiniu.com.w.kunlunsl.com (163.181.66.230) 56(84) bytes of data.
64 bytes from 163.181.66.230 (163.181.66.230): icmp_seq=1 ttl=53 time=2.72 ms
64 bytes from 163.181.66.230 (163.181.66.230): icmp_seq=2 ttl=53 time=3.23 ms

--- onlyv4.overseacdnweb.qiniu.com.w.kunlunsl.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 2.723/2.975/3.227/0.252 ms
```

I found in one of the logs the CDN URL that's used and it's reporting that 2.0.1 is the latest available:

```txt
# curl https://cdn.sipeed.com/nanokvm/latest
2.0.1
```

wj-xiao, 2024-08-27:
It looks like CDN has wrong cache in your area. I get 2.0.5 from the same url.

I just refresh the file resources. Can you try to update again?

or try `curl https://cdn.sipeed.com/nanokvm/latest?random=12345`

ajorg, 2024-08-27:
Some random parameters do work:

```txt
# curl 'https://cdn.sipeed.com/nanokvm/latest?asdfasfd'
2.0.5
```

But the older one is still cached:

```txt
# curl 'https://cdn.sipeed.com/nanokvm/latest'
2.0.1
```

`Cache-Control: no-cache` doesn't help either:

```txt
# curl -H Cache-Control:\ no-cache 'https://cdn.sipeed.com/nanokvm/latest'
2.0.1
```

ajorg, 2024-08-27:
After upgrading the system firmware to 1.2.0 I was able to update the app to 2.0.1 again and Tailscale is working, so I think we can close this issue and will open some others.

wj-xiao, 2024-08-27:
Closed. 

CDN cache problem link to #23 .

ironicbadger, 2024-09-05:
```
# curl https://cdn.sipeed.com/nanokvm/latest
2.0.1
```

Still showing here as `2.0.1` and the tailscale binary chmod's were required.

ajorg, 2024-09-05:
@ironicbadger for me at least, it turned out to not matter that the "app" was 2.0.1.

The firmware comes in two parts, and there's nothing to tie them together. If you haven't manually [updated your system image](https://en.wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) to [the latest](https://github.com/sipeed/NanoKVM/releases) that could explain why you had the chmod issue and I didn't.

FWIW, I wasn't able to use the 1.2.1 pre-release system image for some reason, but the 1.2.0 worked fine for me.

If you're only concerned about still seeing `2.0.1` for the "app" firmware on the CDN, you'll want to watch #23 instead.

ironicbadger, 2024-09-05:
My webUI tells me 2.0.4 for the app, how do I know which system image I am running?

ajorg, 2024-09-05:
I didn't find a way to tell which system image. `/etc/os-release` seemed to say the same thing after upgrade, and I wasn't sure where else to look. @wj-xiao is there a way to see what release / version the installed system image is?

wj-xiao, 2024-09-06:
There are two ways to check the image version.

1. Click Settings -> About in the web page:

[image]

2. execute "cat /boot/ver" in the NanoKVM to retrieve the image release date. The corresponding image version can be found on the [release page](https://github.com/sipeed/NanoKVM/releases).

wj-xiao, 2024-09-06:
It is recommended to update to the latest image firstly by this [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html). Older image versions may contain some bugs. 

Unable to obtain the latest version will be fixed next week.

wj-xiao, 2024-09-06:
It is recommended to update to the latest image by this [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html). Older version images may contain some bugs. 

The issue of being unable to obtain the latest version will be fixed next week.

wj-xiao, 2024-09-10:
The solution can be found #42

closed 2024-09-10 by wj-xiao