sipeed/NanoKVM #22: Linux kernel on system image 1.2.0 might be almost 4 years old

ajorg, 2024-08-27:
The Linux kernel reports version 5.10.4, which was released in December 2020. That means it's missing almost 4 years of critical security fixes. The system firmware needs to be maintained to be more current in order to be safe to use.

ajorg, 2024-08-27:
I think it's worth challenging, more generally, this notion (from [System Overview](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/introduction.html)):

> Firmware updates include major system features and hardware adaptations. These need to be downloaded from GitHub and re-flashed onto the SD card, and are pushed less frequently.

Because of the number of components on the system image, and their use in other projects, security issues will be far more frequently found in components on the system image than in the application. Both need to be updated frequently, and given the difficulty of removing the SD card to flash it, and the loss of any configuration in this case, I think it would be good to use something like [RAUC](https://rauc.io/) or [SWUpdate](https://sbabic.github.io/swupdate/), both of which are supported by Buildroot.

AkechiShiro, 2024-09-09:
I believe `openssh` is also vulnerable to an unauthenticated RCE OpenSSH 9.6 even after updating (on the web interface), it is still vulnerable, if `glibc` is being used then this CVE is impacting the latest nanoKVM release I believe : 
- https://nvd.nist.gov/vuln/detail/CVE-2024-6387

lwbt, 2024-10-25:
See also https://github.com/sipeed/NanoKVM/issues/115

wj-xiao, 2024-11-08:
I'm sorry I need to close this one. Please see #115 .

closed 2024-11-08 by wj-xiao