sipeed/NanoKVM #24: SSL Lets Encrypt Cert

michaelsage, 2024-08-27:
Hi,

One thing that would be great to see is the ability to have the web interface secured by an SSL cert, I believe there is a RISC port so it should be easy enough?

I use DNS challenge on my Pi KVM and it works really well..

Thanks

Michael

polyzium, 2024-08-27:
Please refer to [this page](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/configuration.html) to set up SSL/TLS.

michaelsage, 2024-08-27:
Amazing thank you. Is there a way to install certbot on to the device to enable auto renewals?

polyzium, 2024-08-27:
The NanoKVM runs on Buildroot, and there is no package manager of any kind on the device. This is the bare minimum to get a Linux-based OS up and running. Also, the little thing has about 256MB of RAM, so I highly doubt you can run anything other than the KVM server itself.

michaelsage, 2024-08-27:
Ok, that makes sense, I've setup the renewal on another machine and it copies the cert across daily. What is the best way to restart the web interface to make sure it picks up the new cert? Or should I just schedule a reboot once a week?

patschi, 2024-08-27:
There is also a lightweight ACME implementation such as [acme.sh](https://github.com/acmesh-official/acme.sh) which could work. I don't have NanoKVM myself yet, so not sure if `bash` is installed on the box.

closed 2024-10-04 by michaelsage

AkechiShiro, 2024-10-06:
It would be great if someone documented how to use the `amce.sh` script for the nanoKVM, I guess, just mentioning this as an improvement if anyone would like to do it