sipeed/NanoKVM #25: Keyboard not detected by BIOS/EFI during POST boot sequence

ocochard, 2024-08-28:
My nanoKVM full (version 2.0.5) is able to emulate virtual keyboard once the OS (Windows or FreeBSD) is started, but while in the POST sequence it doesn’t work: I can’t enter in the BIOS menu as example.
Tested on:
- HP EliteDesk
- Beelink APU

n1LWeb, 2024-08-30:
HP Elitebooks, too can't even enter bitlocker credentials

Z2Z-GuGu, 2024-09-11:
We have recently updated version 2.0.7 and attempted to fix this issue. You can update the test on your browser

ocochard, 2024-09-17:
Upgraded to 2.0.7 and tested on Intel NUC, and confirmed it works now.

n1LWeb, 2024-09-18:
Still not working on HP Elitebook

n1LWeb, 2024-09-19:
I Checked again. If this happens the device will also not work on OS level, so this might be a different issue. If this is happening, then also the HID symbol is not lighting up. Resetting HID from the web ui is not working, too. After Replugging the USB it's working (at least on OS level, didn't check bios level again yet) Is there a way to "simulate" a complete re plugin of the USB Port? Maybe something i can test from the terminal interface? I have this issue on other devices on this machine, too, so it could be that it's not NanoKVM's fault.

ocochard, 2024-09-26:
Tested again with (image version 1.2.0, app version 2.0.9) on the same difficult HP EliteDesk: still not able to use keyboard (echap, F10, others) to enter the BIOS.
But once booted, the keyboard/mouse were working well with the OS.

Z2Z-GuGu, 2024-09-27:
@ocochard  Hi, We have added BIOS keyboard test mode in version 2.0.9. Now, you can enable BIOS mode keyboard and mouse by running 'touch/boot/BIOS && reboot '

OrpheeGT, 2024-09-27:
> @ocochard Hi, We have added BIOS keyboard test mode in version 2.0.9. Now, you can enable BIOS mode keyboard and mouse by running 'touch/boot/BIOS && reboot '

personnaly raised the following issue : 
https://github.com/sipeed/NanoKVM/issues/50

I tried your BIOS mode, it does nothing.
System boot, but keyboard does not work.

[messages.log](https://github.com/user-attachments/files/17163381/messages.log)

```
# cat /sys/kernel/config/usb_gadget/g0/functions/hid.GS0/subclass 
1
# 
```

ocochard, 2024-09-30:
> @ocochard Hi, We have added BIOS keyboard test mode in version 2.0.9. Now, you can enable BIOS mode keyboard and mouse by running 'touch/boot/BIOS && reboot '

I’ve tried that, but no change: the BIOS still doesn’t detected the keyboard (but works fine after).

AkechiShiro, 2024-09-30:
Could you give more details is your BIOS running in UEFI/UEFI with CSM/Legacy, are there any options about USB devices in your BIOS such as time waiting for external USB devices ? Any diagnostics vs fastboot option?

Let us know exact hardware model of your device + BIOS version/revision and if you're booting in UEFI or Legacy @ocochard

OrpheeGT, 2024-09-30:
HP Microserver G7 N54L

Definitively NOT UEFI.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

romainreignier, 2024-10-04:
I have the same issue on Neousys embedded computers running Insyde UEFI BIOS. Even in GRUB and resetting USB HID.
I have also tried with the `/boot/BIOS` file but there is no change.

https://github.com/user-attachments/assets/5e0f16e6-1dbd-4902-8f81-6b383eabd215

aka-charos, 2024-10-14:
I have the same issue on  Xeon Glacier Falls W Workstation

romainreignier, 2024-10-15:
This seems to be a known issue for the [PiKVM](https://github.com/pikvm/pikvm/issues?q=is%3Aissue+BIOS+keyboard) and [ArduinoHID](https://github.com/NicoHood/HID/issues?q=is%3Aissue+BIOS) folks.

Sipeed has already added a basic workaround in `/etc/init.d/S03usbdev`:

```sh
if [ -e /boot/BIOS ]
then
    echo 1 > functions/hid.GS1/subclass
fi
```

But we will need to test more options.

From what I have read, some BIOS:
- do not like the Mass Storage Device
- require a single USB descriptor
- require the USB HIB Boot protocol [see Appendix B of the specification (p. 69 of the PDF)](https://www.usb.org/sites/default/files/hid1_12.pdf)

aka-charos, 2024-10-28:
obviously the workaround that Sipeed provided does not work :(

OrpheeGT, 2024-12-05:
Hello,

I received the NanoKVM PCIe, updated to latest 2.1.2, same issue as NanoKVM Full (also updated to 2.1.2) still no keyboard available in BIOS.

ALuck0, 2024-12-05:
Also having this issue. Hope this gets fixed soon since this fundamentally breaks this device.

Qubasa, 2025-01-02:
Yep me too

SpaceAgeHero, 2025-01-03:
Also cannot send inputs to BitLocker unfortunately :-(

eljese, 2025-01-03:
Same issue on HP Prodesk 600 G6. Keyboard does not work during boot so no access to BIOS.

aksultank, 2025-01-18:
Same here, keyboard will not work while pc is booting. I bought NanoKVM specifically for this and it doesn't work.

OrpheeGT, 2025-01-20:
> I saw from PiKVM that an HP machine was unable to recognize HID devices in the BIOS. If the mass storage device could be disabled, it could be detected again. This may be due to the BIOS of DELL and HP machines not allowing mass storage devices and HID devices to coexist. Everyone can try disabling the virtual USB drive and virtual network card in Settings -> Devices and then test in the BIOS.This is indeed a situation we did not consider. We will consider purchasing an HP machine later to try to get them to work together.

Hello,

Already tried to disable everything (virtual LAN and virtual USB) but it did not fix it.

You may consider an old HP N54L for your tests, very cheap.

romainreignier, 2025-01-20:
Hi @Z2Z-GuGu thanks for the tips but I had already tried this.
For me, this is related to the USB HID Boot protocol that might not be enabled or implemented in Linux USB Gadget subsystem.

ciroiriarte, 2025-01-29:
I see the same effect on a Dell PowerEdge R740 server. With the OS up & running the keyboard works, but trying to use the keyboard during POST to select BIOS setup or Boot Manager, there's no reaction.

Image: 1.3.0
Application: 2.1.5

vonasek, 2025-01-30:
> This seems to be a known issue for the [PiKVM](https://github.com/pikvm/pikvm/issues?q=is%3Aissue+BIOS+keyboard) and > Sipeed has already added a basic workaround in `/etc/init.d/S03usbdev`:
> 
> if [ -e /boot/BIOS ]
> then
>     echo 1 > functions/hid.GS1/subclass
> fi
> 
>     * require the USB HIB Boot protocol [see Appendix B of the specification (p. 69 of the PDF)](https://www.usb.org/sites/default/files/hid1_12.pdf)

Not sure if the report descriptor used for hid.GS0 meets all requirements for boot compliant HID, but after changing report_length from 6 to 8, I can now enter bitlocker PIN during startup on my HP notebook. Just do this change in /etc/init.d/S03usbdev and try if it helps:

```
--- S03usbdev.orig_2025-01-30	2025-01-30 16:36:32.901104166 +0100
+++ S03usbdev	2025-01-30 17:42:48.756329489 +0100
@@ -50,7 +50,7 @@
         echo 1 > functions/hid.GS0/wakeup_on_write
     fi
     echo 1 > functions/hid.GS0/protocol
-    echo 6 > functions/hid.GS0/report_length
+    echo 8 > functions/hid.GS0/report_length
 #   echo -ne \\x05\\x01\\x09\\x06\\xa1\\x01\\x05\\x07\\x19\\xe0\\x29\\xe7\\x15\\x00\\x25\\x10\\x75\\x01\\x95\\x08\\x01\\x02\\x95\\x01\\x75\\x08\\x01\\x03\\x95\\x05\\x75\\x01\\x05\\x08\\x19\\x01\\x29\\x05\\x91\\x02\\x95\\x01\\x75\\x03\\x91\\x03\\x95\\x06\\x75\\x08\\x15\\x00\\x25\\x65\\x05\\x07\\x19\\x00\\x29\\x65\\x81\\x00\\xc0 > functions/hid.GS0/report_desc
     echo -ne \\x05\\x01\\x09\\x06\\xa1\\x01\\x05\\x07\\x19\\xe0\\x29\\xe7\\x15\\x00\\x25\\x01\\x75\\x01\\x95\\x08\\x81\\x02\\x95\\x01\\x75\\x08\\x81\\x03\\x95\\x05\\x75\\x01\\x05\\x08\\x19\\x01\\x29\\x05\\x91\\x02\\x95\\x01\\x75\\x03\\x91\\x03\\x95\\x06\\x75\\x08\\x15\\x00\\x25\\x65\\x05\\x07\\x19\\x00\\x29\\x65\\x81\\x00\\xc0 > functions/hid.GS0/report_desc
     ln -s functions/hid.GS0 configs/c.1

```

Btw. /boot/BIOS workaround was also needed to make it working.

vonasek, 2025-01-30:
> I'm glad you found a workaround for your case, hopefully we can build on it for a solution. Unfortunately, even both workarounds together were not enough for being able to enter the bitlocker PIN on my Dell notebook.

You might want to try a different USB port. On my notebook, not all USB ports accept the BIOS keyboard, but once the OS is booted, the keyboard works on all of them.

You could also try another experiment. Before discovering that changing report_length to 8 was sufficient for my needs, I experimented with a secondary keyboard using a report descriptor that I had previously used on an ATtiny85 device. It might be more compliant with a boot-compatible keyboard. A few years ago, I took it from this code: https://web.archive.org/web/20210817082501/http://digistump.com/board/index.php/topic,1096.5/wap2.html

This code adds another keyboard as /dev/hidg3:
```
--- S03usbdev.orig_2025-01-30	2025-01-30 16:36:32.901104166 +0100
+++ S03usbdev	2025-01-30 16:45:23.009015405 +0100
@@ -85,6 +85,24 @@
     echo -ne \\x05\\x01\\x09\\x02\\xa1\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x03\\x15\\x00\\x25\\x01\\x95\\x03\\x75\\x01\\x81\\x02\\x95\\x01\\x75\\x05\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x15\\x00\\x26\\xff\\x7f\\x35\\x00\\x46\\xff\\x7f\\x75\\x10\\x95\\x02\\x81\\x02\\x05\\x01\\x09\\x38\\x15\\x81\\x25\\x7f\\x35\\x00\\x45\\x00\\x75\\x08\\x95\\x01\\x81\\x06\\xc0\\xc0 > functions/hid.GS2/report_desc
     ln -s functions/hid.GS2 configs/c.1
 
+
+    # BIOS keyboard
+    mkdir functions/hid.GS3
+    if [ -e /boot/BIOS ]
+    then
+        echo 1 > functions/hid.GS3/subclass
+    fi
+    if [ ! -e /boot/usb.notwakeup ]
+    then
+        echo 1 > functions/hid.GS3/wakeup_on_write
+    fi
+    echo 1 > functions/hid.GS3/protocol
+    echo 8 > functions/hid.GS3/report_length
+    echo -ne "\x05\x01\x09\x06\xa1\x01\x75\x01\x95\x08\x05\x07\x19\xe0\x29\xe7\x15\x00\x25\x01\x81\x02\x95\x01\x75\x08\x81\x03\x95\x05\x75\x01\x05\x08\x19\x01\x29\x05\x91\x02\x95\x01\x75\x03\x91\x03\x95\x06\x75\x08\x15\x00\x25\x65\x05\x07\x19\x00\x29\x65\x81\x00\xc0" > functions/hid.GS3/report_desc
+    ln -s functions/hid.GS3 configs/c.1
+
+
+
         if [ -e /boot/usb.disk0 ]
         then
                 mkdir functions/mass_storage.disk0

```

This HID device is not used by NanoKVM-Server, but you can manually test whether it works during the boot process in the terminal. The following command writes "1" on the keyboard:

```
echo -ne "\x00\x00\x1e\x00\x00\x00\x00\x00" > /dev/hidg3; echo -ne "\x00\x00\x00\x00\x00\x00\x00\x00" > /dev/hidg3
```

romainreignier, 2025-01-31:
Hi @vonasek good work! Your workaround work for me on the Neousys industrial computer!
Thanks!

OrpheeGT, 2025-01-31:
I tried it to on my HP N54L, but nothing works.

starlit-rocketship, 2025-02-01:
HP Elietedesk does not work in BIOS either. Works in Linux OS just fine once it starts booting to OS.
Tried disabling Virtual Disk and Network and no luck.

Hopefully can be fixed in an update

AkechiShiro, 2025-02-04:
> I see the same effect on a Dell PowerEdge R740 server. With the OS up & running the keyboard works, but trying to use the keyboard during POST to select BIOS setup or Boot Manager, there's no reaction.
> 
> Image: 1.3.0 Application: 2.1.5

Just curious why would you use the nanoKVM on a R740 which already has an iDRAC ? Is it as a backup KVM over IP ?

ciroiriarte, 2025-02-04:
Yup, I use it for work in the field, temporary to reset the password or change IP for iDRAC.

Actually waiting for a NanoKVM-USB to replace a original NanoKVM, IP is superfluous for my use case.

joetretter, 2025-02-12:
I am observing the same issue (I can't enter my bitlocker passcode with the NanoKVM, but it works with my USB Keyboard). The keyboard works fine once the OS is up - I tried with
- Dell Latitude 5430
- Lenovo Yoga 920

mbucc9000, 2025-03-04:
I'm having the same issue with my HP bios not recognizing HID. I've tried all the fixes mentioned in the few GitHub issues that have been posted. (HP Prodesk 600 G3).

WE NEED A REAL FIX!!!

prasaanth-slx, 2025-03-10:
I'm also facing the same issue. My host device is Raspberry pi 5 where both keyboard and mouse are not working during boot. 
We are hoping for a fix asap.

prasaanth-slx, 2025-03-11:
but its not working for me for pi 5

LordKillZone, 2025-03-18:
Yeap HP PC as server and the same problem. I bought nanoKVM specially for theese situation, when i need to remotly run some iso and fix server but i cannot even enter boot options because keyboard is not working when booting with mounted iso file. So basicly nanoKVM is poaintless for me at this point.

slushieken, 2025-03-18:
Very similar or possibly same problem here with an Erying motherboard.

I was able to enter keyboard input only after the OS came up, which does detect the USB device as a SiPeed during the boot process. Before that though, no input was working for neither the BIOS or the GRUB boot menu.

Things I have tried:

- touch/boot/BIOS && reboot
- reset HID
- reboot the SiPeed, was greeted with keyboard input, then rebooted the motherboard, no BIOS or boot menu keyboard

Suggestion to testers/sufferers:

- [There is a problem w/ too long of USB cable outlined here ](https://github.com/sipeed/NanoKVM/issues/87), resolved by using a shorter USB cable of .5 meters. That may help you if the problem is inconsistent.

vonasek, 2025-03-18:
> Things I have tried:
> 
>     * touch/boot/BIOS && reboot
> 
>     * reset HID
> 
>     * reboot the SiPeed, was greeted with keyboard input, then rebooted the motherboard, no BIOS or boot menu keyboard

You should also try to change report_length as described here: https://github.com/sipeed/NanoKVM/issues/25#issuecomment-2625066273

mbucc9000, 2025-03-18:
> > Things I have tried:
> > ```
> > * touch/boot/BIOS && reboot
> > 
> > * reset HID
> > 
> > * reboot the SiPeed, was greeted with keyboard input, then rebooted the motherboard, no BIOS or boot menu keyboard
> > ```
> 
> You should also try to change report_length as described here: [#25 (comment)](https://github.com/sipeed/NanoKVM/issues/25#issuecomment-2625066273)

Turn off “Virtual Disk” in the device settings. I was able to get BIOS HID working on my HP EliteDesk 800 G3

slushieken, 2025-03-18:
> You should also try to change report_length as described here: [#25 (comment)](https://github.com/sipeed/NanoKVM/issues/25#issuecomment-2625066273)

I appreciate the suggestions vonasek, mbucc9000. I added the edits, but I can't narrow down a solution as to why it mostly doesn't gives keyboard input, other than randomly.

- Sporadically, keyboard input will work for a time (but only so far after a reboot of SiPeed).
- After a reboot of either the motherboard or the SiPeed KVM (or maybe a time period?), input stops again.
- One time I got the delete key to work during a motherboard boot, and I reached the BIOS. But as soon as I was in, keyboard input stopped again.

prasaanth-slx, 2025-03-19:
anyone using pi 5 with nanokvm? i can't get hid functionality during boot

starlit-rocketship, 2025-03-25:
Tested on HP Elitedesk G4 and it did not work for me either.

Latest 2.2.4 App Version, 1.3.0 Image

Download.
Extracted.
WinSCP > created folder (did not exist) /init.d in /etc
Moved file to /etc/init.d
chmod +x /etc/init.d/S03usbdev (in web terminal)
Reboot

No USB respone in boot up

OrpheeGT, 2025-03-25:
> WinSCP > created folder (did not exist) /init.d in /etc

If /etc/init.d/ had to be created, then you failed somewhere.

starlit-rocketship, 2025-03-25:
Nevermind, I entered the wrong IP and dropped than on another linux temrinal >_<

Let me try again!

starlit-rocketship, 2025-03-25:
Okay!! Great progress!!! :) They keyboard now works in boot (but mouse mode borken - see below)
Thanks for buying hardware and looking into this.

I can get into bios with the onscreen keyboard (ESC) & and I can use the keyboard to navigate around.

It has broken my mouse mode in the sense of Absolute mode no longer works in BIOS or more importantly, OS and I have to use Realtive Mode. It was working fine before.

Any ideas on this one?

eljese, 2025-03-25:
Tested with boot image 1.3.0 and application 2.2.4. No success with HP Prodesk 600 G6.

Z2Z-GuGu, 2025-03-25:
> It has however broken my mouse mode in the sense of Absolute mode no longer works in BIOS or more importantly my OS and I have to use Realtive Mode. It was working fine before.

I didn't include the relative mode in the testing script, but I will add the complete script to the new version (2.2.5), which is expected to be updated within a day.

Additionally, version 2.2.4 may have caused service anomalies in a few cases due to my oversight, which could be the reason @OrpheeGT encountered issues. You can try again after receiving the new version. In any case, I will keep following up on this issue. Thank you all for your feedback!

OrpheeGT, 2025-03-25:
@Z2Z-GuGu let me know if I can help in any way...
If you need a temporary SSH access to my NanoKVM or HP N54L, we can discuss by mail.

Z2Z-GuGu, 2025-03-25:
> [@Z2Z-GuGu](https://github.com/Z2Z-GuGu) let me know if I can help in any way... If you need a temporary SSH access to my NanoKVM or HP N54L, we can discuss by mail.

Thank you! I will contact you when necessary. But before that, I need to know your email address. If you're concerned about exposing your email, you can send a message to my email: z2zmoujietongxue@126.com

OrpheeGT, 2025-03-25:
> Thank you! I will contact you when necessary. But before that, I need to know your email address. If you're concerned about exposing your email, you can send a message to my email: [...]

mail sent

ocochard, 2025-03-25:
> 
> If I receive successful feedback from you, I will include this feature in the new version. Thank you, everyone!

Tested on HP EliteDesk 705 G4 with upgraded nanokvm (v1.4.0 image and v2.2.4 app) using the S03usbdev file:
- Fix the keyboard detection in BIOS;
- Mouse doesn’t work anymore in Windows in the standard absolute mode, need to use the relative mode to fix that.

OrpheeGT, 2025-03-25:
So I just updated mine to v1.4.0 image as it was stil in v1.3.0, re-updated to nanokvm 2.2.4, re-applied the fix..
But no go...

I tried to changed report_lengh to 8 too, but still doesn't work on my HP N54L.

diamondiron, 2025-03-25:
I can report success with HP BIOS. I'm running 1.3.0 application version and edited the report_length from 6 to 8 in the /etc/init.d/S03usbdev and then rebooted the KVM, appears to be working but don't think its surviving reboots.

Z2Z-GuGu, 2025-03-26:
Hey, guys! We just updated to the latest version 2.2.5, and you should now receive the version update. In this latest version, we added a "HID Only" mode, which enhances HID compatibility and fixes the issue with absolute mouse functionality mentioned in the previous script (of course, some BIOS do not support absolute position mice and can only enable relative mice). Additionally, the previous versions may have had unstable HID performance on some host systems (like the Mac mini M4), and using the "HID Only" mode can fix that as well.

However, the "HID Only" mode will disable the virtual USB drive and virtual network card. If you do not need these additional features, you can prioritize using "HID Only."

You can select the "HID Only" mode under the mouse icon. To ensure a successful switch, the system will restart during the transition. After about 16-23 seconds, you can refresh the webpage automatically or manually, and you will successfully switch to "HID Only" mode. 

I tested this on my HP EliteDesk 800 G3, and this mode can reboot smoothly into the BIOS, allowing control with the keyboard and relative mouse.

Z2Z-GuGu, 2025-03-26:
> So I just updated mine to v1.4.0 image as it was stil in v1.3.0, re-updated to nanokvm 2.2.4, re-applied the fix.. But no go...
> 
> I tried to changed report_lengh to 8 too, but still doesn't work on my HP N54L.

Please update to version 2.2.5 and try using the 'HID Only' mode again. This may resolve your issue

OrpheeGT, 2025-03-26:
> Please update to version 2.2.5 and try using the 'HID Only' mode again. This may resolve your issue

This is a no go for me...

I don't understand why...

[image]

no reaction at all, bios, grub, nothing...

Z2Z-GuGu, 2025-03-26:
> This is a no go for me...

I’m also curious—can you access and operate the BIOS using a regular keyboard?

OrpheeGT, 2025-03-26:
> I’m also curious—can you access and operate the BIOS using a regular keyboard?

Yes, I have an old mac keyboard, and it works fine to access bios or grub.

OrpheeGT, 2025-03-26:
I sent you an email yesterday, did you receive it, I also have Telegram if needed.

OrpheeGT, 2025-03-26:
HID-mode enabled : 

[image]

I must say the ssh connection to nanokvm is quite unstable... most of the time I have to reboot it to be able to re-login with SSH...

[14 more replies]