sipeed/NanoKVM #26: Ethernet link doesn't work in some conditions

gamelaster, 2024-08-29:
When I connect target PC HDMI cable into NanoKVM, as soon as the outer shell of USB or HDMI  touches and connects the ground of the PC to the KVM, Ethernet stops entirely. Until that, connection works perfectly. When I connect same Ethernet cable to the target PC itself, Ethernet on PC works.

I looked at the schematics, and I'm not experienced in ethernet but have done other design work for digital PCB design; and the layout of the interface and isolation doesn't match what I would expect to see. The design shown doesn't appear to provide the required isolation or match any design I've seen used with a PHY:
[image]

Could you please confirm if the schematic is correct and matches the hardware? Could you suggest what is going on here and how I can have the unit operate correctly? Thanks

Skylancer187, 2024-09-01:
I'm having similar issues and I thought it was a software bug... Reloaded the software and I can see that once the HDMI/USB is connected to the Host Computer, the ETH on the displays keeps blinking as if it's connecting and disconnecting.

I ordered 5 units and all 5 exhibited the same behavior. I'm also not an engineer, but it makes sense what I see after testing all 5 units and connecting the HDMI/USB to a host computer with the Ethernet plugged in. Wifi appears to be ok, however, if I use Ethernet, the connection is spotty, and unable to use the KVM properly.

The switch that the NanoKVM Pro is plugged into also shows link and unlinked about every 2-3 seconds. Again, only when the HDMI and Ethernet are connected to the NanoKVM Pro and the HDMI is plugged into a host computer. I have to unplug the KVM to reset it again. I'll have to see what happens if I use a dedicated USB wall wart connected in the AUX to see if that fixes a grounding issue. But so far, just a USB in the HID and Ethernet connected, it starts to have network issues within 30 seconds.

I hope it's not a hardware bug, that seems like a massive oversight in testing.

Skylancer187, 2024-09-01:
Nope, the same behavior with a dedicated power supply on the AUX causes the ETH to not work properly.

Connects to my network and disconnects ever 2-3 seconds with just the ETH and AUX (to a 2.4A 5v PSU).

Are there any bog/jump wires that I can test on one of these units to create better isolation? Willing to test something on one of the units.

gamelaster, 2024-09-01:
@Skylancer187 for me, it helped to connect Serial header GND on NanoKVM, to the protective earth on power outlet. That finally let me use NanoKVM.

Z2Z-GuGu, 2024-09-06:
@Skylancer187 Do you have any new progress? It may be possible to try using other power sources separately to power the AUX interface, such as a power bank (without connecting HDMI and HID at the same time to see if it can stably obtain IP). In addition, I have released an [image](https://github.com/sipeed/NanoKVM/releases/tag/v1.2.1) to address this issue, which may be fixed in software. For detailed procedures, please refer to [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)

Z2Z-GuGu, 2024-09-06:
> I looked at the schematics, and I'm not experienced in ethernet but have done other design work for digital PCB design; and the layout of the interface and isolation doesn't match what I would [...]

Hi, This is a circuit called capacitive isolation, which is different from inductive isolation. The circuit reference is [here](https://atta.szlcsc.com/upload/public/pdf/source/20220512/F2C4FA6759918E4E15104FDF33BE5815.pdf). The advantage is that it has a small size. In order to test the stability of this circuit, we tested it with all possible equipment of the company and found that it can work normally. However, there have been problems in the usage environment of a small number of users, and we are collecting the problems and trying to fix them through software updates.

gamelaster, 2024-09-07:
Hi @Z2Z-GuGu , I appreciate your answer.

Sadly, I am not quite sure this is possible to fix with software updates , as this seems to be hardware issue.

I figured out where is problem, and that the power supply of target computer I use, have "virtual grounding", which is not connected with common ground from power grid, and that causes problems.

Z2Z-GuGu, 2024-09-07:
@gamelaster The use of modified software can improve the sensitivity of network cable detection, and we have indeed received feedback from some users that it can be detected by V1.2.1 firmware while not being detected by V1.2.0. But this is not fundamentally solving the problem, and I cannot confirm whether such modifications can solve everyone's problem. We are currently designing a new KVM called NanoKVM PCIe, and due to sufficient PCB space, the circuit here has been modified back to traditional inductive isolation.

Z2Z-GuGu, 2024-09-18:
I recently tested the traditional inductive isolation scheme on our latest NanoKVM PCIe, and it has been proven to be effective. We tested it for a long time in different environments and even PoE, and found that it can work stably. You can follow our new product on the wiki and download site, and we will soon update the introduction and materials of the new product.

closed 2024-09-18 by Z2Z-GuGu

shalupov, 2024-09-18:
@Z2Z-GuGu Will it be changed in newer editions of nanokvm full or it's considered a finished product?

Z2Z-GuGu, 2024-09-19:
The new product called NanoKVM PCIe will completely solve this problem, but there are currently no plans to modify NanoKVM Full, and accessories may be added in the future to solve it
[image]

heppu, 2024-09-30:
The pci version looks awesome! I've been looking for poe kvm and this looks perfect. Any idea when this will be available to buy?

tigattack, 2024-11-11:
I'm experiencing this issue. This feels like quite a significant problem with the device, rendering it unusable without very inconvenient and frankly quite janky workarounds, such as the suggestion of connecting the serial GND to the aux. PSU GND... really?! I'm glad if that resolves the issue for some, but that's entirely unfeasible for me.

I know it wasn't a Sipeed employee who made this suggestion, and I'm glad they shared the suggestion, however it feels crazy that this is the only real, physical "solution" so far.

I'd imagine I'm not alone when I say I'd appreciate a little more than a "this might help sometimes" software fix from Sipeed (which, FWIW, had no effect on the issue for me, so this device happens to be completely unusable for my intended use case).

As an aside, when running the v1.2.1 image, both the display and the web interface instruct that an update should be performed. I felt this worth noting.

avg-I, 2025-01-07:
What helped me is using an ethernet cable with shielded connectors, the cable is probably shielded as well (otherwise those connectors wouldn't make sense).
By shielded connectors, I mean metal ones unlike the typical plastic ones.
I guess that maybe that connects the ground.

gamelaster, 2025-01-07:
@avg-I for me even that did not worked. I guess for you it works because the other (not-KVM) side is properly grounded, and it connects the grounds through shielding, so it solves the issue.

codingJWilliams, 2025-01-12:
I am also experiencing this, this makes the product almost unusable. @Z2Z-GuGu Will we see a proper fix for this? Or a hardware recall?

gxcreator, 2025-01-15:
Same here. Ethernet almost not working even with 10M half-duplex

gxcreator, 2025-01-16:
After another reboot board just heated to degree it starts smelling burnt plasric and now it's not booting

CRCinAU, 2025-01-24:
This still seems to be an issue... I'm having problems with 2 of 3 of the full units randomly not getting ethernet link when they boot up.

It seems to matter what order I connect cables, and if I reboot the KVM at all, it's a 50/50 as to if the network will start again.

codingJWilliams, 2025-01-24:
What I have found is that if you use an unmanaged switch with a DC barrel jack between your main switch and your NanoKVM it seems to work reliably. I assume this is because the unmanaged switch uses a different ground or something like that, either way that has got it working properly for me

CRCinAU, 2025-01-24:
My switch does use a DC barrel jack - but its a PoE switch. It straight up doesn't work on my 2.5Gbit switch...

Not having reliable networking is kinda fatal on a network based KVM....

CRCinAU, 2025-01-24:
> [@shalupov](https://github.com/shalupov) The new product called NanoKVM PCIe will completely solve this problem, but there are currently no plans to modify NanoKVM Full, and accessories may be added in the future to solve it !

So given that the NanoKVM Full doesn't work correctly, what do you suggest people do who purchased these devices?

Surely the solution isn't "buy the PCIe version as well"?

CRCinAU, 2025-01-24:
I notice that the FAQ now says:

> 1.3. Unable to Obtain IP Address[#](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#Unable-to-Obtain-IP-Address)
> ... 
> The NanoKVM Cube (including NanoKVM Full and NanoKVM Lite) may be unable to obtain an IP address when connecting to certain power sources/HDMI. Please confirm the following:
> Disconnect all interfaces and power the device using a power bank, then connect the network cable to see if an IP can be obtained. If an IP can be obtained, reconnect HDMI/computer USB to check if the IP exists.
> If the IP exists only when powered by the power bank, but disappears after connecting HDMI/computer USB, this indicates the issue. Please contact customer service to purchase an isolator to resolve it.

Anyone have any idea what this 'isolator' is?

dotX12, 2025-02-01:
> Same here. Ethernet almost not working even with 10M half-duplex

see https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317

dotX12, 2025-02-01:
The device does try to work better (but understandably doesn't) on 10mbit compared to 100 mbit, as there is less interference on 10 mbit, but it is still impossible to work with. buy a cable with insulation and a metal plug. 
https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317

Z2Z-GuGu, 2025-02-20:
Perhaps everyone can pay attention to the update of F&Q in the [wiki](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#Unable-to-Obtain-IP-Address) if it is confirmed that this is the case, you can ask customer service to purchase an isolator to solve it

CRCinAU, 2025-02-20:
As mentioned, I have a PoE power adapter on the way to test if this improves being able to get link.

Won't know if this helps until I have it in-hand.

Z2Z-GuGu, 2025-02-20:
> Won't know if this helps until I have it in-hand.

In our tests, when the power/HDMI was connected to a power source with significant noise, using a separate Ethernet cable resulted in an inability to obtain an IP address. However, by connecting a PoE isolator in this manner, the NanoKVM was able to quickly acquire an IP address, and the communication speed was normal.
[image]

CRCinAU, 2025-02-20:
Yep - this is the same device that the support folks sent me a picture of that I assume is still in transit.

If that solves the problem, they have said they'll send me another 4 - one for each Full unit I bought.

Z2Z-GuGu, 2025-02-20:
It can be left as is, because in the official version of NanoKVM-Full, the power supplies of these two sets are internally isolated.

CRCinAU, 2025-03-07:
I received the PoE adapter, and I can confirm that with this adapter in place, the nano KVM works as it should do.

@Z2Z-GuGu - Given I purchased 5 x NanoKVM Full devices, I asked for another 4 of these adapters to be sent so I could use the other 4 devices I have, but was told I would have to purchase them.

As far as I'm concerned, this should be a warranty matter - as without the POE adapter, the devices just straight up don't work.

How will these be handled? As I feel strongly this should be covered under the unit warranty if the adapter is a required component...

gamelaster, 2025-03-09:
If I bought NanoKVM from Sipeed website, where I can request the PoE adapter? Thanks

stefanskotte, 2025-03-25:
Yeah, its gonna be a PiKVM for me, this is too hacky.

joshuataylor, 2025-07-05:
So this is probably why 2/6 of my devices have "randomly" died, glad it didn't take anything else out...

Biliskn3r, 2025-11-26:
NIC on mine is dead. Worked for 6months. then one day just dead. LCD screen works though. lol

Mifody-07, 2026-10-23:
# Disclaimer: I'm not affiliated with Sipeed, just a user who ran into the same problem and decided to figure it out.

## Problem
NanoKVM Cube Full. Ethernet never comes up while HDMI or USB is connected to the target PC. Unplug them and the network is up in seconds. Cables, switches, PSU and firmware don't matter.

## Cause
Small power adapters with a 2-pin plug (no ground pin) leave the device ground at about 100 V AC relative to earth (the ground pin of a wall outlet). The current is tiny and harmless, it's the same slight tingle you feel touching a metal-body laptop on its charger.

Both my NanoKVM and my switch use such adapters, so both "float" at ~100 V. While they float together, there is no difference between them and the network works.

When you plug HDMI or USB into a PC with a grounded plug, NanoKVM is pulled to 0 V. Now the switch is the only device left floating at ~100 V, and its only connection to NanoKVM is the Ethernet cable. So this voltage goes through the cable into the NanoKVM Ethernet chip, the chip can't see the data signal anymore, and the link drops.

Every normal Ethernet port (PCs, routers, switches) has a small transformer (Ethernet magnetics) that passes the data signal but blocks voltage between the grounds of connected devices. NanoKVM doesn't have one. This is a serious design mistake by Sipeed engineers: NanoKVM can't handle a normal home setup with mixed grounded and ungrounded devices. And a KVM is useless this way: its whole purpose is to have HDMI, USB and network connected at the same time, but here you have to choose between seeing the screen and having network access. On top of that, if this voltage keeps hitting the Ethernet chip, it can eventually break it, which is exactly what some people in this thread report.

The fix is to connect NanoKVM ground and switch ground, so the voltage goes through a wire instead of the Ethernet chip.

## Fix at home (two wires, nothing to buy)
I used two solid cores from an old UTP patch cord.
1. Power everything off.
2. NanoKVM: open the case and separate the screen board from the main board. Take one core from a UTP cable and strip 2-3 mm. Use a solid core: a stranded one may also work, but it's more hassle and a stray strand can easily short neighboring pins. Insert it into the GND hole of the header (see the official pinout), then push the screen board back on: its pin goes into the same hole and clamps the wire. No soldering needed, a UTP core is thin enough to fit next to the pin. Make sure no bare wire sticks out of the hole and touches anything else. Route the wire out through the gap near the LAN port.

[image]

[image]

3. Switch: this depends on the model. If your switch has a metal case or a ground screw, just use that. Mine (MikroTik CSS106) has a plastic case, so I opened it, found a large ground pad with a multimeter (continuity to the metal shell of an Ethernet jack) and soldered the second core to it.

[image]

4. Screw both wires to the metal case of a device with a grounded plug (I used my NAS). If you don't have one, just connect the two wires together, that already removes the difference.
(photo will be added later)

Result: network is stable with HDMI and USB connected.

Why separate ground wires are better than a shielded cable: leakage current goes straight to earth through its own wires instead of passing through NanoKVM and the HDMI/USB ports of the target PC.

## Without opening anything
A shielded FTP/STP patch cord can connect the grounds through the cable shield, but only if both are true:
- the cable really has a shield (foil or braid) inside, connected to the metal shells of both plugs. Some cheap cables have metal plugs but no shield, or the shield isn't connected to the plugs;
- the Ethernet jacks on both NanoKVM and the switch have metal shells connected to the device ground. A metal-looking jack is not always connected to anything.

Check it with a multimeter.

Keep in mind that in this case leakage current still flows through the HDMI/USB ports of the target PC.

A transformer Ethernet isolator also works (cheap 100 Mbit ones are enough for NanoKVM): data passes through it via a magnetic field, with no direct wire connection, so the voltage between the grounds can't get through. It costs more than my two wires, which cost me nothing, but it's more convenient: no need to open anything.

Never put wires into a mains outlet.
                                                          
Sipeed, please document this and add proper Ethernet isolation in future revisions.

joshuataylor, 2026-10-23:
Can you let us know when you burn your house down pls 😂.