sipeed/NanoKVM #27: Keyboard layout

OrpheeGT, 2024-08-29:
Hello

It seems there is only QWERTY supported ?

I'm on azerty keyboard, and typing with it inside nanokvm screen result as qwerty.

kind of tricky when trying to unlock computer session with password ******

Can it be addressed ?

Thank you

Z2Z-GuGu, 2024-09-05:
Hi, we have tested and found that the keyboard layout settings in the system will affect your input characters. Please try modifying the keyboard layout in the settings of the controlled host.

OrpheeGT, 2024-09-05:
Hi, I'm not sure to understand.

My controlled laptop is running with azerty keyboard under windows.

I'm running NanoKVM virtual screen inside a Linux Mint also with azerty keyboard layout.

But typing from Linux virtual screen from web browser seems to be in QWERTY, if I raise the virtual keyboard popup from web browser, it is also in QWERTY

OrpheeGT, 2024-09-05:
Update, actually you are right.

I made a mistake, but it seems some keys are bad identified.

There is and @ in my password.

So I usually type on my AZERTY Keyboard : Atl-GR (right Alt) + 0 (à / @)

But it seems AltGR is not recognized.
I have to use : Left CTRL + Left Alt + 0 (à / @)
[image]

[image]

wj-xiao, 2024-09-30:
fixed in 2.0.9

closed 2024-09-30 by wj-xiao