sipeed/NanoKVM #28: Static MAC Address

cedartinkerer, 2024-08-29:
I know you can set a static IP but I'd like to be able to set a static MAC address.  Or at least I assume so. Every reboot it grabs a new IP. And if I am moving between networks being able to 'lock in' a MAC address so the DHCP server would keep handing out the same IP would be really helpful. Ideally it could be in the GUI like a check box 'Lock MAC Address' or something.

cedartinkerer, 2024-08-31:
I believe I fixed it for myself by using the command:

echo "pre-up ifconfig eth0 hw ether whatever:mac:address" >> /etc/network/interfaces

It would be nice if not in the UI to at least add this in the technical documentation next to the static IP option

cedartinkerer, 2024-09-09:
And it looks like I spoke to soon. That actually did not resolve the issue. It seemed to keep it for a reboot or two but then it goes back to cycling.

cedartinkerer, 2024-10-04:
@Jakdaw  - That seems to have held better. Thanks for that pointer. I ended up putting the mac in /boot/ethmac and just creating /boot/eth.mac but I agree it looks like an error.

Interestingly on one of my networks even though the mac was the same I was getting different IPs.  Once I hard coded a reservation it worked. I need to try another network to see if it is a network issue or NanoKVM oddity.  I'm suspecting network.

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The boot identity script accepts a persistent Ethernet MAC override in /boot/eth.mac and now reads the same filename it checks. It validates the address before changing the interface. Without an override, the address is derived consistently from the chip UID.

This is a file-based setting, not a GUI lock checkbox. A stable MAC alone does not guarantee the same DHCP address on different networks.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/firmware/buildroot/board/enhanced/init.d/S10uuid#L50).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!