sipeed/NanoKVM #29: WireGuard network interfaces?

razuritta, 2024-08-30:
Hello, could you please add wireguard kmod support for it?
Using `modprobe wireguard` or adding wireguard device `ip link add wg0 type wireguard` both does not work.
wg command seems present though.

it would be pretty useful when using wireguard to connect to the management network segment.

Thanks. :)

razuritta, 2024-08-31:
> The wireguard kmod is part of the kernel since 5.6. NanoKVM has 5.10.4. Maybe you meant `ip link add dev wg0 type wireguard` ? https://www.wireguard.com/quickstart/

Oh, sorry,
I mean, using `ip link add` command does not create a wireguard interface.
It produces the following output.
```
# ip link add wg0 type wireguard
ip: RTNETLINK answers: Not supported
```

razuritta, 2024-08-31:
I guess it is not possible to create any network interfaces?
```
# ip
BusyBox v1.36.1 () multi-call binary.

Usage: ip [OPTIONS] address|route|link|tunnel|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
        [promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
        [master IFACE | nomaster] [netns PID]
ip tunnel add|change|del|show [NAME]
        [mode ipip|gre|sit] [remote ADDR] [local ADDR] [ttl TTL]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
```
and it does not show anything like `ip link add`.
I cannot create gre/vxlan tunnel network interfaces as well as linux bridge on it.

Wondering whether adding complete version of iproute2 instead of that in the busybox solves this problem.

razuritta, 2024-09-01:
> ip link add **dev** wg0 type wireguard

it has following output:
```
# ip link add dev wg0 type wireguard
ip: RTNETLINK answers: Not supported
```

wj-xiao, 2024-09-06:
Could you please confirm the image version? Cause WireGuard is available from the version v1.1.0. 

Or you can update to the latest [image](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) and try again.

razuritta, 2024-09-06:
> Could you please confirm the image version? Cause WireGuard is available from the version v1.1.0.
> 
> Or you can update to the latest [image](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) and try again.

I see, it has a version 2.0.6 for application, and 1.0.0 for image version.
Is it possible to upgrade image version via web or by placing image file to some specific place and then reboot?
Or is it a must to upgrade image version by pulling out tf card and flash it directly? :)

wj-xiao, 2024-09-06:
Unfortunately, the TF card must be pulled out to upgrade the image. 
To avoid frequent updates, a stable image version will be released in the future.

razuritta, 2024-09-07:
> Unfortunately, the TF card must be pulled out to upgrade the image. To avoid frequent updates, a stable image version will be released in the future.

I'm able to add wireguard interface manually on image version 1.2.0.
Thank you for your help.

closed 2024-09-07 by razuritta