sipeed/NanoKVM #31: Feature request: VNC

hadicharara, 2024-08-31:
Hello,

Would it be possible to add VNC support to the NanoKVM?
If this isn't a priority, would it be possible to release more documentation on how one would implement the daemon, as I think the docs are lacking.

Thank you.

polyzium, 2024-09-13:
Theoretically, you could create a proxy daemon that takes the MJPEG stream and websockets for the inputs provided by the NanoKVM and forwards them to a VNC server. Unfortunately, we don't have source code for the backend yet (it will be released when the repository reaches 2k stars), so this is the only workaround I could think of.

hadicharara, 2024-09-13:
Yes I'm aware we need the backend source code. Since they won't release it just yet I figured ai might ask them to implement it maybe.

That gradual open-sourcing of components is so weird. They make it seem like open source is a privilege. I get they want to increased visibility from starring the repo, but it might also get them bad rep.

Zepan, 2024-09-18:
> Yes I'm aware we need the backend source code. Since they won't release it just yet I figured I might ask them to implement it maybe.
> 
> That gradual open-sourcing of components is so weird. They make it seem like open source is a privilege. I get they want to increased visibility from starring the repo, but it might also get them bad rep.

In fact we will opensource after ship out all preorders.  Just don't want there have many clones before we ship out..

hadicharara, 2024-09-18:
Okay great then! Looking forward to receiving my pre-orders and contributing.