sipeed/NanoKVM #33: Display cover to tight for buttons and have 2-3 display protection foils attached

Stimmenhotel, 2024-09-02:
I have just received my preorder.

First thing I saw was the display cover protection foil is still attached, so I unscrew the display cover and removed one from the front and one from the back. I am not sure, but I think this sould be removed on assembly (for release version).
But also not a big thing.

There was a third foil, anotherone in the inside. It is a frosted foil, so I think this is intentional attached to the inside? Lightspreading and make it less transparent?

**Now the main problem:**
After assembly I discovered the Power and Reset buttons were not working properly.
I loosen the display cover screws just a bit and it was working again.
So I think the tolerances are a bit to tight and the buttons could use some more space to the display cover.

Zepan, 2024-09-02:
Hi, thank you for your feedback!
the beta version is what you descript, and in release verison (ship after 9.10) will remove the cover which inside the case.
and release version also improve the buttons:
[image]

closed 2024-09-02 by Stimmenhotel

Zepan, 2024-09-02:
right, BOOT is used when system damaged.  early case version need unscrew to get TF card to write.
It is printed via SLA