sipeed/NanoKVM #34: NanoKVM Lite Power Button Trigger

beatles1, 2024-09-02:
To be able to trigger the power button on the host with the NanoKVM Lite do we need any additional hardware. I can see pin A23 is for this but would that need to trigger a relay/transistor?

Additionally would it be possible to apply 5v somewhere as a secondary power supply or again does that need more hardware?

Thanks in advance.

Z2Z-GuGu, 2024-09-04:
Hi, it need to use a solid state relay to control the computer power, and can't use the pins to directly connect to the power button, and we have open-sourced the circuit design of KVM Full [here](https://cn.dl.sipeed.com/shareURL/KVM/nanoKVM/HDK/02_Schematic) for reference.
In addition, 5V can be used to supply external power (which will simultaneously increase the burden on the power adapter), but the VBUS pin is preferred, and using VSYS may damage your NanoKVM.

beatles1, 2024-09-04:
Oh thank you, that's really useful. Is there a reason you need a solid state relay rather than just an optocoupler? Does the power button actually carry much current?

Z2Z-GuGu, 2024-09-04:
Optocouplers are theoretically feasible, but we were forced to use solid-state relays due to a hardware design mistake a long time ago, and later did not have enough time to validate the new schematic.

beatles1, 2024-09-04:
Thank you, optocouplers just seem a lot cheaper to get only a few of. I'll have a play and see if I can get it to work.

For the 5V quesiton, am I understanding correctly that we can connect a 5V power supply to the VBUS and ground pins and that will power the NanoKVM?

Z2Z-GuGu, 2024-09-04:
> For the 5V quesiton, am I understanding correctly that we can connect a 5V power supply to the VBUS and ground pins and that will power the NanoKVM?

Yeah, you're right, NanoKVM does not require too much power (approximately 1W), ensuring a power supply in the range of 4.5-5.5V is ok.

closed 2024-09-04 by beatles1