sipeed/NanoKVM #35: Video Output on 2.0.6

cedartinkerer, 2024-09-03:
Hi,
I upgraded the app to 2.0.6 and now HDMI display on the web browser doesn't work.  I have the full version with image 1.20.  Previously it worked. I've tried refreshing the web page, log in back out and back in, and rebooting. Nothing shows.  

The LCD screen shows HDMI is connected and the frame rate changes as well  but nothing appears but a small computer icon with an x on the webpage.

CrazyCoder, 2024-09-04:
How did you get the 2.0.6 update? Mine says 2.0.4 is the latest.

cedartinkerer, 2024-09-04:
Nothing out of the ordinary.  My full told me 2.0.6 was available on the GUI and so I hit update. ::Shrugs::

CrazyCoder, 2024-09-04:
I tried a clean flash and it updates only to 2.0.4. Probably some region limitation or A/B testing.

polyzium, 2024-09-04:
I've seen reports that Sipeed's CDN caches the results and that's why not everyone may be getting the latest version. Mine is 2.0.6.

cedartinkerer, 2024-09-09:
Oddly, I reflashed the microSD with 1.2.1 and upgraded to 2.0.4 and it was working again. Then later in that day without upgrading to 2.0.6 the HDMI stopped working in the same way.

wj-xiao, 2024-09-10:
For cdn cache, please refer to #42

wj-xiao, 2024-09-10:
@cedartinkerer If the X icon appears, it means the streaming service has stopped.
Please update to version 2.0.7, which fixes some HDMI issues. But not sure if it can solve your problem.

cedartinkerer, 2024-09-11:
That is really helpful to know that the icon means the streaming server had stopped! 

It appears 2.0.7 has made it more stable and working now for 24 hours.  I'll consider it closed.