sipeed/NanoKVM #36: Implement USB Audio Emulation (Microphone and Headset) Using Linux USB Gadget on NanoKVM

thimico, 2024-09-03:
**Acceptance Criteria for USB Audio Emulation on NanoKVM**

As a final client, I want my NanoKVM device to emulate a USB microphone and headset over a USB connection, so that I can have two-way audio communication (both talking and listening) with a connected laptop or host device without needing to install any additional software on the host.

### **Acceptance Criteria:**

1. **USB Audio Emulation:**
   - When I connect the NanoKVM to my laptop or host device via USB, the NanoKVM should automatically be recognized as a standard USB audio device.
   - The laptop should see the NanoKVM as both a microphone (for audio input) and a headset (for audio output).

2. **Plug-and-Play Functionality:**
   - The USB audio emulation should work immediately upon connecting the NanoKVM to the laptop, with no additional drivers or software installation required on the host device.

3. **Two-Way Audio Communication:**
   - I should be able to use the NanoKVM as a microphone to capture my voice or other audio and transmit it to the connected laptop.
   - I should be able to listen to audio from the connected laptop through the NanoKVM, as if it were a standard headset or speaker.

4. **Compatibility:**
   - The USB audio emulation should work seamlessly with various operating systems, including Windows, macOS, and Linux.
   - The functionality should be compatible with common applications like video conferencing tools (Zoom, Microsoft Teams, Skype) and audio recording software.

5. **Reliability and Quality:**
   - The audio quality should be clear and without noticeable lag, allowing for real-time communication.
   - The connection should be stable and should not drop during use.

6. **Ease of Use:**
   - I want to be able to easily enable or disable the USB audio functionality from the NanoKVM's control interface without needing to use complex commands or configurations.
   - There should be clear indicators or feedback showing whether the USB audio emulation is active or not.

7. **No Impact on Other NanoKVM Functions:**
   - Enabling USB audio emulation should not interfere with other functionalities of the NanoKVM, such as video capture or keyboard/mouse input.

### **Completion Criteria:**

- The NanoKVM is recognized as a USB audio device by the host when connected.
- Two-way audio communication (microphone input and headset output) is successfully tested using standard audio applications on the host device.
- The feature works across different operating systems and applications as expected.
- I can enable/disable the audio feature easily from the NanoKVM control interface.

**Note:** If additional support or guidance is needed to implement this feature, please provide any source code or detailed instructions that can assist in completing this task effectively.

polyzium, 2024-09-04:
I don't think audio support is the goal of NanoKVM. Besides, even PiKVM does not support microphone emulation (at least not yet). I think it is better to couple this with a remote desktop service on the OS side than to have the KVM emulate it.

Z2Z-GuGu, 2024-09-18:
@thimico Hi, This is theoretically achievable, but we have some high priority issues recently. After these issues are resolved, I will test the functionality of adding USB audio.

corrpel, 2024-12-19:
This is the only thing preventing me to use it via internet. Because of this, i use it locally with a bluetooth headset connected to the main laptop. It would be nice to be free from main device...

hardingt, 2025-02-20:
> [@thimico](https://github.com/thimico) Hi, This is theoretically achievable, but we have some high priority issues recently. After these issues are resolved, I will test the functionality of adding USB audio.

This would be fantastic, this or pulling the stereo stream out of the HDMI, even worthy of a newer beefier product IMO

CaFeXYZ, 2025-03-19:
> [@thimico](https://github.com/thimico) Hi, This is theoretically achievable, but we have some high priority issues recently. After these issues are resolved, I will test the functionality of adding USB audio.

In the meantime can we simply utilize the built-in mic for basic audio streaming? (maybe via ssh) I assume that is much easier to archive than building a USB speaker emulator? Also apparently PiKVM supports 2 way audio now.

USB音频可能比较费时间，能不能直接先启用自带的麦克风录音？根据下面的网页nanokvm显然是可以在ssh开启录音的，encode成mp3并把data传回ssh client（而非保存在本地）也不困难。但我还没弄明白要怎么播放出来。

https://telefoncek.si/2025/02/2025-02-10-hidden-microphone-on-nanokvm/

https://docs.pikvm.org/audio/

CaFeXYZ, 2025-03-20:
I got audio working finally! It's actually just using the built-in mic to record audio from the machine, so the quality isn't quite there, but it's good enough as a temporary solution.

From Windows Command Prompt (PowerShell will not work. You might be able to get a modified command working but you need the Use-RawPipeline module.):
`ssh root@<nanokvm_ip_address> "arecord -Dhw:0,0 -r 48000 -f S16_LE --buffer-time=20000" | ffplay -f s16le -ar 48000 -analyzeduration 0 -probesize 32 -> nul 2>&1`

From Linux terminal (not tested)
`ssh root@<nanokvm_ip_address> "arecord -Dhw:0,0 -r 48000 -f S16_LE --buffer-time=20000" | aplay`

corrpel, 2026-03-17:
@Z2Z-GuGu hello :) do you have any updates on this? Thank you!