sipeed/NanoKVM #37: USB keyboard/mouse problem:  [error] [hid.go:60] write to hid failed: invalid argument

CrazyCoder, 2024-09-04:
NanoKVM is connected via USB to the N100 mini PC running OPNsense (FreeBSD 14.1-RELEASE-p3).

USB devices are visible on the FreeBSD side:

`usbconfig`:
```
ugen1.2: <flyingrtx licheervnano> at usbus1, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=ON (120mA)
```

`dmesg`:
```
ugen1.2: <flyingrtx licheervnano> at usbus1
ukbd0 on uhub0
ukbd0: <flyingrtx licheervnano, class 0/0, rev 2.00/5.10, addr 36> on usbus1
kbd1 at ukbd0
uhid0 on uhub0
uhid0: <flyingrtx licheervnano, class 0/0, rev 2.00/5.10, addr 36> on usbus1
uhid1 on uhub0
uhid1: <flyingrtx licheervnano, class 0/0, rev 2.00/5.10, addr 36> on usbus1
umass0 on uhub0
umass0: <flyingrtx licheervnano, class 0/0, rev 2.00/5.10, addr 36> on usbus1
umass0:  SCSI over Bulk-Only; quirks = 0x0000
umass0:2:0: Attached to scbus2
da0 at umass-sim0 bus 0 scbus2 target 0 lun 0
da0: <Linux File-Stor Gadget 0510> Removable Direct Access SCSI-2 device
da0: 40.000MB/s transfers
da0: 21983MB (45021184 512 byte sectors)
da0: quirks=0x2<NO_6_BYTE>
```

NanoKVM Linux spills the following errors when I move the mouse cursor, type any key on a physical or on a virtual keyboard:

```
[error] [hid.go:60] write to hid failed: invalid argument
```

Rebooting NanoKVM doesn't help. Reset HID doesn't help. Connecting to other USB ports doesn't help.

HDMI works, I can see the terminal screen with the login prompt, but can't type anything.

The weird thing is that it used to work when I tried it for the first time, but not anymore. Maybe rebooting the OPNsense will help (don't want to interrupt my connection at the moment).

What else can I try on the NanoKVM side and on the FreeBSD side to make it work again?

CrazyCoder, 2024-09-04:
Flashed 1.2.1 image from scratch and it seems to fix the keyboard/mouse issue.

I was also confused by the tty showing only debug messages and no login prompt. Something got wrong with ttyv0 and switching to ttyv1 using Ctrl+Alt+F2 worked.

I'll see if USB stops working again, but at least I no longer have these errors in NanoKVM console.

wj-xiao, 2024-09-06:
There are indeed some issues with HID that can cause it to become unresponsive in certain scenarios. 
We are trying to fix it.

wj-xiao, 2024-09-10:
We have made some changes in version 2.0.7. But a complete fix requires a firmware update, so it will take some time.