sipeed/NanoKVM #38: USB keyboard/mouse problem: Compute hangs on boot with keyboard plugged in

mchugh19, 2024-09-04:
When booting an Intel Desktop Board DQ77MK with the KVM connected via the keyboard/mouse usb cable, the computer doesn't boot. The monitor shows the initial splash screen with the prompt to press keys to boot to bios settings, but hangs at that screen. No amount of keyboard input allows booting. 

If the device is rebooted without the keyboard/mouse cable connected, all is fine and it proceeds.

wj-xiao, 2024-09-10:
We have fix it in version 2.0.7. Please update.

wj-xiao, 2024-09-11:
It seems the remote host is booting from the NanoKVM virtual USB disk instead of the system boot disk. This might be related to the default setting of the motherboard. 

You could try disabling the NanoKVM virtual USB disk first, then reboot the remote host to see if it works properly.

[image]

mchugh19, 2024-09-11:
Yep, that did it! Thanks for the suggestion!

closed 2024-09-11 by mchugh19