sipeed/NanoKVM #40: Power to host via 5V USB?

geerlingguy, 2024-09-06:
I've been testing NanoKVM with a Sipeed Lichee Pi 3A board, and I'm not sure if this is a flaw in the USB implementation on the Lichee Pi carrier board, or on the NanoKVM, or expected behavior... but:

  1. If I plug in USB-C power input on the Lichee Pi
  2. And have the NanoKVM plugged in from HID (USB-C) to a USB-A port on the Lichee Pi
  3. And boot up the system

Everything works fine... _but_, when I shut down the Lichee Pi either by holding down the PWR button, or by running `sudo shutdown` via SSH, the Lichee Pi seems to try shutting down, but runs into a lower power state (the fan spins down slightly).

Further, if I turn off wall power to the Lichee Pi's USB-C power input, the USB red power LED right next to the USB-C power input stays lit and the fan still spins.

It stops spinning and the red LED goes out when I unplug the USB-A plug from the NanoKVM.

Which seems to indicate there's enough backfed power from NanoKVM into this carrier board for at least a watt or two of power—enough to make it so I can't reboot the board (not even by power cycling the wall plug!). I have to unplug the NanoKVM's USB HID connection every time I want to shut down or reboot the Lichee Pi board.

geerlingguy, 2024-09-06:
I have a USB-C power supply plugged into the AUX port on the NanoKVM (since I would like it to keep running even when the host system is powered off, and potentially not serving power through the host USB port).

korhojoa, 2024-09-07:
I plugged in a raspberry pi 5 and it powers up and boots just fine through the port with no power provided except through the NanoKVM, so I'm just going to assume there's no backfeeding protection.

Not great, but the documentation doesn't really say anything about it, so one would have to use a cable that doesn't have vbus connected if that's not desired functionality. This should at least be configurable via jumper or something.

geerlingguy, 2024-09-09:
I am in communication via email with Sipeed and it sounds like a newer revision will include a diode in place of a 0V resistor to prevent power backfeeding on 5V through the USB host port.

I will see if I can mod my own units to work that way to confirm.

@korhojoa - They also mentioned many SBCs don't include USB backfeed protection, though a Pi 5 seems like it wouldn't get enough juice through USB! Interesting—did you encounter any stability issues running like that?

Zepan, 2024-09-18:
Just copy the mail content for more users reference:
KVM-A board (the small board stack on the LicheeRV Nano) connect 5V power to LicheeRV Nano's Vsys pin, and Vsys pin defaultly connect to VBUS via 0R.
When SBC poweroff, the power from VBUS cutoff, but the power from KVM-A board's 5V0 is still there, and come back to VBUS via 0R.
The solution is remove the 0R between Vsys and VBUS, and replace with a diode like 1N5189.
[image]
The 0R is in the backside of licheerv nano, You can remove it with a hot iron.
And you can solder a through-hole diode like 1N5189  between the 2 holes right side, don't make wrong the direction.
[image]
We have found this bug earier, but at that time most user use NanoKVM with PC, and most PC board have reverse current protection circuit, so it is not a big problem.

But recently more and more user use it on SBC, and SBC usually don't have  reverse current protection circuit, so it requires we does it on NanoKVM.

In the new version, we add not only  reverse current protection circuit (not a simple diode, but an ideal diode),  but also add high voltage protection circuit, as we discovered that some users have been using non-compliant USB PD power sources that output a voltage of 7V to 12V without performing a proper handshake during power-up, which can damage the board.
[image]

voltagex, 2024-09-18:
@Zepan will you offer replacements for users who are unable to desolder that component themselves?

geerlingguy, 2024-09-18:
@voltagex - FYI just removing the diode by picking at it with a soldering iron works to solve the backfeeding bug (but it is super duper tiny... like so small I don't have an iron that will allow me to also use my precision tweezers to pull it—I had to flick it off with the hot iron lol).

The picture above is quite magnified!

ddscentral, 2024-09-19:
I sincerely hoped not to find this issue in the list. Ah well...
Would removing that 0-ohm link disable power via HID ?

geerlingguy, 2024-09-19:
@ddscentral - Yes; thus the best fix is a diode to replace the resistor. But if you're okay powering it via separate 5V USB-C power input into the Aux jack, it works out just removing the resistor.

ddscentral, 2024-09-20:
Thanks. I planned to use an external brick anyways, don't want my KVM to depend on the host system for power.
I would prefer not to modify the board though.
I think a modded USB cable with 5V pin disconnected should also work to fix the backfeed issue.

cat2devnull, 2024-09-25:
Just thought I would post here; Partly to thank @geerlingguy and @Zepan for saving me from hours of reverse engineering. But also to include a couple of detailed images so others can work out what to do.
I received my order last week and seems Sipeed are still shipping the version with the solder bridge between VSYS and VBUS which played havoc with my N5105, N100 and N305 servers. I couldn't even get them to turn on reliably and all USB ports were dead if I have the NanoKVM connected with an AUX-USB connected.
In the end, removing the solder bridge with my soldering iron took 30sec and problem solved. I might put a diode in some day but I'm not planning on running with power supplied over the PC-USB port since this defeats the purpose (turn the computer off and loose power to the KVM).
I hope this helps anyone else who was a bit confused about this process.
[image]
[image]
[image]

geerlingguy, 2024-09-25:
@cat2devnull - Oh that's strange! My units did not have that solder bridge—instead, it had a tiny 0 ohm resistor on the '5V' pad — which it looks like is not present in your photos. Interesting! I wonder if maybe the batch you have received is an even earlier one?

cat2devnull, 2024-09-25:
> Oh that's strange! My units did not have that solder bridge—instead, it had a tiny 0 ohm resistor on the '5V' pad

@geerlingguy - Ok, I wasn't sure looking at your photos because they didn't match mine. When I reached out to Sipeed directly they were initially confused but then they said I had units from "the other batch" so there must be two versions being sold at the moment. Ones like yours with a small 0 ohm resister on the 5V pads and mine with the solder bridge on the VSYS and VBUS pads.

ddscentral, 2024-09-25:
Modding a cheap USB cable to disconnect the 5V pin also worked for me. 
IIRC, mine did not have the solder bridge, only a tiny zero-ohm link.

nickw100, 2024-09-26:
Mine arrived today, it doesn't have the diode or the bridge across the pins, but the pins on the top board are bridged and i think they connect directly to the ones on the bottom.
[image]

ddscentral, 2024-09-26:
So the new "revision" has simply replaced the zero ohm link on 5V with a... different kind of zero-ohm link on 5V. 
Question is, why ?

geerlingguy, 2024-09-26:
@ddscentral - According to Sipeed, the revision I have is not the latest—the new-new-new revision is supposed to have a diode there, so power can only come in from the host USB port, not go out if you have aux power plugged into the NanoKVM.

nickw100, 2024-09-27:
@ddscentral I pre-ordered the beta version when they had soldout the first lot, but they shipped it when the new ones where in stock so I'm unsure which version this is, it could just be another beta revision.

ddscentral, 2024-09-27:
I have ordered another unit, will let you know about any changes when it arrives.

guibom, 2024-09-29:
I've just received mine, and it's exactly the same as @nickw100.
So I just double-checked and yep, there is still 5V on the HID USB port.

Jakdaw, 2024-09-30:
Had my NanoKVM arrive from Aliexpress this week - looks just like what @nickw100 is reporting. Can anyone confirm that breaking the bridge that @nickw100 highlighted next to the display will isolate the NanoKVM AUX 5v from the HID 5v?

I don't want to rely upon USB protections - I don't want to backfeed 5v into the HID cable, nor for that matter back 5v into the PSU that I'm powering the NanoKVM from!

pheuzoune, 2024-10-08:
I received mine last week. 

I ran some tests without using auxiliary power, and everything was fine. Then, I connected the NanoKVM to my Lenovo 65W USB-C power adapter on aux power to ensure the NanoKVM would stay powered while rebooting my PC. That’s when it happened:

The flaw in the design caused too much power to be sent to the USB port on my computer. Dark smoke and a strong burning smell came from the computer, and now it’s completely dead.

To be clear: beware, the NanoKVM can be dangerous for your hardware.

The PC was a Lenovo ThinkCentre M910 Tiny.

Zepan, 2024-10-09:
which "Lenovo 65W USB-C power adapter" part number?
some power adapter will output 20V voltage without protocol handshake, and our protect only upto 12V.
if you plug  power adapter that design for laptop, it is possiblely output 20V.
you can tell your "Lenovo 65W USB-C power adapter" part number, and we will buy and test it.
but we still suggest use 5V only power adapter, as we don't know if PD power adapter really follow the protocol.

patschi, 2024-10-09:
In above scenario, when measuring between Molex-GND on PSU and USB-GND on NanoKVM-AUX, there is 0.024-0.035V flowing:
[image]

Not sure if that could trip some fault protection in the PSU, as there really seems power flowing back.

patschi, 2024-10-15:
Would anyone mind double-checking me, so that I know it's a "me-problem" or general hardware design, as I'm not really a "electrical-stuff-experienced-person"? The only workaround I'd see, is somehow removing the power wires from the USB-HID...

ddscentral, 2024-10-15:
I've pre-ordered the PCI-E version as well, will report if it has the same issue when it arrives.
For my NanoKVM, I use a cheap modded USB 2.0 A to C wire with 5V pin disconnected. Works fine to fix the issue.

pheuzoune, 2024-10-15:
> which "Lenovo 65W USB-C power adapter" part number? some power adapter will output 20V voltage without protocol handshake, and our protect only upto 12V. if you plug power adapter that design for [...]

The model : LENOVO adlx65yscc3A

you can't tell the issue is on the power adapter, I use it every day to charge headsets, earbuds, smartphones, smartwatehes and basically every USB stuff I have at home. 

There's a flow identified in the power management of tne nano kvm since beginning of september and Sipeed continues to send device with issues. 
I burnt a PC with that. Support via aliexpress does not answer. 

I was ready to buy multiple PCIE version, but I won't ever by devices like this. Won't take the risk to burn other motherboards.

cat2devnull, 2024-10-16:
> Would anyone mind double-checking me, so that I know it's a "me-problem" or general hardware design, as I'm not really a "electrical-stuff-experienced-person"? The only workaround I'd see, is somehow removing the power wires from the USB-HID...

I would just work out which version you have, zero ohm resister or solder bridge and remove it with a soldering iron as per the FAQ. https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html
I think it is funny that the photo on their FAQ was actually taken from my post here 3 weeks ago. :)

patschi, 2024-10-16:
> I would just work out which version you have, zero ohm resister or solder bridge and remove it with a soldering iron as per the FAQ.

If I'm not mistaken, I do have the fixed issue and I can also confirm that power is not flowing from AUX to USB-HID:
[image]

In my case, the power however is flowing from USB-HID back to AUX - the other way around. To workaround this issue, I purchased the ["PortaPow Cased Power Blocker" from Amazon here](https://www.amazon.de/dp/B094G4P3P4) and connected it to the USB-HID port. Now I can confirm to power flowing the wrong way.

Sipeed responded very fast upon reaching out with my initial email and mis-reading my explanation (they assumed I meant power flowing from AUX to USB-HID). But since highlighting it is the other way around (from USB-HID to AUX), they have been ignoring me. 

While I can understand them being a small team and (very) busy for sure, a small "You could be right, let us check and we report back" would have been nice instead of ignoring...

ddragic, 2024-10-19:
> 
> If I'm not mistaken, I do have the fixed revision and I can also confirm that power is not flowing from AUX to USB-HID:
>...
> In my case, the power however is flowing from USB-HID back to AUX - so the other way around than what is stated in FAQ/here. To workaround this issue, I purchased the ["PortaPow Cased Power Blocker" from Amazon here](https://www.amazon.de/dp/B094G4P3P4) and connected it to the USB-HID port. Now I can confirm to power flowing the wrong way.

Just got my NanoKVM and had the same issue, VBUS and VSYS are not bridged but it's backfeeding in the other direction, from USB HID to AUX.

I solved it by cutting and pulling out the VBUS pin that connects the LicheeRV Nano to the KVM-A board. Now it can be powered only from the AUX connector over the VSYS pin and there will be no connection to VBUS (it can't be powered from the HID connector but I'm okay with that).
[image]

Also the provided schematic for KVM-A is mostly useless, older revision maybe? Didn't feel like it to tear it down and capture it.

ddscentral, 2024-10-21:
Well, the diode blocks power only one way, that is from AUX to HID. There will still be backfeed from the HID to AUX. 
A diode can't fix that.

I can see two straightforward solutions which would not require significant modifications to the board:
1. Disable powering via HID connector altogether. May not be convenient for some users.
2. Supply an USB cable with 5V pin disconnected in case user wants to power via AUX. 
Perhaps would be the cheapest way for them.

atippe, 2024-10-25:
Confirming the power backfeeding issue on my NanoKVM - arrived in the same configuration as reported by @nickw100 (bridged pins). USB tester shows 5.00V being fed back through the USB hid port.

[image]

After removing the solder bridge between VBUS and VSYS, the USB tester now shows 0.00V, confirming the backfeeding issue is resolved.

[image]

Zepan, 2024-10-28:
> Would anyone mind double-checking me, so that I know it's a "me-problem" or general hardware design, as I'm not really a "electrical-stuff-experienced-person"? The only workaround I'd see, is somehow removing the power wires from the USB-HID...
> 
> **EDIT**: I'm also concerned, if this also applies to the upcoming PCIe variant. If USB-HID is attached and power is flowing back from USB-HID back to the PCIe port... that might not be ideal...

Hi, sorry for late reply, we are busy for ship last batch in last weeks...
We add new diode in your direction in new batch (Nov. batch)
and don't worry for PCIe version, we add many diodes for all source~

Zepan, 2024-10-28:
> I've pre-ordered the PCI-E version as well, will report if it has the same issue when it arrives. For my NanoKVM, I use a cheap modded USB 2.0 A to C cable with 5V pin disconnected. Works fine to fix the issue.
Hi, sorry for late reply, we are busy for ship last batch in last weeks...
don't worry for PCIe version, we add many diodes for all source~
[image]

patschi, 2024-10-28:
> Hi, sorry for late reply, we are busy for ship last batch in last weeks...

Thank you very much for replying and confirming, @Zepan! While it's unfortunate, it makes attempting and using workarounds more comfortable.

I can totally understand that you're very busy - I really can. But ignoring my 2-3 reply emails (while responding on my initial email in just a few hours, which is amazing), has not felt welcomed. A very, very quick "We are looking into it" and "You are right. We will add diode in next revision." would have been all I wanted.

But anyway - thanks a lot for taking care!

PatrickHuetter, 2024-10-28:
@Zepan If i order today some nanoKVM devices at aliexpress, are these already from the new batch with the fix included? Or do i have to wait?

WegnerDan, 2024-11-02:
I just received five nanokvm lite today from aliexpress, ordered on 30.09.24.

I think mine still has the 0 ohm resistors, but I'm not sure.

[image]

PatrickHuetter, 2024-11-15:
Any news on this?

Bogey, 2024-11-25:
@Zepan, would you mind sharing an update please?

MrChunks, 2024-12-05:
Mine arrived today, and of course I found this thread as I was poking around, getting it updated etc. 
I am confused by a couple of things:

1. Do I have an updated board/how can I tell?
2. It sounds like there are possibly two different power path issues? But one of the fixes resolves both potential problems?
3. Does the "revised version" of the board only fix one of the path issue, so no matter what, I should be performing a fix?

I'm really hesitant to plug this in now, mostly because I'm not confident in my understanding of the problem itself.

While writing this, before posting, I took mine apart again and took photos. Comparing mine to the one posted by @atippe above, it looks like mine shipped with the pins un-bridged already. I'm going to see if I can find my USB tester and do the same confirmation he did.
[image]

brian4120, 2024-12-06:
@MrChunks 
I just discovered this today as well. As far as I can tell reading this thread, if the solder bridge between vbus and vsys (the pins you pointed to) is not present and the resistor at R63 has been removed (the one with the 5V silkscreen on the LicheeRV, the opposite side of where the OLED is near the USB C port) you are not at risk of back-feeding your computer via HID when using AUX power. 

Here is a pic I just took of the "5v" resistor removed. If you dont see a component between the two solder points, you should be good. You can double check this by plugging in a 5v power adapter via USB A to C cable to the HID port. If it stays off, the 5V resistor has been removed and will not backfeed power to the computer. 

[image]

As a side note, I have a Lenovo P710 that I am trying to break out the front panel connector for and noticed that the power button LED was being backfed via the NanoKVM. Hence me removing R63. Now it acts as expected. As others have pointed out, a diode between vbus and vsys pins is the real fix to allow both HID and AUX power.

MrChunks, 2024-12-06:
> Here is a pic I just took of the "5v" resistor removed. If you dont see a component between the two solder points, you should be good. You can double check this by plugging in a 5v power adapter via [...]

Thank you @brian4120 ! It's kind of shocking how small that diode (I guess?) is. Even with a photo I can't really see the device there (yay for getting old). But when I plugged my **AC Adapter --> USB A to USB C cable --> HID Port** the kvm powered up. So I guess I see if I can poke the diode off with a soldering iron, and then I can relax with all this. 

For comparison, here's the best macro my phone can do :) 
[image]

Solverz-0, 2024-12-30:
@brian4120 

> I am trying to break out the front panel connector for and noticed that the power button LED was being backfed via the NanoKVM. Hence me removing R63.
I cannot see how this is even possible? The 5v power from the USB-HID nor the AUX USB is even connected to the "Front Panel connection USB C port". Are you sure on this? How did you test?

>  As others have pointed out, a diode between vbus and vsys pins is the real fix to allow both HID and AUX power.
It is not though, as power is still flow in the other direction as others have proven. USB-HID ->  USB-AUX.

ckotte, 2025-03-19:
I'm sorry. I don't want to hijack this thread. 

I connected my NanoKVM via the USB-HID port to my computer and via the AUX port to an external power supply. Based on the documentation the PWR LED should glow green. In my case (I also see on a screenshot here), only the 5V LED glows in blue. 

I read somehing in the FAQ about STA LED and if it doesn't flash then there might be an issue with the hardware. Does the PWR LED glows at all or is this outdated information in the documentation/FAQ

LeonStraathof, 2025-05-07:
Can anyone tell me where this modification has to be done on the PCIe version? I seem to have the same problem that the PC will be stuck at boot 90% of the attempts at the BIOS splash screen when the HID connection is made. As soon as i remove the HID connection it boots fine. Also video seems to be much more responsive if HID is not connected, so it seems to impact the NanoKVM as well.