sipeed/NanoKVM #41: HID stops working when an image is inserted

mangelajo, 2024-09-09:
The kvm managed device is an intel N100 NUC with linux, video works correctly.

1) Boot up, connect
2) I can type on the linux console of the NUC.
3) Insert an iso image
4) I can't type on the linux console of the NUC.
5) Eject the iso image
6) I can type again.

sometimes Reset HID will help (instead of ejecting the ISO image), sometimes it won't.

razuritta, 2024-09-09:
Yup it seems that there's issue with hid emulation when changing usb config like turning on and off virtual disk/ethernet and image mounting.
You might be able to see when using `watch lsusb -tv` command to see which part is missing tho.

modzilla99, 2024-09-11:
I am affected as well, the issue still persists on `v2.0.7`.

> You might be able to see when using watch lsusb -tv command to see which part is missing tho.
The output does not change (Virtual Network=Off and Virtual USB=On) and still shows:

```
Bus 001 Device 001: ID 1d6b:0002
```

This is the kernel log when I attach an ISO, I truncated it a bit since it will spam the `signalling skipped: is not allowed by host` and `usb_gadget_wakeup` msg:

```log
Sep 11 18:40:06 kvm-4a10 daemon.info init: Id "acm" respawning too fast: disabled for 5 minutes
Sep 11 18:40:12 kvm-4a10 kern.info kernel: [  124.876906] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 11 18:40:12 kvm-4a10 kern.err kernel: [  124.876922] configfs-gadget gadget: usb_gadget_wakeup
Sep 11 18:40:12 kvm-4a10 kern.info kernel: [  124.924029] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 11 18:40:12 kvm-4a10 kern.err kernel: [  124.924045] configfs-gadget gadget: usb_gadget_wakeup
...
Sep 11 18:40:15 kvm-4a10 kern.info kernel: [  127.700560] dwc2 4340000.usb: bound driver configfs-gadget
Sep 11 18:40:15 kvm-4a10 kern.info kernel: [  127.818114] dwc2 4340000.usb: new device is high-speed
Sep 11 18:40:16 kvm-4a10 kern.info kernel: [  127.992103] dwc2 4340000.usb: new address 20
Sep 11 18:40:16 kvm-4a10 kern.info kernel: [  128.001259] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 11 18:40:16 kvm-4a10 kern.err kernel: [  128.001276] configfs-gadget gadget: usb_gadget_wakeup
Sep 11 18:40:16 kvm-4a10 kern.info kernel: [  128.303250] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 11 18:40:16 kvm-4a10 kern.err kernel: [  128.303265] configfs-gadget gadget: usb_gadget_wakeup
Sep 11 18:40:16 kvm-4a10 kern.info kernel: [  128.303354] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 11 18:40:16 kvm-4a10 kern.err kernel: [  128.303365] configfs-gadget gadget: usb_gadget_wakeup
...
```

SmartyKNX, 2024-11-12:
Any news on this topic? Using a KVM without the ability to mount an ISO image to reinstall a remote server is pretty useless to me.

modzilla99, 2024-11-12:
I was going to mention that it is now fixed, but I didn't find the issue anymore.

SmartyKNX, 2024-11-12:
I can't confirm that this has been fixed. Currently my nanoKVM is connected to an HP Prodesk and as soon as I mount an ISO image I can no longer use the keyboard.

modzilla99, 2024-11-13:
Have you tried different USB ports, specifically USB 2.0 ones?

SmartyKNX, 2024-11-15:
I'm curious. Is this a known bug that someone at Sipeed is trying to fix, or are my 5 devices some kind of expensive doorstop?

I mean, a remote KVM has to be completely reliable because the only reason for a KVM is to be able to use it without being at a remote location. If I have to go there because the KVM is unstable or not working at all, I don't see many use cases for a KVM like this.

larryqiann, 2024-11-20:
> I'm curious. Is this a known bug that someone at Sipeed is trying to fix, or are my 5 devices some kind of expensive doorstop?
>
> I mean, a remote KVM has to be completely reliable because the only [...]

I'd second this. While it's impressive that they shipped H.264 support pretty quickly, I think it's much more important for the base functionality to work correctly before any addons like this or Tailscale etc.

nwgat, 2024-11-20:
Try usb 3.0 cable

FlorianHeigl, 2024-12-07:
@nwgat doesn't help.

Z2Z-GuGu, 2025-01-20:
Currently, it is known that the keyboard cannot be used in the BIOS of DELL and HP models. However, I really have no clue about this issue. In the latest version 2.1.5 of the application, I added a feature that executes `touch /boot/usb.notwakeup && reboot` to disable USB wakeup. I know the hope is slim, but I still hope someone can help me try it out.Thank you! If it doesn't work, don't forget to use `rm /boot/usb.notwakeup && reboot` to re-enable the USB wakeup feature.

Z2Z-GuGu, 2025-01-20:
The BIOS of HP and DELL may have more requirements for HID devices. If the above method still doesn't work, I'm planning to buy a HP/DELL machine for testing, our entire company can't find any machine from these two manufacturers (>.<)

FlorianHeigl, 2025-01-21:
Hi,

> Am 20.01.2025 um 04:29 schrieb BuGu ***@***.***>:
> 
> 
> The BIOS of HP and DELL may have more requirements for HID devices. If the above method still doesn't work, I'm planning to buy a HP/DELL machine for testing, our entire company can't find any machine from these two manufacturers (>.<)
> 

Maybe hold off a bit with the hardware.

I've been having that even with an Intel NUC (Gen5, I think it was a Pro model with AMT).
The problem came up in more tests though while I just didn't understand what was going on, possible that one was a cheap gigabyte or supermicro server.

(And sorry, I can't test at the moment due to no physical access (it needed a power cycle or something to come back))

razuritta, 2025-04-13:
> I am affected as well, the issue still persists on `v2.0.7`.
> 
> > Eject the iso image
> > I can type again.
> 
> That's something I actually cannot confirm...
> 
> **EDIT**: Now I do, did not disable the `Virtual USB` toggle, unmounting the iso did not work.
> 
> **EDIT2**: When I toggle the button first, it does not work. Doing it the way around does though
> 
> > You might be able to see when using watch lsusb -tv command to see which part is missing tho.
> 
> The output does not change (Virtual Network=Off and Virtual USB=On) and still shows:
> 
> ```
> Bus 001 Device 001: ID 1d6b:0002
> ```

I mean type `lsusb` from host machine, which is the machine being controlled by nanokvm.
(just providing a way to diagnose which function is lost during toggling on/off.)

razuritta, 2025-04-13:
> The BIOS of HP and DELL may have more requirements for HID devices. If the above method still doesn't work, I'm planning to buy a HP/DELL machine for testing, our entire company can't find any machine from these two manufacturers (>.<)

I would like to help.
I have some hp machines lying around.
Is there any ways to help you with this issue?

foorschtbar, 2025-04-15:
Same problem here with a Shuttle XPC Nano

Z2Z-GuGu, 2025-06-13:
Recently, we verified some HP and DELL hosts and found that their BIOS has very strict requirements for USB HID. This may be due to security considerations, as it does not allow USB HID devices and mass storage devices to be combined into the same USB device. This has led to the situation you encountered. 

If remote installation is necessary, you will need to use a USB drive with a preloaded image in combination with the NanoKVM in "HID Only" mode.

larryqiann, 2025-06-13:
Do the NanoKVM-Pro or any other version of the hardware have any workarounds to this? I remember someone mentioning at some point that there was some hardware USB HID interface on the Pro.