sipeed/NanoKVM #42: Can't Update from 2.0.4 to 2.0.6

teedoff087, 2024-09-09:
I receive the message that an update is available:
[image]

I click Confirm to update and it acts like it's updating:
[image]

It operates for about 10-15 seconds then boots me back to the login screen, but checking the version and it's still 2.0.4.

Is there a way to update this through SSH?

SimHat, 2024-09-09:
Same issue here.

cascaabreu, 2024-09-09:
Was on 2.0.6, got the notification for 2.0.7...
After 15/30 seconds it was back online at the login screen.

Now it is back in version 2.0.4, notifying 2.0.7 is available but still does not work.
Tailscale lost its connection when it rolled back to 2.0.4, had to login again into my tailnet.

ajorg, 2024-09-10:
Same problem here (2.0.4 to 2.0.6).

What's happening here is related to #23.

When the app looks for an update, it's getting https://cdn.sipeed.com/nanokvm/latest which is just a text file that's supposed to say `2.0.7` but says `2.0.6` still because Sipeed hasn't been able to completely invalidate the cache on the CDN.

When the app tries to update, what it's really doing is downloading https://cdn.sipeed.com/nanokvm/latest.zip which actually contains the `2.0.4` version. So the update succeeds, but has no useful effect (unless you're on a version earlier than `2.0.4`, and looks like it's failing.

If you're lucky enough to get a notification for `2.0.7` that means your CDN endpoint has at least that `nanokvm/latest` object updated, but since your CDN endpoint still has `2.0.4` you can actually get downgraded.

I haven't quite figured out how to update manually. It looks just slightly more complicated that unzipping that .zip file into `/tmp`.

An improvement that would be wise would be to store the app .zip files at `nanokvm/app-<version>.zip` (in addition to `latest.zip` to support older clients) so that at the very least you'd always get the version you were trying to get, and you'd be able to downgrade if you had to I guess.

Another improvement might be to add a `?nocache=<random-string>` to the `nanokvm/latest` URL, to bypass the CDN cache, at least optionally. It's possible it's intentional that the CDN rolls out the update incrementally, so you might not want to always bypass it.

wj-xiao, 2024-09-10:
Could you try checking for updates again? I just refreshed the file cache.

@ajorg Thanks for your suggestion.

> An improvement that would be wise would be to store the app .zip files at nanokvm/app-<version>.zip (in addition to latest.zip to support older clients) so that at the very least you'd always get the version you were trying to get, and you'd be able to downgrade if you had to I guess.

The latest version includes incompatible changes. Updating from the new version to an older one might cause issues.

> Another improvement might be to add a ?nocache=<random-string> to the nanokvm/latest URL, to bypass the CDN cache, at least optionally. It's possible it's intentional that the CDN rolls out the update incrementally, so you might not want to always bypass it.

In the latest version (2.0.7), a URL parameter has been added to bypass the cache. But you'll need to update to this version first. :(

I recommend executing the update through the web. But if you can't access the latest version, I'll provide an update script.

wj-xiao, 2024-09-10:
[update-nanokvm.py](https://github.com/user-attachments/files/16939079/update-nanokvm.py.zip)

Use this script to manually update the NanoKVM.
1. Download the file and unzip it
2. Execute `python update-nanokvm.py`
3. Wait for the update to complete

**Note:** NanoKVM will automatically reboot after the update, please wait.

cedartinkerer, 2024-09-10:
I just tried again and it reports 2.0.7 is ready but it still is staying at 2.0.4

Running the script it says it updated to 2.0.7 though the toggles I saw in 2.0.6 are gone and it still has check marks next to Virtual Network and USB.  So not sure if that is indicative to anything.

wj-xiao, 2024-09-10:
Maybe the browser has cache. You can try to force refresh the web page.

Or disable browser cache:
1. Open the browser Developer Tools(F12);
2. Go to the `Network` tab;
3. Check `Disable cache` option;
4. Refresh the page.

CrazyCoder, 2024-09-10:
Updating via the `update-nanokvm.py` worked for me.

teedoff087, 2024-09-10:
> Updating via the `update-nanokvm.py` worked for me.

Same here.

geerlingguy, 2024-09-10:
Ditto with the built-in autoupdate, it showed I could go from 2.0.4 to 2.0.7, but after going through the update, it was still on 2.0.4. Trying `update-nanokvm.py` next.

cedartinkerer, 2024-09-10:
I think the logout / login back in must have refreshed the screen as it took 2x then the UI showed correctly for 2.0.7. 

So the script worked great just took more cycles for the UI to show the features of 2.0.7

ajorg, 2024-09-11:
The script did seem to work for me too. Thanks!

junglemusic, 2024-09-12:
I had the same problem. When i got told me that it wanted to update and never did until a day or two later. It got downgraded today. Wanted to update it again and it didn't work. Managed to nuke the webserver by pulling the plug when i tried it again on the third or fourth try. 
Fortunatley i was able to ssh into the shell and updated via the script above.

TheSp1der, 2024-09-13:
It should be noted that you can't use the web-shell interface to run this script. One of the first things the script does is shutdown the web-interface, killing your session. SSH directly to the NanoKVM to run the script. I had to force a reboot of the NanoKVM after the upgrade process was completed, but I'm now at the latest version. Thanks @wj-xiao for providing the upgrade script!

bfayers, 2024-09-20:
> [update-nanokvm.py.zip](https://github.com/user-attachments/files/16939944/update-nanokvm.py.zip)
> 
> Use this script to manually update the NanoKVM.
> 
> 1. Download the file and unzip it
> 2. Execute `python update-nanokvm.py`
> 3. Wait for the update to complete
> 
> **Note:** NanoKVM will automatically reboot after the update, please wait.

This script does _not_ work on my NanoKVM Lite. Exploring an SD Card image taken from a full NanoKVM had a different value in `/device_key` and putting that ID on my device temporarily allowed the update to work fine.

With my key the endpoint for download_lib responds with:
```json
{
    "code": -2,
    "msg": "encrypt failed"
}
```

It seems like not all device_keys are on the sipeed server side...

wj-xiao, 2024-09-24:
> It seems like not all device_keys are on the sipeed server side...

Please refer to #49 .

selten, 2024-09-28:
I'm currently facing while running the update-nanokvm.py 
```
download lib...
update failed
HTTPSConnectionPool(host='maixvision.sipeed.com', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?uid=[device-id] (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x3ff58d9a90>, 'Connection to maixvision.sipeed.com timed out. (connect timeout=None)'))
```

colemickens, 2024-09-29:
I cannot wait for the kernel sources to get released so a proper distribution, with proper update mechanisms can be ported.

MetalSamurai99, 2024-10-02:
Still a problem with 4 brand new NanoKVMs I just received.

Check for update.
2.0.7 -> 2.0.9
An update is available. Are you sure to update?

Confirm

Update started. Please wait...

Update failed. Please retry.

...And repeat endlessly.

The Python update script successfully updates to 2.0.9.

Bengt, 2024-12-06:
I think it is sketchy to get the updater script from a user content location, so I hosted a new implementation here:

<https://github.com/Bengt/NanoKVM-Updater/blob/main/nanokvm-updater.py>

Feel free to collaborate on it there. Patches welcome!

duffysk, 2024-12-09:
I triggered update in WebUI. After a while my WebUI became unresponsive (SSH still works).
I tried to follow python updater way, here is what I get:

`# python update-nanokvm.py
stop service...
killall: NanoKVM-Server: no process killed
OK
start update......
create temporary directory /root/nanokvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1733766313
download firmware done
download lib...
download lib done
change permissions done
update to 2.1.2 success.
restart service
the nanokvm will reboot
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
rm: can't remove '/etc/init.d/S02udisk': No such file or directory
rm: can't remove '/etc/init.d/S30wifi': No such file or directory
mkdir: can't create directory '/kvmapp/kvm': File exists
mkdir: can't create directory '/etc/kvm': File exists
# /etc/kvm/hw exist
hw = beta!
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
rmmod: can't unload module 'i2c_gpio': Resource temporarily unavailable
rmmod: can't unload module 'i2c_algo_bit': Resource temporarily unavailable
insmod: can't insert '/mnt/system/ko/i2c-algo-bit.ko': File exists
insmod: can't insert '/mnt/system/ko/i2c-gpio.ko': File exists
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
OK
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
192.168.1.18
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
ISP Vipipe(0) Allocate pa(0x8ace8000) va(0x0x3fdbf83000) size(291424)
pstSize->u32Width = 1920
pstSize->u32Height = 1080
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fdc6a62e8
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe]
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[lt6911_probe] -208 = exit success
pstSize->u32Width = 1920
pstSize->u32Height = 1080
20241209 17:50:43.606 0731 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20241209 17:50:43.615 0731 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    6911

PQBIN message
gerritId:
              commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

author:        lxowalle       desc:          gc4653 30fps
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1920
pstSize->u32Height = 1080
maix multi-media init ok
try release vio ok
try release venc ok
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
2024/12/09 17:50:43 NOTICE: Authentication is disabled! Please ensure your service is secure!
2024/12/09 17:50:43 config loaded successfully
[2024-12-09 17:50:43] [info] [logger.go:50] logger set success
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
proto: https, port: 80 443
maix multi-media driver destroyed.
ISP Vipipe(0) Free pa(0x8ace8000) va(0x0x3fdbf83000)
-- [E] Trigger signal, code:SIGSEGV(11)!`

WebUI is still not accessible still.

duffysk, 2024-12-09:
@Bengt  I ran your update script, the result looks more promising but even after reboot, WebUI is still not accessible.

```
# python nanokvm-updater.py
2024-12-09 18:55:01,065 - INFO - Starting firmware update process...
2024-12-09 18:55:01,178 - INFO - Service stop initiated
2024-12-09 18:55:01,183 - INFO - Created temporary directory /root/nanokvm-cache
2024-12-09 18:55:01,185 - INFO - Downloading firmware...
2024-12-09 18:58:34,461 - INFO - Firmware download complete
2024-12-09 18:58:34,479 - INFO - Downloading library...
2024-12-09 19:00:43,816 - INFO - Library download complete
2024-12-09 19:00:43,913 - INFO - Permissions updated
2024-12-09 19:00:44,055 - INFO - Update to version 2.1.2 successful
2024-12-09 19:00:44,057 - INFO - Starting service in background...
2024-12-09 19:00:46,073 - INFO - Service restart initiated
```

Bengt, 2024-12-10:
@duffysk Thanks for trying my script. Unfortunately, I do not know the root cause of your issue. Apparently, the service was not running initially, so maybe look into that.

joshuagamboa, 2025-02-05:
hi, what about this error? what can be done with it?
# python /etc/kvm/nanokvm-updater.py
1970-01-01 00:26:42,705 - INFO - Starting firmware update process...
1970-01-01 00:26:42,773 - INFO - Service stop initiated
1970-01-01 00:26:42,777 - INFO - Created temporary directory /root/nanokvm-cache
1970-01-01 00:26:42,779 - INFO - Downloading firmware...
/usr/lib/python3.11/site-packages/urllib3/connection.py:634: SystemTimeWarning: System time is way off (before 2022-01-01). This will probably lead to SSL verification errors
  warnings.warn(
1970-01-01 00:26:43,436 - ERROR - Firmware download failed: HTTPSConnectionPool(host='cdn.sipeed.com', port=443): Max retries exceeded with url: /nanokvm/latest.zip?n=1602 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1006)')))
1970-01-01 00:26:43,439 - ERROR - Update failed: HTTPSConnectionPool(host='cdn.sipeed.com', port=443): Max retries exceeded with url: /nanokvm/latest.zip?n=1602 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1006)')))
1970-01-01 00:26:43,461 - ERROR - Update process failed: HTTPSConnectionPool(host='cdn.sipeed.com', port=443): Max retries exceeded with url: /nanokvm/latest.zip?n=1602 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1006)')))

Freekers, 2025-02-06:
> hi, what about this error? what can be done with it?
> # python /etc/kvm/nanokvm-updater.py
>
> 1970-01-01 00:26:42,705 - INFO - Starting firmware update process... 1970-01-01 00:26:42,773 - INFO - [...]

Try the script from @Bengt , it solved the issue for me: https://github.com/sipeed/NanoKVM/issues/42#issuecomment-2521833720
Thanks @Bengt  :)

joshuagamboa, 2025-02-08:
Thank you! That's exactly what I did. The loop continued but I just left it. When I came back, it is already working normally! Have enabled Tailscale and it rocks. Hope they add these fixes and workarounds on the FAQs.

wj-xiao, 2025-02-14:
Version 2.1.6 is now available. Please refer to [CHANGELOG.md](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md) for details.

The installation package has been released on [GitHub](https://github.com/sipeed/NanoKVM/releases/tag/2.1.6).
If you are still unable to get the update, you can refer to the [README.md](https://github.com/sipeed/NanoKVM/blob/main/server/README.md#manually-update) for manual update.

closed 2025-03-11 by Z2Z-GuGu

phyco1991, 2025-06-09:
Not sure why this issue is closed, I'm seeing the same problem currently trying to upgrade from 2.2.6 to 2.2.8

I've tried SSHing to the KVM and running the script but it just reinstalled 2.2.6...

mrjackson, 2025-07-06:
I think something like this has happened again?
https://cdn.sipeed.com/nanokvm/latest shows 2.2.8
https://cdn.sipeed.com/nanokvm/latest.zip shows 2.2.9

chf-6309, 2025-07-15:
I try to update from 2.2.5 to 2.2.8 via the nanoKVM webpage, notice that 2.2.9 don't show up (?), and it seem verrrrryyyy long. Tried to download the zip files from https://cdn.sipeed.com/nanokvm/latest.zip and FireFox display a download speed of 9,7kB/S or 12,1kB/s (on my ~400Mb/s fiber connection)... any problem with the update server?

mihalski, 2025-10-26:
I still can't figure out how to update. I can't download the update python zip file on the nanokvm. Neither wget or curl can download it.