sipeed/NanoKVM #43: HDMI Resolution support

carrierone, 2024-09-10:
I am thinking about buying one or more 5 packs of these NanoKVMs however most of my old servers have SVGA output that I want to use them with, I am concerned about this because the specifications say it supports "1080p" and that is all it mentions.  With the PiKVM 4 by contrast I tried out several VGA->HDMI adapter cables/boxes which convert the analog signal to digital, but it struggles to sync up to different resolutions even if its 720p for example.  The adapter cable/boxes tend to output a signal in various formats that would be I think comparable to DVI signals with various resolutions.  Anyway, I was wondering if the NanoKVM can support tweaking the HDMI/CSI input so that I could get these to work typically they output 640x480 or 800x600 during boot process to text based BIOS and linux OSes is all I need to work.  If not, I guess I can just use these things for media emulation because most of these servers have working KVM/IPMI however the ISO emulation is broken with newer java, but I do have some linux OSes that don't get along with the Supermicro IPMI KVM for whatever reason...  Thanks!

carrierone, 2024-09-10:
I found this in the user manual, but its got a huge note at the end about the host resolution.... and no detail about the options available for host system settings

> 2.1. Resolution, Frame Rate, and Image Quality Settings[#](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html#Resolution%2C-Frame-Rate%2C-and-Image-Quality-Settings)
> NanoKVM supports 1080P, 720P, 600P, and 480P image transmission. Different resolutions can be selected in Image Settings -> Resolution. Higher resolutions use more bandwidth and lower real-time frame rates.
> 
> Note: This only changes the transmitted image size, not the HDMI input image size. To change that, please go to the host system's settings menu.

polyzium, 2024-09-10:
If you're asking if the NanoKVM syncs properly with these VGA-to-HDMI adapters, I don't think it does at the time of posting this comment. They recently fixed a bug where the NanoKVM would not display resolutions other than 1080p. See https://github.com/sipeed/NanoKVM/issues/5#issuecomment-2338590727 for more details.

Z2Z-GuGu, 2024-09-11:
@carrierone Hi, in previous versions of KVMAPP, NanoKVM only supported a physical resolution of HDMI 1920 * 1080, so when switching to other resolutions, there was an issue where it could not be displayed (Windows prioritizes changing the desktop resolution when modifying the resolution, and the physical resolution will not be modified, so the above problem will not occur). We have recently provided an update 2.0.7, which supports more resolutions (1920 * 1080; 1600 * 900; 1440 * 1080; 1440 * 900; 1280 * 1024; 1280 * 960; 1280 * 800; 1280 * 720; 1152 * 864; 1024 * 768; 800 * 600), but due to the encoder of the SG2002 chip, some resolutions may experience screen tearing issues (1680 * 1050; 1440). We are trying to modify the EDID of HDMI to mask these problematic resolutions (1050; 1368 * 768; 1366 * 768; 720 * 576)

carrierone, 2024-09-11:
That is great @Z2Z-GuGu  I will have to buy a nanokvm when it is released and try it out since I already have a bunch of VGA->HDMI and let you know if it works

Z2Z-GuGu, 2024-09-12:
We once published a video on X about using VGA to HDMI to connect a computer and NanoKVM. You can refer to [it](https://x.com/SipeedIO/status/1831518671065227603)

closed 2024-09-18 by Z2Z-GuGu