sipeed/NanoKVM #44: Virtual USB does not work on certain environments.

JP-Dora1080, 2024-09-10:
What I am trying to do now is to combine my StarTech KVM switch with NanoKVM Lite to allow me to switch screens virtually. If you watch the video I posted on my X, it will be easier to understand. In that video, I am using a physical keyboard connected to the KVM switch, and I have the NanoKVM Lite plugged into a USB port on the StarTech KVM switch, but it is only providing power and the keyboard and mouse on the remote are not working. I have tried changing the cables, but that has not fixed the problem. I connected the USB directly to the server and the keyboard worked only on that server. I would like to be able to control all servers from the KVM switch. Can you fix this problem? Thank you in advance.

https://x.com/DORA_0907/status/1833134326923640963

Benchwork, 2024-09-11:
what firmware are you using? 
I was using the stock firmware and was having issues with keyboard/mouse. I updated to the most current firmware (requires rewriting the SD card, and now it is working

JP-Dora1080, 2024-09-12:
All are up-to-date with the latest version of NanoKVM, but this does not solve the problem. Is there any other solution?

[image]

polyzium, 2024-09-12:
Look in the dmesg of the other server where the keyboard doesn't work, and paste the log here. My guess is that either the NanoKVM or the switch is not responding to the USB initialization procedure.

JP-Dora1080, 2024-09-12:
DMESG Log. The log shows that the physical keyboard is recognized on the server by relaying through the USB port of the KVM switch, but nothing is recognized on NanoKVM's Virtual USB. If the physical keyboard is recognized, I assume that there is some kind of bug on the NanoKVM side that is preventing it from being recognized.

[image]

polyzium, 2024-09-12:
Hm, then I'll leave this question to the Sipeed employees.

onix331, 2024-11-16:
It also does not work with memtest86+ v7.00 (latest) on the latest KVM version (2.1.1).
While the screen says it detected hid devices, it does not react when i press any key