sipeed/NanoKVM #45: Having HID plugged in stops PC from booting

rusty1256, 2024-09-13:
I have my media PC set to always power on after a power loss. With HID cable plugged in the PC does not start until I remove the HID cable from the PC or power on using nanokvm FULL.

Any way around this?

geerlingguy, 2024-09-17:
I'm 99% sure this has to do with #40 — I noticed if I have it plugged in, while I also have power plugged into NanoKVM separately, the PC won't boot. I'm guessing some motherboards have USB power backfeed protection that will block the computer from booting if a device is sending power into the 5V input, for protection.

Z2Z-GuGu, 2024-09-18:
I noticed a problem before: inserting a USB flash drive caused the computer to fail to boot up. Perhaps you can try setting up the virtual USB flash drive on the webpage and turning it off.

rusty1256, 2024-09-20:
I have tried a power blocker usb i have.
Test conditions:
HID - powerblocker -PC
AUX power connect to mains.

Issue still occurs. So not sure its a backfeed issue

I also have my motherboard to power on when power is restored. I have a managed PSU so I use that to cut and restore power as needed