sipeed/NanoKVM #46: Has anyone received their order yet?

corndog2000, 2024-09-13:
I've emailed Sipeed twice about the status of my order. I have not heard back from them. Is this product legit, or should I try and get a refund through PayPal?

pdostal, 2024-09-13:
I did receive out of nowhere last week. I'm in Czechia - Central Europe.

polyzium, 2024-09-13:
If you used the pre-order page, this is probably why. I received mine via AliExpress within 9 days. Regardless, Sipeed's neglect of pre-orders seems deeply concerning.

patschi, 2024-09-13:
I ordered mine via Aliexpress 4 weeks ago and received nothing yet. But from what I heard, they're shipping based on the date of ordering and expect to send out all before end of September.

They received approx. 2000 orders on Aliexpress in the last ~1.5 months I believe, so possibly a bit overwhelmed with the amount of orders as they are a small (?) team.

polyzium, 2024-09-13:
Sounds like a plausible scenario. A lot of people are looking at the product, given the aggressive price point compared to the PiKVM.

ccdoggy, 2024-09-13:
Reach out to support@sipeed.com if it was a preorder and has been a long time, I pre-ordered July 15th and just received it yesterday (US) which is just shy of 2 months.  Support was helpful and gave me tracking info.  Its clear they are doing their best to manufacture a fairly unique product with relatively high demand while also developing it further.

TheSp1der, 2024-09-13:
I was also worried that I had tossed my cash into the ether, but these things are legit. They really are wonderful little devices and seem to work well. I was like you and also questioned why it was taking so long to receive them, it took a few days form them to get back to me when I inquired as to my order status (support@sipeed.com), but they did eventually reply with a tracking number. If you do ask them for a status try to include all your preorder details in the message.

corndog2000, 2024-09-13:
I've emailed support twice with my order information. No reply after a week.

egallis31, 2024-09-15:
received mine out of nowhere today - west coast us

corndog2000, 2024-09-15:
Support finally responded saying my pre-order would ship mid to late October.

closed 2024-09-15 by corndog2000