sipeed/NanoKVM #47: AliExpress link in Readme is broken

corndog2000, 2024-09-13:
The AliExpress link loads, but then goes to a 404 page. I'm in the US so it's being redirected to aliexpress.us.

polyzium, 2024-09-13:
AliExpress always gives you a 404 if you have a US IP (I think this is related to the Chinese sanctions). From a French IP, the page opens fine.

[image]

closed 2024-09-14 by corndog2000

cblknittights, 2024-09-22:
> AliExpress always gives you a 404 if you have a US IP (I think this is related to the Chinese sanctions). From a French IP, >the page opens fine.

What sanctions??? I order all kinds of tech stuff on AX all the time. Got MB's etc.. all from AX... 

Its more than IP.. I can use my VPN node in Luxemborg, and its fine, TILL I log in and my US account, starts the 404 redirection. With out logging in and using my Luxenborg node, just switching to US shipping, 404 redirect. 

SO this is "more than IP" based. 

Don't see what the issue here is.. more 
 
> For the USA you need to use the pre-order page in the README.

IF the tech is sanctioned then selling it in the US is an issue, period. This "pre-order" v. AX seems to be something more...
I'd like more details.

Honestly I could order 10 Full versions right now, and thats just my own use. I could easily order 10x that for other clients etc...

This thing just got a HUGE EXPOSURE via a tech TV show, attempt to purchase it, and I am met with huge hurdles... guess what... 98% of the customers from this exposure, when they hit the issue on AX, left! They ain't coming back. 

So what ever this "issue" is with AX, solve, and get them.

I am over here waving... my $$$$ TAKE IT! ! ! But can't... sigh...