sipeed/NanoKVM #49: download lib failed

dcompagn, 2024-09-14:
Hi, 

I've got the NanoKVM Lite. 
I flashed the SD card with the firmware v1.2.0 but the video capture is not working. 
Then, I run the `update-nanokvm.py` script but I got this error

`download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1726304869
download firmware done
download lib...
update failed
download lib failed, content_type: application/json; charset=utf-8` 

Digging into the script I see that a device key is part of the URL 
`"https://maixvision.sipeed.com/api/v1/nanokvm/encryption?uid={device_key}"`
The device key is read from `/device_key`.

The error message I get is "encrypt failed":

`application/json; charset=utf-8, b'{"code":-2,"msg":"encrypt failed"}'`

wj-xiao, 2024-09-18:
You are right. Apologize for the issue.
Could you please provide your device key?

You can get the device key by executing `cat /device_key`.

bfayers, 2024-09-21:
> You are right. Apologize for the issue. Could you please provide your device key?
> 
> You can get the device key by executing `cat /device_key`.

I've got the same issue with `d1c15bae1526b59a`

wj-xiao, 2024-09-24:
@bfayers Sorry for that. The device key has been added. Please try again.

TharsisZen, 2024-09-29:
Same issue with 2 separate new devices delivered a few days ago.
Lite: d1ca1f0c1359c262
Full: d1c188ba1526b59a

dcompagn, 2024-10-05:
Hi, this is my device key.
`d1c11b1b1526b59a`

wj-xiao, 2024-10-08:
@dcompagn Added. Please try again.

closed 2024-11-08 by wj-xiao

Evs91, 2024-12-19:
this is mine; if you could check for it
`d1ccc40f1359c262`

Kellanji, 2024-12-26:
Hi!
My device key is d1ccca1d1359c262

muness, 2025-01-23:
Hi, my device key is `d1be23991359c262`. Can you add it, @wj-xiao ? Thanks!

Finalemu, 2025-06-24:
I have the same problem. My device id is d1c1913c1526b59a