sipeed/NanoKVM #50: HID/USB not working

OrpheeGT, 2024-09-14:
Hello,

I'm using a HP N54L (Microserver G7)

The keyboard (and mouse I guess) is not working. the HID icon on NanoKVM screen off.

Under BIOS, it seems to be detected as HUB (the keyboard detected is a real keyboard plugged to be able to access BIOS).
[image]

Under Linux, the NanoKVM is detected as
```
Bus 002 Device 006: ID 3346:1009 flyingrtx licheervnano
```
[image]

I tried to reset HID from NanoKVM menu...
[image]

detailed lsusb : 

```
root@rr:/opt/rr# lsusb -v -d 3346:1009

Bus 002 Device 008: ID 3346:1009 flyingrtx licheervnano
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x3346 flyingrtx
  idProduct          0x1009 licheervnano
  bcdDevice            5.10
  iManufacturer           1 flyingrtx
  iProduct                2 licheervnano
  iSerial                 3 0123456789ABCDEF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0069
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 licheervnano
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              120mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              5 HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      63
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0006  1x 6 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0006  1x 6 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      2 Mouse
      iInterface              7 HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      52
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      2 Mouse
      iInterface              9 HID Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      74
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0006  1x 6 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0006  1x 6 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

OrpheeGT, 2024-09-14:
```
Sep 14 17:38:08 kvm-e457 auth.err getty[3887]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 14 17:38:18 kvm-e457 auth.err getty[3913]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 14 17:38:28 kvm-e457 auth.err getty[3938]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 14 17:38:29 kvm-e457 kern.err kernel: [ 4074.315137] configfs-gadget gadget: End Point Request ERROR: -108
Sep 14 17:38:34 kvm-e457 kern.warn kernel: [ 4078.997832] LT6911--:-- INT!!##
Sep 14 17:38:34 kvm-e457 kern.info kernel: [ 4079.107255] close channel no. 0
Sep 14 17:38:35 kvm-e457 kern.info kernel: [ 4079.608975] vi_event_handler exit
Sep 14 17:38:35 kvm-e457 kern.info kernel: [ 4079.860366] vpss_release
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 14 17:38:35 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:714:SAMPLE_COMM_VI_StopViChn(): CVI_VI_DisableChn failed with 0xc00e8007!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 14 17:38:35 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:948:SAMPLE_COMM_VI_StopSingleViPipe(): CVI_VI_StopPipe failed with 0xc00e8007!
Sep 14 17:38:35 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:971:SAMPLE_COMM_VI_StopViPipe(): SAMPLE_COMM_VI_StopViPipe is fail
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 14 17:38:35 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:606:SAMPLE_COMM_VI_StopDev(): CVI_VI_DisableDev failed with 0xc00e8007!
Sep 14 17:38:35 kvm-e457 kern.err kernel: [ 4079.873228] vb_get_pool_info:536():  NULL VB HANDLE
Sep 14 17:38:35 kvm-e457 kern.err kernel: [ 4079.874723] vb_get_pool_info:536():  NULL VB HANDLE
Sep 14 17:38:35 kvm-e457 kern.info kernel: [ 4079.875428] vpss_open
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:214:lt6911_probe(): read sensor id error. 
Sep 14 17:38:35 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:222:lt6911_probe(): Sensor ID Mismatch! Use the wrong sensor??
Sep 14 17:38:35 kvm-e457 local5.warn : [SYS-WRN] sample_common_bin.c:36:SAMPLE_COMM_BIN_ReadParaFrombin(): Bin exist (/mnt/cfg/param/cvi_sdr_bin)
Sep 14 17:38:36 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:36 kvm-e457 kern.err kernel: [ 4080.977635] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:37 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:37 kvm-e457 kern.err kernel: [ 4082.001624] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:38 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:38 kvm-e457 kern.err kernel: [ 4083.025717] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:38 kvm-e457 auth.err getty[3996]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 14 17:38:39 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:39 kvm-e457 kern.err kernel: [ 4084.049765] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:40 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:40 kvm-e457 kern.err kernel: [ 4085.073843] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:41 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Sep 14 17:38:41 kvm-e457 kern.err kernel: [ 4086.097900] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Sep 14 17:38:42 kvm-e457 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
[... cut ...]
Sep 14 17:41:26 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:222:lt6911_probe(): Sensor ID Mismatch! Use the wrong sensor??
Sep 14 17:41:26 kvm-e457 local5.warn : [SYS-WRN] sample_common_bin.c:36:SAMPLE_COMM_BIN_ReadParaFrombin(): Bin exist (/mnt/cfg/param/cvi_sdr_bin)
Sep 14 17:41:27 kvm-e457 kern.info kernel: [ 4251.672655] open channel no. 0
Sep 14 17:41:27 kvm-e457 kern.info kernel: [ 4251.673038] close channel no. 0
Sep 14 17:41:27 kvm-e457 kern.info kernel: [ 4251.673101] open channel no. 0
Sep 14 17:41:29 kvm-e457 kern.info kernel: [ 4253.997247] dwc2 4340000.usb: new device is high-speed
```

OrpheeGT, 2024-09-14:
It actually seems to be detected once an advanced OS boots : 

[image]

but this OS actually does not have keyboard/mouse available once booted... only remote web access (NAS)

So basically Keyboard seems to be available once I don't need it anymore... I need it under BIOS and at GRUB boot...

theadlee, 2024-09-17:
Same issue here.

Z2Z-GuGu, 2024-09-18:
Perhaps you can modify the configuration file to achieve BIOS compatibility:
`vi /etc/init.d/S03usbdev`
# keyboard
mkdir functions/hid.GS0
# echo 1 > functions/hid. GS0/subclass
Cancel the comment on the previous line and `reboot`

OrpheeGT, 2024-09-18:
I tried.

It freeze the POST boot menu.
[image]

```
# cd /var/log
# tail -f messages 
Sep 18 13:36:17 kvm-e457 kern.info kernel: [   38.052869] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:36:17 kvm-e457 kern.err kernel: [   38.052885] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:36:19 kvm-e457 auth.err getty[610]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 18 13:36:29 kvm-e457 auth.err getty[636]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 18 13:36:35 kvm-e457 kern.info kernel: [   55.563855] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:36:35 kvm-e457 kern.err kernel: [   55.563871] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:36:35 kvm-e457 kern.info kernel: [   55.563956] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:36:35 kvm-e457 kern.err kernel: [   55.563967] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:36:38 kvm-e457 auth.info sshd[656]: Accepted publickey for root from 192.168.1.132 port 60360 ssh2: RSA SHA256:ZjL77cRATr9WudUDkb29FSGa28JOczP+tU2cj7V2xSo
Sep 18 13:36:39 kvm-e457 auth.err getty[665]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 18 13:36:43 kvm-e457 kern.info kernel: [   63.671909] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:36:43 kvm-e457 kern.err kernel: [   63.671925] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:36:43 kvm-e457 kern.info kernel: [   63.672008] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:36:43 kvm-e457 kern.err kernel: [   63.672018] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:36:49 kvm-e457 auth.err getty[692]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 18 13:36:54 kvm-e457 kern.err kernel: [   75.252634] configfs-gadget gadget: End Point Request ERROR: -108
Sep 18 13:36:57 kvm-e457 kern.warn kernel: [   78.342127] LT6911--:-- INT!!##
Sep 18 13:36:58 kvm-e457 kern.info kernel: [   78.447637] close channel no. 0
Sep 18 13:36:58 kvm-e457 kern.info kernel: [   78.548762] vi_event_handler exit
Sep 18 13:36:58 kvm-e457 kern.info kernel: [   78.800930] vpss_release
Sep 18 13:36:59 kvm-e457 auth.err getty[719]: can't open '/dev/ttyGS0': No such file or directory^M
Sep 18 13:37:04 kvm-e457 kern.warn kernel: [   85.016412] LT6911--:-- INT!!##
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 18 13:37:04 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:714:SAMPLE_COMM_VI_StopViChn(): CVI_VI_DisableChn failed with 0xc00e8007!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 18 13:37:04 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:948:SAMPLE_COMM_VI_StopSingleViPipe(): CVI_VI_StopPipe failed with 0xc00e8007!
Sep 18 13:37:04 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:971:SAMPLE_COMM_VI_StopViPipe(): SAMPLE_COMM_VI_StopViPipe is fail
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] cvi_vi.c:101:CHECK_VI_CTX_NULL_PTR(): Call SetDevAttr first
Sep 18 13:37:04 kvm-e457 local5.err : [LOG-ERR] sample_common_vi.c:606:SAMPLE_COMM_VI_StopDev(): CVI_VI_DisableDev failed with 0xc00e8007!
Sep 18 13:37:04 kvm-e457 kern.err kernel: [   85.069894] vb_get_pool_info:536():  NULL VB HANDLE
Sep 18 13:37:04 kvm-e457 kern.err kernel: [   85.070988] vb_get_pool_info:536():  NULL VB HANDLE
Sep 18 13:37:04 kvm-e457 kern.info kernel: [   85.071663] vpss_open
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:85:lt6911_read_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:136:lt6911_write_register(): I2C_WRITE error!
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:214:lt6911_probe(): read sensor id error. 
Sep 18 13:37:04 kvm-e457 local5.err : [VI-ERR] lt6911_sensor_ctl.c:222:lt6911_probe(): Sensor ID Mismatch! Use the wrong sensor??
Sep 18 13:37:04 kvm-e457 local5.warn : [SYS-WRN] sample_common_bin.c:36:SAMPLE_COMM_BIN_ReadParaFrombin(): Bin exist (/mnt/cfg/param/cvi_sdr_bin)
Sep 18 13:37:05 kvm-e457 kern.info kernel: [   85.697857] open channel no. 0
Sep 18 13:37:05 kvm-e457 kern.info kernel: [   85.698224] close channel no. 0
Sep 18 13:37:05 kvm-e457 kern.info kernel: [   85.698293] open channel no. 0
[... cut ...]
Sep 18 13:41:51 kvm-e457 daemon.info ntpd[415]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Sep 18 13:42:13 kvm-e457 kern.info kernel: [  393.449351] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
Sep 18 13:42:13 kvm-e457 kern.err kernel: [  393.449367] configfs-gadget gadget: usb_gadget_wakeup
Sep 18 13:42:13 kvm-e457 kern.err kernel: [  393.469731] configfs-gadget gadget: End Point Request ERROR: -108
Sep 18 13:42:13 kvm-e457 kern.info kernel: [  393.509274] dwc2 4340000.usb: bound driver configfs-gadget
```

OrpheeGT, 2024-09-19:
2.0.9 release does not fix the issue.

theadlee, 2024-09-19:
I am seeing that same behavior with an HP elitedesk G4 mini.  Legacy USB enabled.  Cannot hit get into BIOS (ESC or F10) and Grub menu does not recognize keyboard.

AkechiShiro, 2024-09-29:
Maybe this could help : https://forum.ultimatehackingkeyboard.com/t/bios-legacy-hid-mode/715/5