sipeed/NanoKVM #51: Rebooting the NanoKVM resets the PC

depau, 2024-09-16:
Hi,
I received the NanoKVM full and I tested it. I connected it to the motherboard connector, tested the buttons on the device and they work as expected.

However, I'm experiencing two issues:

- When the NanoKVM reboots and the ATX board is connected, the PC is reset
- If the PC is off and I try to remotely press the power button to power it on, it powers on for half a second then it powers back off (but that doesn't happen if I press the physical buttons on the NanoKVM device).

What could be the issue?

Z2Z-GuGu, 2024-09-18:
This seems to be a widespread issue on older hardware versions, I will add an explanation on the wiki later.

closed 2024-09-18 by Z2Z-GuGu

depau, 2024-09-25:
I saw your comment on the other thread and looking at the schematics I noticed soldering the resistors to the B board shouldn't make that big of a difference (please correct me if I'm wrong), I didn't wanna solder them in such a packed enclosure.

[image]

Now it looks like remote power on/off works as expected. Unfortunately it still resets the PC on reboot. Should I move the resistors to the Lichee board?

Z2Z-GuGu, 2024-09-26:
> Now it looks like remote power on/off works as expected. Unfortunately it still resets the PC on reboot. Should I move the resistors to the Lichee board?

I replicated this issue on early hardware and attempted to add resistors to the host, but unfortunately the problem still persists. Through further testing, I found that the pin controlling the reset of the chip experiences a brief high level during the initial startup phase (not weak pull-up of other pins), which prevents the problem from occurring no matter how small the resistor is pulled down
If we want to avoid this issue, we can only disconnect RESET from KVM-B. This phenomenon occurred in early hardware, but later versions have modified the control pins to solve this problem.

depau, 2024-09-26:
I see, thank you. Is this pulse short enough to be smoothed down with a capacitor?

Z2Z-GuGu, 2024-09-26:
> I see, thank you. Is this pulse short enough to be smoothed down with a capacitor?

It may be difficult because it takes over 1 second and requires a big capacitor

depau, 2024-09-26:
> takes over 1 second

That would definitely require a *big* capacitor.

Would it be possible to get replacement PCBs given the hardest issue? 

TBH it wasn't really clear from the marketing material that the preorder version would be an early revision with potential issues and no Wi-Fi

kocane, 2024-09-29:
Does this mean my NanoKVM ordered Jul 21 definitely has this issue?

Z2Z-GuGu, 2024-09-30:
> Does this mean my NanoKVM ordered Jul 21 definitely has this issue?

Please execute `cat /etc/kvm/hw`. If the output is `alpha`, then your NanoKVM will have this issue

Z2Z-GuGu, 2024-09-30:
> Would it be possible to get replacement PCBs given the hardest issue?

@kocane Hi, how many NanoKVM Full do you have? Also, in which region are you located? I need to inquire if there is a possibility of sending them.

After internal discussions, the WiFi version will not appear on the NanoKVM Full form, but will exist on the future NanoKVM PCIe.

depau, 2024-09-30:
> > Would it be possible to get replacement PCBs given the hardest issue?

Sorry for the typo, I meant to type "hardware issue".

> Hi, how many NanoKVM Full do you have? Also, in which region are you located? I need to inquire if there is a possibility of sending them.

I have one NanoKVM full (alpha). I live in Italy. My GitHub profile info should match the info I specified in the preorder.

> After internal discussions, the WiFi version will not appear on the NanoKVM Full form, but will exist on the future NanoKVM PCIe.

NanoKVM PCIe sounds exciting!

kocane, 2024-10-01:
> > Does this mean my NanoKVM ordered Jul 21 definitely has this issue?
> 
> Please execute `cat /etc/kvm/hw`. If the output is `alpha`, then your NanoKVM will have this issue

Thanks. It seems like mine is beta according to /etc/kvm/hw. So I'm good :-)?