sipeed/NanoKVM #52: Video update slow

wilpig, 2024-09-16:
Got my NanoKVM lite in and it's pretty nifty.  I am liking the size and was able to get the firmware loaded fine, and with the help of the python script got it to 2.0.7.

I've futzed around with the resolution on my host machine to make them happy and in sync at 1280x720, I bumped the refresh up to 60, tried disabling the frame detection but i'm still only getting screen updates about every 10 - 15 seconds.  I have vnc on that system so I could watch and the cursor is keeping up with my movements in the browser just fine it's just the video image that is delayed.  Is there a setting i'm missing or is something else at play here that i'm missing?

geerlingguy, 2024-09-17:
It seems like I've been hitting the same thing on my Windows desktop—firmware 2.0.7 seems to be very sluggish, compared to 2.0.3 that I was running before.

geerlingguy, 2024-09-17:
Found out the issue *seems* to be related to #40 — I unscrewed the bottom and removed the heatsink, then used a soldering iron to remove the tiny (very tiny) 0 ohm resistor in the little '5V' area.

This fixed the USB power backfeed issue and seemed to not only fix the slowness, but also allow USB HID stuff to work properly.

I also had to set my refresh rate in Windows settings to 24Hz to remove some strange screen tearing that was showing up.

wj-xiao, 2024-11-08:
We have release version 2.1.0 which includes support for H.264.
Could you please update and verify if this issue still exists?

MorganMLGman, 2024-12-02:
Hi, I also received my NanoKVM lite today, straight away flashed the newest firmware on sd card and… it was terrible, almost unusable. Best case 8 seconds between frames worst case no video. I found in this issue https://github.com/sipeed/NanoKVM/issues/86 that this could be "not enough power". So in addition to this tiny resistor labelled 5V I added thick solder joint between VBUS and VSYS pins. Didn't have much hope in this solution but to my own surprise like night and day, almost instant video response. I don't know how to measure latency on this thing, but it doesn't feel far from regular vnc.

wilpig, 2025-01-02:
Finally got a chance to update mine to the latest, which is 2.1.3 today, and it is working quickly as expected without having to touch the 5v.

avg-I, 2025-01-07:
Recently received my NanoKVM, just started playing with it and see very slow response times.
Actually, they vary quite a lot, sometimes it's like a second or two, sometimes it's more than 5.
I've verified that the kernel gets input events very promptly, so it must be something with capturing / transmitting video output. I tried to play with settings, but nothing improved the situation.

Ping times are good, about half a millisecond, but there is also a 10% packet loss for some reason.

Firmware version is 2.1.3.

fjluquerodriguez, 2025-02-09:
Versión 2.1.5 here and same problem. H264 doest works and jpeg very slow. Very very slow. Pung less than 10ms.