sipeed/NanoKVM #53: Support Netbird VPN

nazarewk, 2024-09-17:
It would be great to run Netbird on NanoKVM in addition to Tailscale and others.

Not sure what exactly is the process here, but I have created a [RISC-V feature request ](https://github.com/netbirdio/netbird/issues/2611)

wj-xiao, 2024-09-18:
If the app has a RISC-V version, simply download and run it.
Treat NanoKVM as a RISC-V Linux distribution.

matu6968, 2024-09-23:
> If the app has a RISC-V version, simply download and run it. Treat NanoKVM as a RISC-V Linux distribution.

Officially Netbird does not support RISC-V support, but golang **does** support cross compiling (with a few shell variables). To cross compile the client to RISC-V simply follow the [Netbird client build instructions](https://github.com/netbirdio/netbird/blob/main/CONTRIBUTING.md#build-and-start) and add the following lines before CGO_ENABLED=0: `env GOOS=linux GOARCH=riscv64` (adding env to the commands may not be needed)

v01ded, 2025-04-21:
Hope the see that! Love Netbird abilities to have access control. Useful for all NanoKVM to control who have access to the NanoKVM. Thanks

trbutler, 2025-07-15:
This is fantastic. Thanks @bc547 !

dormancygrace, 2026-09-20:
Hi! I've added NetBird support to [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my community firmware for NanoKVM. If you've been looking for a way to use NetBird on your device, you're welcome to try it and share your feedback!