sipeed/NanoKVM #55: IPv6

Gunni, 2024-09-18:
I cannot find it in the specs, does this device support IPv6?

Does it function without IPv4?

If no, why not?

lion7, 2024-09-26:
I can confirm it works with IPv6. Received my NanoKVM yesterday and it works straight out of the box in a dual-stack network on both IPv4 and IPv6 addresses.

closed 2024-09-26 by Gunni

hardillb, 2024-12-05:
Likewise I only see link-local IPv6 addresses

kedama222, 2024-12-06:
I noticed that the file `/etc/sysctl.d/99-tailscale.conf` contains the setting:
`net.ipv6.conf.all.forwarding = 1`.

While this configuration is documented on [Tailscale's How-to Guides](https://tailscale.com/kb/1019/subnets#enable-ip-forwarding) as part of enabling subnet routing, I believe NanoKVM is not primarily intended to function as a subnet router.

kedama222, 2024-12-14:
related #197