sipeed/NanoKVM #56: Video isn't restored after reboot, so I can't see the bootloader

voltagex, 2024-09-18:
[image]

This is the last thing I see with the HDMI capture plugged into my Framework laptop.

wj-xiao, 2024-09-19:
Could you check your application version (setting -> About NanoKVM)?
Cause a related issue was fixed in version 2.0.7.

wj-xiao, 2024-09-19:
Or you can try to close the frame detect
[image]

voltagex, 2024-09-20:
@wj-xiao yes, I'm on 2.0.7.

I have tried disabling Frame Detect.

I think this might be a Framework Laptop issue, I'll try with another system later.

CRCinAU, 2024-10-09:
Just to check this - I noticed an Intel NUC will shut off power to the USB ports when rebooting - so if you're not also powering the IP KVM via an external method, the IPKVM will also reboot - meaning you won't see the BIOS screens etc...

Might be worth checking if this is the same problem - ie power your IP KVM from a different USB source.

patschi, 2024-10-09:
> Might be worth checking if this is the same problem - ie power your IP KVM from a different USB source.

Sorry for off-topic, but I'd like to make aware about this issue on some earlier NanoKVM revisions: https://github.com/sipeed/NanoKVM/issues/40

wj-xiao, 2024-11-08:
Please update to version 2.1.0 in the webpage. We have added support for H.264.