sipeed/NanoKVM #61: Can't stream any video at all

bfayers, 2024-09-21:
I've had my NanoKVM plugged into my macbook for testing but I never get any stream output in the webui. 

If I manually run ` /kvmapp/kvm_stream/kvm_stream` I get a few errors, but these are the logs towards the end when it inevitably crashes.

```
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1920
pstSize->u32Height = 1080
maix multi-media init ok
try release vio ok
try release venc ok
CONFIG_MAIXCAM_LIB_ENC-0
CONFIG_MAIXCAM_LIB_ENC-1
CONFIG_MAIXCAM_LIB_ENC-2
-- [E] Trigger signal, code:SIGSEGV(11)!

maix multi-media driver destroyed.
ISP Vipipe(0) Free pa(0x8cae8000) va(0x0x3fe2364000)```

abuss, 2024-09-23:
I have the same issue with my NanoKVM Like using release 1.2.0 and app version 2.0.7.
The output I'm getting running the same command `/kvmapp/kvm_stream/kvm_stream` is:

```
# /kvmapp/kvm_stream/kvm_stream
[KVM_stream]:start
kvm_cfg.fps = 60
kvm_cfg.qlty = 60
kvm_cfg.width = 1920
kvm_cfg.height = 1080
VI RES = 1920 * 1080 
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
1920 * 1080
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
You may have repeatedly initialized maix multi-media![_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 1920
pstSize->u32Height = 1080
alloc_buf_size(294912): 294912, phy_addr(8cae8000): 0x8cae8000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8cae8000, ion_data.name:ISP_SHARED_BUFFER_0
relese PoolId: 0, PhysAddr: 0x8b300000, BlkSize: 6266880, BlkCnt: 4 Free: 2
try release poolid:0 phy:0x8b300000
try release poolid:0 phy:0x8b8fa000
try release poolid:0 phy:0x8bef4000
try release poolid:0 phy:0x8c4ee000
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
ISP Vipipe(0) Allocate pa(0x8d11f000) va(0x0x3fcdc38000) size(291424)
pstSize->u32Width = 1920
pstSize->u32Height = 1080
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fce37f2e8
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
pstSize->u32Width = 1920
pstSize->u32Height = 1080
20240923 19:44:06.601 2870 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20240923 19:44:06.621 2870 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647       
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1              
sensorName0    6911           

PQBIN message
gerritId:      
              commitId:      fc6ce647       
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1              
sensorName0    4653           

author:        lxowalle       desc:          gc4653 30fps   
createTime:    2024-03-26 13:41:34version:                      
tool Version:       v3.0.5.48           mode:      
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1920
pstSize->u32Height = 1080
VI_SDK_IOC_S_CTRL - vi_sdk_enable_chn NG, No buffer space available
ISP Vipipe(0) Free pa(0x8d11f000) va(0x0x3fcdc38000)
-- [E] Trigger signal, code:SIGSEGV(11)!
```

Z2Z-GuGu, 2024-09-24:
Hi, please unplug the HDMI cable and execute '/etc/init.d/S95nanokvm restart_stream'. I need to check the log at this time.

bfayers, 2024-09-24:
> Hi, please unplug the HDMI cable and execute '/etc/init.d/S95nanokvm restart_stream'. I need to check the log at this time.

[log.txt](https://github.com/user-attachments/files/17111940/log.txt)

Z2Z-GuGu, 2024-09-24:
@bfayers I think I've found the cause, please execute the following command in the terminal
``` shell
echo 1920 > /kvmapp/kvm/width
echo 1080 > /kvmapp/kvm/height 
sync
/etc/init.d/S95nanokvm restart_stream
```
may be effective

bfayers, 2024-09-24:
> @bfayers I think I've found the cause, please execute the following command in the terminal
> 
> ```shell
> echo 1920 > /kvmapp/kvm/width
> echo 1080 > /kvmapp/kvm/height 
> sync
> /etc/init.d/S95nanokvm restart_stream
> ```
> 
> may be effective

Unfortunately still nothing 

[log2.txt](https://github.com/user-attachments/files/17113419/log2.txt)

Z2Z-GuGu, 2024-09-24:
> Unfortunately still nothing
It looks like the problem is still not solved, emm, perhaps you can try to re-burn the image to solve this problem, refer to [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)

bfayers, 2024-09-24:
> > Unfortunately still nothing
> 
> It looks like the problem is still not solved, emm, perhaps you can try to re-burn the image to solve this problem, refer to [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)

Completely fresh burned image + updated from UI, instead of with the command line and it works.

I suspect the issues were to do with using another device's device_key and now that mine was added in #49 it's all good.

closed 2024-09-24 by bfayers