sipeed/NanoKVM #62: Feature request: Support USB-C DisplayPort and Wifi

ptrinh, 2024-09-22:
So that the whole NanoKVM could have power, data, video on a single USB-C connection, like a type-C thumb drive.

Zepan, 2024-09-24:
thank you, we will conside it in our future product!

Zepan, 2024-09-24:
Just make a twitter vote: https://x.com/SipeedIO/status/1838420384233771038

ptrinh, 2024-09-25:
Seems majority agreed with this. Do you have an ETA for the next USB KVM product?

vldm, 2024-09-26:
@ptrinh what is idea behind this design?
1. You can't power-on/reset computer
2. I am not sure if it possible to see bios at all.
3. You only limited in thunderbolt/usb4 devices, which are usually laptops.

Is it just to have simple "remote assistant"?

ptrinh, 2024-09-26:
1. I don't need power-on/hard-reset, as my mini PC servers are always on. I only need this KVM device to reinstall the OS.
2. Yes, with USB-C DP Altmode.
3. Not just laptop, but for the mini PC servers like Minisforum MS-02, which is a new trend for homelab servers.

philhug, 2025-02-20:
@ptrinh what hat are you using for USB-C altmode input on the Rpi Zero 2W?