sipeed/NanoKVM #63: Root partition of 8GB

SuperKali, 2024-09-23:
Hi, I received the NANOKVM today, I noticed that the root partition is maximum 8GB but in the kit there is a 32gb sd card, it would be good to porter use the remaining gb for the /data folder for the iso. there's way to solve the problem? Thanks!

matu6968, 2024-09-23:
> Hi, I received the NANOKVM today, I noticed that the root partition is maximum 8GB but in the kit there is a 32gb sd card, it would be good to porter use the remaining gb for the /data folder for the iso. there's way to solve the problem? Thanks!

You can resize the /data partition using something like gparted.

SuperKali, 2024-09-23:
> You can resize the /data partition using something like gparted.

But this should be made default

matu6968, 2024-09-23:
> But this should be made default

IDK why did Sipeed did not bother with resizing the SD card by default (even a simple one time init script if it detects the /data size is not sufficient will do resizing by itself)

Z2Z-GuGu, 2024-09-24:
Hi, I have set the root partition to be fixed at 8GB (to facilitate developers to add various functions to NanoKVM). The remaining space on the TF card is used to create an exfat format partition, mainly for mounting virtual USB storage disks. The Full version defaults to a 32GB TF card, and the remaining approximately 23GB of space is allocated as a third partition, mounted in the/data directory. I consider 8GB/23G to be sufficient for developers/virtual USB drives, but if you have different needs, you can modify the/etc/init.d/S01fs modification ratio under the root partition in the TF card after burning the system (before powering on NanoKVM)

SuperKali, 2024-09-24:
> Hi, I have set the root partition to be fixed at 8GB (to facilitate developers to add various functions to NanoKVM). The remaining space on the TF card is used to create an exfat format partition, [...]

Thank you for your answer, I have to check this, actually i can't, i will update this issue soon i can

closed 2025-06-20 by SuperKali