sipeed/NanoKVM #66: Only 128 MB RAM ?

ddscentral, 2024-09-25:
The system only shows 128MB RAM even though the specs declare 256MB
During boot, it seems part of memory is reserved for something called "Ion", did a quick search but could not find what this is used for. Well, apart from a comment in a patch for the Lichee NanoRV board, (which increased the Ion reservation to 105 MB) mentioning "multimedia cache".

I believe it would be good to mention the RAM situation somewhere in the docs.

Z2Z-GuGu, 2024-09-27:
SG2002 requires pre allocation of memory for the multimedia module during compilation, and the 131M seen in the system is the remaining memory for user space. We have continuously modified the proportion of this memory in multiple tests. As of now, 105M is used for the multimedia module, and NanoKVM will use this memory for image acquisition and subsequent image processing. I will add an explanation of this in the future

closed 2024-09-27 by Z2Z-GuGu

ddscentral, 2024-09-27:
Understood, thanks.