sipeed/NanoKVM #67: Main app running from /tmp ?

ddscentral, 2024-09-25:
More of a question than an issue, do drop me a link if there's a better place to ask this.

Is there a reason the server and streaming apps are executed from /tmp instead of their native location of /kvmapp ?
/tmp resides in RAM and the device only has 128 MB accessible. The apps in /tmp take quite a sizeable chunk of this.

FlorianHeigl, 2024-10-04:
not knowing the original reasons for it and I haven't taken a peek at this yet, but launching them from a ramdisk can mean the service doesn't die as easily on SD card failure. not as nice as booting to ram or so, but for 128M it can be a good choice for resilience.

wj-xiao, 2024-10-10:
> Is there a reason the server and streaming apps are executed from /tmp instead of their native location of /kvmapp ?

The services are executed from /tmp to avoid file corruption and potential service outages. 
You can review and modify the startup method by this script `/etc/init.d/S95nanokvm`. 

> /tmp resides in RAM and the device only has 128 MB accessible. The apps in /tmp take quite a sizeable chunk of this.

128MB of memory is allocated to the kernel. This might be adjusted in the future to allocate more memory to the user.

closed 2024-11-08 by wj-xiao