sipeed/NanoKVM #71: Feature request: OpenVPN

ddscentral, 2024-09-27:
It would be beneficial to have OpenVPN added to the stock distribution as it is more universally supported than Tailscale. This would add another way to access KVM devices behind NAT.

I have cross-compiled OpenVPN from source using the LicheeRV-Nano sysroot and toolchain and it runs fine. I can access the KVM through it. Only needs a few extra libs installled, all of which are easy to cross-compile using the provided toolchain and sysroot.

BTW, if anyone needs OpenVPN on the NanoKVM right now, I can provide a tarball with prebuilt binaries.

SuperKali, 2024-09-27:
I think it is an overkill feature, the nanokvm have a constant CPU utilization, openvpn on only a single core would make the stream go unstable

ddscentral, 2024-09-27:
Well, it never went unstable for me.
It handled 720p terminal over VPN just fine for me even with a slow 5G as upstream (my backup connection).
With 1080p latency did increase, but it was still fine when accessing locally even with stream over VPN running simultaneously so my guess the increase is caused by slow internet connection.

But jeez, MJPEG sure eats a ton of bandwidth. A better codec would really help with latency when connecting remotely. We really need H264 on this thing...

ptrinh, 2024-09-29:
Just use Tailscale. Much better.

ddscentral, 2024-09-30:
I would not call it "much better". First of all, anything that requires me to create an account on a third party website just to manage local network resources is a showstopper for me. Wireguard network ain't rocket science to set up, you do not need any accounts for this.

ddscentral, 2024-09-30:
> Just use Tailscale. Much better.

I would not call it "much better". First of all, anything that requires me to create an account on a third party website just to manage local network resources is a showstopper for me. Wireguard network ain't rocket science to set up, you do not need any accounts for this.

ddscentral, 2024-10-03:
A bit offtopic. I've also compiled SoftEther client for for testing. It's an L2 VPN but is about twice as fast as OpenVPN even over TCP. Not as easy to setup though.
Won't ask to add it as it's niche and not as well known.

MatejKovacic, 2025-01-09:
I would also like to see OpenVPN on this device. Especially because it supports TCP connections.

Also, @ddscentral can you provide instructions how to compile OpenVPN binary?

ddscentral, 2025-01-10:
@MatejKovacic I would not attempt this, unless you have experience with Linux and cross-compiling software since this is non-trivial to set up. Cross-compiling almost never is.

Baisically, it boils down to building the the LicheeRV-Nano buildroot, which can be found here: https://github.com/sipeed/LicheeRV-Nano-Build and then cross-compiling OpenVPN using the same RISC V toolchain against this buildroot.
From what I recall, there are some extra dependencies which you will also need to build against the same buildroot and install them into it. 
After building OpenVPN, you will need to upload the main binary as well as any missing dynamically linked libraries to the NanoKVM. 

To use OpenVPN, you will need to setup a Linux service to auto-connect at boot. Or connect manually from terminal.

If you want, I can upload the pre-built binaries somewhere.

MatejKovacic, 2025-01-10:
Well, I am tempted to, but you know... don't trust things you did not build by yourself. :)

(And yes, I understand I am trusting NanoKVM developers...) :)

ddscentral, 2025-01-11:
That is true :) You should not trust random binaries from the internet.

If you know how to build from source, the major different between native compile and cross-build is the extra setup options required. Different CC and CXX compilers, PKG_CONFIG, build prefix, etc. 
Also, different build systems (autoconf, cmake, meson, etc.) need different options.

Sipeed uses Docker for LicheeRV Nano buildroot environment, which already has everything you need to compile binaries for LicheeRV Nano board, which NanoKVM is based on.

ddscentral, 2025-06-12:
I have made a tarball for my internal use, I will try to dig it out of one of my KVM units,
Github CI is possible but would take some effort, since the build depends on NanoKVMs buildroot and also some extra depenencies.
Ideally, you'd want a Docker image with everything pre-built inside and use that.

I have since switched to SoftEther since their client is more lightweight and seems better optimized than OpenVPN (server-side does support OpenVPN too). But it's even harder to get cross-built than OpenVPN.

Ruinland, 2025-12-01:
Looking forward to an OpenVPN integration as well.
Sometimes machines are behind heavily restricted environment and only HTTP proxies are allowed.
By setting OpenVPN server node listening on 443 port with encapsulating HTTP traffics, this could be circumvented. I managed to hack around a solution but would love to see it gets integrated.

dormancygrace, 2026-09-14:
Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

OpenVPN is integrated into Settings → VPN, with .ovpn profile import, connection controls and persistent profiles. The GUI uses OpenVPN 3 Core with the upstream Linux ovpn DCO driver.

Routed TUN profiles are supported. TAP, executable scripts and interactive SSO/MFA are not supported, and arbitrary provider compatibility is not guaranteed.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/docs/VPN.md#openvpn).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!